crankyoldgit/IRremoteESP8266 #389: Responding to IR signals (Unknown decoding)

Birdbrainuk, 2018-01-07:
I have successfully extracted raw data using irecvdump V2 and can also send the raw data to control my device (ceiling fan and light).

What I am trying to do is detect when a button is pressed on the real remote and respond depending on which button is pressed so my home automation will know if the device has been tuned on directly

The data is 48 bits long and although unknown encoding the code values are fairly consistent.

How can I compare the code value with maybe an if statement.

Hope you can help

crankyoldgit, 2018-01-08:
The answer you are looking for is _sort of_ located here: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#my-remotes-signal-is-being-reported-as-an-unknown-code-what-does-that-mean-how-do-i-send-an-unknown-code

The _UNKNOWN_ code value is about as good an option you are going to get for matching a signal without a _proper_ decode routine. e.g. It will _try_ to give you the same "code" for the same message again and again, but it is FAR from perfect/good/reliable. If you want something reliable, you are going to need to write a proper decoder for it.

**If** all you are using the IR receiver connected to the ESP is for capturing this one IR device, you can probably significantly reduce the `TIMEOUT` values, which will reduce the chance a good signal will collect some additional IR noise at the end of it. For example, if you set it to time out after 20ms or so, rather than say 70ms, there is less chance the IR receiver will pick up some random noise and throw off the computed UNKNOWN hash value/code.

> How can I compare the code value with maybe an if statement.

Try something like:
```
void loop() {
  // Check if the IR code has been received.
  if (irrecv.decode(&results)) {
   if (results->decode_type == UNKNOWN && results->value == CODEYOUWANT) {
    // Do action for code/value CODEYOUWANT etc here.
   }
  }
}
```

Birdbrainuk, 2018-01-08:
Thanks for the support got it working.

closed 2018-01-09 by crankyoldgit