crankyoldgit/IRremoteESP8266 #392: Cover for IRBlaster using IRremoteESP8266 library

leobel96, 2018-01-13:
Hi, I created a 3d printed cover and a step by step guide to make a IRBlaster using your library [here](https://www.thingiverse.com/thing:2756041). Feel free to include it in your wiki!

roidayan, 2018-01-13:
Nice!
I started working on an enclosure as well and I want to include the IR receiver led, temperature sensor, light sensor and maybe rgb led.

leobel96, 2018-01-13:
Great! If you want you can start modifying the one I designed. On thingiverse I included the file .step

crankyoldgit, 2018-01-14:
This is a much better design than the square (horrible) box I used for mine. Top stuff.

I've added a Wiki page for user projects. Yours is at the top of the list. ;-)
https://github.com/markszabo/IRremoteESP8266/wiki/Related-User-Projects

closed 2018-01-14 by crankyoldgit

leobel96, 2018-01-14:
As I've written on thingiverse under the project, i'm Pretty new in 3d modelling so there could be imperfections in the model and I didn't care about tolerances because I wanted everything fitting perfectly and staying fixed without glue. So feel free to edit the CAD file I uploaded according to you needs

crankyoldgit, 2018-01-14:
Let me know if you want to upload the CAD/mesh files to this git repository directly, or if you have your own etc. Otherwise, the thingiverse link is probably sufficient.

leobel96, 2018-01-14:
It's indifferent: feel free to download the CAD/mesh from thingiverse and upload it here if you want.

crankyoldgit, 2018-01-14:
I'll leave it as is.