crankyoldgit/IRremoteESP8266 #396: Default garbage

Legantmar, 2018-01-20:
I use esp8266-12 and VS1838
After upload script and start I see garbage code in serial port monitor.
But I not send IR code
What is ?
what could be the problem?
how to solve it?
[screen link](https://yadi.sk/i/zuN78R1-3RdTah)

crankyoldgit, 2018-01-21:
Try using [IRrecvDumpV2.ino](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) instead of `IRrecvDemo.ino`

The "default garbage code" is likely Infra-red/Signal noise.

**Note:** You are using a IR module designed for **5 Volts** (The Uno's GPIOs are 5V, the ESP8266's are 3.3V) It will probably work most of the time however.

Most of the other hex values reported are [UNKNOWN](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#my-remotes-signal-is-being-reported-as-an-unknown-code-what-does-that-mean-how-do-i-send-an-unknown-code) signals, if they bother you you can disable UNKNOWN detection by changing the the value of `DECODE_HASH` to `false` [here](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L57).

Legantmar, 2018-01-21:
crankyoldgit, thank you!
DECODE_HASH to false - it helped me
VS1838 Vcc 2,2 - 6.0 V https://yadi.sk/i/zxIPCAaK3RdpPq
this works for 3.3V

closed 2018-01-21 by crankyoldgit