crankyoldgit/IRremoteESP8266 #398: Where is gpio.h?

SteveRMann, 2018-01-22:
I downloaded the zip and used the IDE library manager to install IRremoteESP8266 (_v2.3.1 (20171229)).
I loaded the example IRrecvDumpV2 and got an error, missing "gpio.h"

Is there a GPIO library that I should also add to my IDE?  There's not one in the Libraries Manager.

crankyoldgit, 2018-01-22:
I'm guessing the "gpio.h" file is this one: https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/gpio.h

Did you install the support for the ESP8266  devices via the "Board Manager" and selected an ESP8266 board as the build target/platform?
(See https://github.com/esp8266/Arduino#installing-with-boards-manager)

It (_gpio.h_) is certainly **not** something that is supplied by our library. I believe, it's part of the core Arduino _stuff_.

SteveRMann, 2018-01-23:
DOH!
Yes, I've been working with various ESP8266 devices for a while, so the various libraries for the ESP8266 have been installed.  Mostly through the Board Manager.  The DOH! is the second part of your reply.  I had been working recently on the UNO, and I had not changed the Board Type.
So, I learned something today.
It compiles just fine now.
For a background, I have been working with a few ESP8266 nodes around the house as IOT modules.  I have a UNO that records and transmits IR codes for everything but my VIZIO TV.  I just cannot get the VIZIO to respond to the recorded codes.  Which is why I wanted to try the NodeMCU board and this library.  Ultimately I want to be able to tell Alexa to turn on NBC NEWS, and an esp8266 node will respond by sending the IR codes to turn on the TV and the Satellite receiver then set the channel and maybe dim the lights a bit...  The only piece of the chain that I am missing is the IR control of the VIZIO TV.
Thanks very much for our quick reply.

closed 2018-01-23 by SteveRMann

crankyoldgit, 2018-01-23:
@SteveRMann  RE: Vizio codes. It _could_ be that they are at a different modulation frequency than the IR receiver module you are using. e.g. The most common IR modulation frequency is 38kHz. Most receivers are typically that. However, you can get subtle garbage in a capture if the transmitter and receiver freq. are not the same. (see the troubleshooting guide. This issue has affected a number of people many times.

Alternatively, you could try using a [GlobalCache code](https://irdb.globalcache.com/Home/Database)  (via `sendGC()`) for Vizio devices and see if it responds. If it does, you can see if the library detects it by using https://github.com/markszabo/IRremoteESP8266/blob/master/tools/gc_decode.cpp. It might give you are more manageable native protocol/code to use.

Or try the Pronto codes [here](http://www.remotecentral.com/cgi-bin/forums/viewpost.cgi?925066) via `sendPronto()`, they might work for you too.

`gc_decode` will similarly convert pronto codes into native codes if the library supports it.