crankyoldgit/IRremoteESP8266 #399: One more question, please

SteveRMann, 2018-01-23:
In the Wiki page (wiki#ir-receiving), the following appears:

> If ESP8266 interrupt pin is attached to a LED then make sure to add the following line in setup. Serial.begin(115200,SERIAL_8N1,SERIAL_TX_ONLY);

What interrupt pin?

Thanks

crankyoldgit, 2018-01-23:
Thanks for the feedback. Yeah, that's completely unreadable. (Thankfully I didn't write it, this time ;-)

I've updated/corrected it. Can you please take another look and see if that makes it clear(er)?

SteveRMann, 2018-01-23:
That makes a lot more sense.  Maybe a cut-n-paste error?  I can see where incoming serial data could screw with the timing.  I am not very experienced with Git, but is this the appropriate vector for asking such questions?

closed 2018-01-23 by SteveRMann

crankyoldgit, 2018-01-25:
@SteveRMann I have no qualms with this sort of question logged as an issue. So, keep reporting things you find. :-)