crankyoldgit/IRremoteESP8266 #400: sendJVC() freezes on sending Power command (0xC5E8)

schelm87, 2018-01-25:
### Version/revison of the library used
 "version": "2.3.1",

### Expected behavior
- recorded PowerONOFF Command of my JVC Receiver ( 0xC5E8 ) 
- send 0xC5E8 via `irsend.sendJVC(JVCPower, 16, 1);`
- code stopps working (see below)

For verification, I tried another key press (Sleep) - which was recorded as 0xC5C0. 
This works fine. Command was done correctly by my device. 

### Code 
```
//sleep Button pressed...
	//code received...
	//	Timestamp : 000011.600
	//	Encoding : JVC
	//	Code : C5C0(16 bits)

//Power Button pressed
	//code received...
	//	Timestamp : 000014.601
	//	Encoding : JVC
	//	Code : C5E8(16 bits)

#define JVCPower          0xC5E8 //doesn't work...
#define JVCSleep			  0xC5C0  //works fine
	Serial.println("send sleep...");

	irsend.sendJVC(JVCSleep, 16, 1);
	Serial.println("1...");
	delay(100); 
	Serial.println("send power...");
	irsend.sendJVC(JVCPower, 16, 1); 
	Serial.println("3...");
	delay(100);
	Serial.println("4...");

//SERIAL OUTPUT: 
//	send sleep...
//	1...
//	send power...
// <NO FURTHER OUTPUT AFTER THAT>
```

The ordering of sending those two commands does not have any effect - JVCPower always stops the code working, regardless if send first or second.

crankyoldgit, 2018-01-25:
Thanks for the detailed report. Yep, that's a bug alright. Sorry it affected you.
Can you please try the fix in this branch https://github.com/markszabo/IRremoteESP8266/tree/issue-400 and let me know if that fixes it for you or not?

schelm87, 2018-01-26:
Hi,
**thanks for your very quick fix.** 
3382348 works perfect for 0xc5e8 and still works for 0xc5c0. 

Just for clearification: 
the Sample "JVCPanasonicSendDemo" also works well now. 

It could be simplified (at least with my device) by removing the second send command and adding a repeat to the first. 
```
  irsend.sendJVC(JVCPower, 16, 1);  // hex value, 16 bits, **REPEAT**
  // see http://www.sbprojects.com/knowledge/ir/jvc.php for information
  //delayMicroseconds(50);
  //irsend.sendJVC(JVCPower, 16, 1);  // hex value, 16 bits, repeat
  //delayMicroseconds(50);
```

closed 2018-01-26 by schelm87

crankyoldgit, 2018-01-26:
FYI, I've just release v2.3.2 of the library which contains this fix. If you are using the Arduino IDE, it should be available via the Library Manager in less than 24 hours.