crankyoldgit/IRremoteESP8266 #404: Haier AC protocol

kuzin2006, 2018-01-27:
Hello, as far as I see, your lib does not have Haier AC remote protocol support. I've made reverse engineering of my AC protocol, hope it will be useful for you. All these codes are tested and fully functional with my AC unit. I use it in my home automation project, so I have no relevant universal code to provide, hope this data will help somebody to add support for one more AC protocol.

Haier HSU07-HEA03 Remote control protocol.

Known protocol details
   Pulses 1,2,3 - 2850..3100
   Pulse 4 - 4300..4350
   Marks - 500
   Spaces used to transmit data:
       Short(0) - 600..850
       Long(1) - 1700..1950

   Full Raw Data packet length - 149 pulses. Pulses 5-149 encode the data packet
   Full Data Packet length - 9 bytes / 72 bits / 144 pulses

   Last byte is CRC - SUM of previous 8 bytes modulo 0x100

Code length is 8 byte + 1 byte CRC

Bit numbers:

[0-7] Prefix
0xA5 10100101

[8-11] TEMP
0x00 0000 0 - 16C
...
0x0E 1110 14 - 30C

[12-15] CMD
0000 - Off
0001 - On
0010 - Mode
0011 - Fan
0110 - Temp UP
0111 - Temp DOWN
1000 - Sleep
1100 - Health (turns on ion generator)
1001 - Timer SET
1010 - Timer Cancel

[16-17] Swing
00 - Off
01 - Up
10 - Down
11 - Change

[18] Not Used
1

[19-23] CUR H24 - current time, hours (as set on remote)
0-23 hours

[24-25] Timer Mode
00 - Cancel
01 - Off Timer
10 - On Timer
11 - On/Off Timer

[26-31] CUR MINS - current time, minutes (as set on remote)
0-59 minutes

[32-33] Not Used
00

[34] Health
0 - Off
1 - On

[35-39] Timer OFF HR24
0-23 hours
 
[40-41] Fan speed
00 - Auto
01 - 3 (max)
10 - 2 
11 - 1 (min)

[42-47] Timer OFF MINS
0-59 minutes

[48-50] Mode
000 - Auto
001 - Cool
010 - Dry
011 - Heat
100 - Fan

[51-55] Timer ON HR24
0-23 Hours

[56] Not Used
0

[57] Sleep
0 - Off
1 - On

[58-63] Timer ON Mins
0-59 Minutes

[64-71] CRC, SUM of 8 previous bytes

crankyoldgit, 2018-01-28:
Thanks for that data. Hopefully someone might take up creating a send() & decode() routines etc for it.

Can you please provide some raw captures from `IRrecvDumpV2` to help them, and a description of what settings were on the remote for that message etc.? That will aid them knowing if the code they have works as expected.
Also, what frequency is required? (e.g. 38kHz etc) and do you happen to know the duty cycle at all?

FYI, "pulses" (also known as marks) are only the odd numbered entries in the raw data. The even ones are known as spaces (or a complete absence of any pulses). And to even get more technical each "mark" is actually made up of lots of very small short "pulses" and "gaps" typically at a freq of 38kHz. The ratio of the duration of these mini pulses and gaps in a "mark" is the duty-cycle ratio for the signal. (we default to 50:50 if unknown).

kuzin2006, 2018-01-28:
I have my results in [MS Excel table](https://www.dropbox.com/s/mecyib3lhdxc8c6/IR%20data%20reverse%20engineering.xlsx?dl=0), already with durations conversion to 0s and 1s. First attempts to read codes were with Arduino Uno (IR Remote by shirrif), and raw durations were slightly different from your Lib.

Now on NodeMCU 0.9 i have stable detection with durations range 500-800 for 0 and 1600-1950 for 1.

Stable rawData sending is with 3050, 3150, 3050, 4400 for starting pulses, 750 for 0, 1950 for 1, and 560 for marks (odd). I transmit at 38 kHz and did not experiment with freq variations yet.

crankyoldgit, 2018-01-28:
@kuzin2006 The reason I'm asking for the _Raw_ data is we have a number of [tools](https://github.com/markszabo/IRremoteESP8266/tree/master/tools) and [tests](https://github.com/markszabo/IRremoteESP8266/blob/master/test/ir_Daikin_test.cpp#L753) which can use that data to replay it to ensure that any sending code produces something similar, and that any decode routines can handle a "real" message correctly.

You've done excellent work decoding it. That is certainly the part that requires the most effort. This is just to help any library developers who come along later to be able to validate stuff when they don't have access to the devices in question.

The _raw_ data as reported from the Uno ([IRremote arduino library IRrecvDumpV2](https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino)) is fine.

kuzin2006, 2018-01-28:
collected some raw data:

ON COOL 16C FAN LOW ALL OFF 12:00AM
uint16_t rawData[149] = {3030, 3044,  3030, 4304,  576, 1694,  550, 582,  552, 1704,  552, 714,  550, 582,  550, 1706,  552, 582,  550, 1836,  552, 582,  578, 568,  550, 582,  550, 714,  550, 582,  550, 582,  552, 582,  550, 1836,  552, 582,  552, 580,  580, 1692,  550, 712,  552, 582,  550, 582,  552, 580,  550, 714,  552, 582,  550, 582,  552, 582,  578, 698,  552, 580,  552, 582,  552, 582,  552, 1836,  552, 580,  552, 582,  552, 582,  550, 714,  578, 568,  550, 582,  550, 582,  552, 714,  550, 1706,  550, 1706,  550, 582,  550, 714,  552, 582,  580, 566,  552, 582,  550, 714,  552, 580,  552, 580,  552, 1706,  550, 714,  550, 582,  552, 582,  578, 568,  552, 712,  552, 582,  550, 582,  550, 582,  550, 712,  552, 582,  550, 582,  552, 582,  578, 722,  552, 1704,  550, 582,  550, 1706,  550, 736,  550, 582,  550, 1706,  550, 1704,  552, 1704,  578};  // UNKNOWN D22E7E78
OFF
uint16_t rawData[149] = {3026, 3044,  3032, 4304,  578, 1692,  550, 582,  550, 1706,  552, 712,  552, 580,  552, 1704,  552, 582,  552, 1838,  550, 582,  580, 566,  550, 582,  550, 714,  552, 582,  550, 582,  552, 582,  550, 714,  550, 582,  552, 582,  578, 1690,  552, 712,  552, 582,  550, 582,  550, 582,  550, 714,  550, 582,  552, 582,  550, 582,  578, 700,  550, 582,  550, 582,  550, 582,  550, 1838,  550, 582,  552, 580,  552, 582,  552, 712,  578, 568,  550, 582,  550, 582,  550, 714,  550, 1704,  552, 1706,  550, 582,  550, 714,  550, 582,  578, 566,  552, 582,  552, 712,  550, 582,  552, 580,  552, 1704,  552, 714,  550, 582,  552, 582,  578, 568,  550, 714,  552, 582,  552, 582,  550, 582,  550, 714,  550, 582,  552, 582,  552, 582,  578, 722,  552, 1706,  550, 582,  552, 1704,  552, 736,  552, 582,  550, 1704,  552, 1704,  552, 582,  580};  // UNKNOWN 9BB5B2B5
ON
uint16_t rawData[149] = {3032, 3046,  3028, 4304,  576, 1692,  552, 582,  550, 1706,  550, 714,  550, 582,  552, 1706,  550, 582,  552, 1836,  552, 582,  578, 568,  552, 580,  552, 714,  550, 582,  550, 582,  552, 582,  550, 1836,  552, 582,  550, 582,  578, 1692,  550, 712,  552, 582,  552, 582,  550, 582,  550, 714,  552, 582,  550, 582,  552, 582,  578, 700,  550, 582,  550, 582,  550, 582,  552, 1838,  550, 582,  552, 582,  552, 582,  550, 714,  578, 568,  550, 582,  550, 582,  550, 714,  552, 1706,  550, 1704,  550, 582,  552, 714,  550, 582,  578, 568,  550, 582,  550, 714,  552, 582,  550, 582,  550, 1706,  550, 714,  550, 582,  550, 582,  578, 568,  550, 714,  550, 582,  550, 582,  552, 582,  550, 714,  550, 582,  552, 582,  552, 582,  578, 722,  552, 1704,  552, 582,  550, 1706,  552, 736,  550, 584,  550, 1706,  550, 1706,  550, 1704,  578};  // UNKNOWN D22E7E78
TEMP 16C
uint16_t rawData[149] = {3030, 3042,  3032, 4304,  576, 1694,  550, 582,  552, 1706,  550, 712,  552, 580,  552, 1706,  550, 582,  550, 1836,  550, 582,  578, 568,  552, 582,  552, 1834,  552, 582,  550, 1706,  552, 1706,  548, 714,  552, 580,  552, 582,  578, 1690,  552, 714,  550, 582,  552, 582,  550, 582,  552, 712,  550, 582,  550, 582,  552, 582,  578, 700,  550, 582,  552, 582,  552, 582,  550, 1836,  552, 582,  550, 584,  550, 582,  550, 712,  580, 568,  550, 582,  550, 584,  550, 712,  552, 1704,  552, 1704,  552, 580,  552, 714,  550, 582,  580, 566,  552, 580,  552, 712,  552, 582,  550, 582,  552, 1704,  552, 712,  552, 582,  550, 582,  580, 566,  552, 712,  552, 582,  552, 582,  550, 582,  550, 714,  550, 582,  550, 582,  550, 582,  578, 722,  550, 1706,  552, 582,  550, 1704,  552, 1860,  552, 1704,  552, 1704,  552, 582,  550, 582,  580};  // UNKNOWN F35ACDF0
TEMP 17C
uint16_t rawData[149] = {3030, 3042,  3032, 4306,  574, 1692,  550, 582,  550, 1704,  552, 712,  552, 582,  552, 1704,  550, 582,  550, 1836,  552, 582,  578, 568,  552, 1704,  552, 714,  550, 582,  550, 1706,  550, 1706,  552, 712,  550, 582,  552, 580,  578, 1688,  554, 712,  552, 582,  550, 582,  552, 582,  550, 714,  550, 582,  552, 582,  550, 582,  580, 698,  550, 582,  552, 582,  550, 582,  550, 1838,  552, 580,  550, 582,  550, 582,  552, 712,  580, 568,  550, 582,  550, 582,  552, 714,  550, 1704,  552, 1706,  552, 580,  552, 712,  550, 582,  578, 568,  550, 582,  550, 714,  550, 582,  552, 582,  550, 1704,  552, 714,  552, 582,  550, 582,  578, 568,  550, 714,  550, 582,  552, 582,  550, 582,  550, 714,  552, 582,  550, 582,  550, 582,  578, 722,  552, 1704,  552, 1706,  552, 582,  550, 738,  552, 1706,  552, 1706,  550, 582,  552, 582,  578};  // UNKNOWN 4046F041
TEMP 18C
uint16_t rawData[149] = {3030, 3070,  3004, 4304,  576, 1694,  552, 580,  550, 1706,  550, 714,  550, 582,  552, 1706,  550, 580,  552, 1836,  552, 582,  578, 568,  550, 1706,  550, 1838,  550, 580,  552, 1704,  552, 1706,  550, 712,  550, 582,  552, 582,  578, 1692,  552, 714,  550, 582,  550, 582,  550, 582,  552, 714,  550, 582,  550, 582,  550, 582,  578, 698,  552, 582,  550, 582,  550, 582,  552, 1836,  552, 580,  550, 584,  550, 582,  552, 712,  578, 568,  552, 582,  550, 582,  550, 712,  552, 1706,  554, 1702,  552, 580,  550, 714,  552, 582,  578, 568,  550, 582,  552, 712,  550, 582,  550, 582,  552, 1704,  554, 710,  552, 582,  552, 582,  578, 568,  552, 712,  552, 582,  550, 582,  552, 582,  550, 714,  550, 582,  550, 582,  550, 582,  580, 722,  550, 1704,  552, 1704,  552, 582,  550, 1862,  550, 1704,  552, 1704,  552, 582,  550, 582,  578};  // UNKNOWN 1575D184
TEMP 19C
[... cut ...] 2,  550, 582,  550, 1836,  552, 582,  552, 582,  550, 582,  550, 714,  578, 568,  550, 582,  550, 582,  552, 712,  552, 1702,  554, 1704,  552, 582,  550, 714,  550, 582,  578, 568,  552, 582,  550, 714,  550, 582,  552, 582,  550, 1706,  552, 714,  550, 582,  550, 582,  578, 568,  552, 712,  550, 582,  552, 582,  552, 582,  550, 714,  550, 582,  550, 582,  550, 582,  580, 722,  552, 1706,  552, 580,  550, 582,  550, 1862,  552, 1704,  552, 582,  550, 1706,  550, 582,  578};  // UNKNOWN 79D9736A

crankyoldgit, 2018-01-28:
Many thanks. That's perfect.

crankyoldgit, 2018-02-04:
@kuzin2006 I've coded up support for Haier devices. Can you have a look at the code in PR #409 and can you please test it out on the real device(s) by using the [haier_ac branch](https://github.com/markszabo/IRremoteESP8266/tree/haier_ac)? I'm fairly confident it will work.

kuzin2006, 2018-02-09:
Almost finished my project, will try to test soon.

crankyoldgit, 2018-03-02:
This is now live in the v2.3.3 release

closed 2018-03-02 by crankyoldgit

nygma2004, 2018-03-17:
Hi Guys,

Can you help me here. I am trying to set this up for a Haier unit, but I have a YR-H10 remote. When I run the recvdumpv2 program, it is giving me the following code for power off:
uint16_t rawData[149] = {2952, 3124,  2952, 4500,  524, 1696,  524, 586,  526, 1694,  522, 586,  526, 584,  526, 1692,  526, 586,  526, 1678,  524, 1720,  500, 586,  524, 584,  528, 584,  526, 586,  526, 584,  528, 582,  526, 576,  526, 586,  524, 584,  528, 1696,  522, 1696,  524, 584,  526, 1694,  524, 584,  526, 578,  526, 586,  526, 584,  526, 586,  526, 1694,  526, 1692,  526, 1696,  522, 584,  526, 576,  526, 584,  526, 584,  526, 586,  524, 586,  526, 1694,  524, 1696,  522, 586,  526, 576,  526, 1696,  522, 586,  526, 586,  524, 584,  528, 540,  570, 586,  524, 586,  526, 576,  526, 586,  526, 584,  526, 1696,  524, 584,  526, 584,  526, 584,  526, 584,  526, 576,  526, 586,  526, 584,  528, 584,  526, 586,  524, 584,  528, 584,  526, 586,  526, 578,  524, 586,  526, 586,  526, 1696,  522, 584,  526, 586,  526, 586,  526, 586,  526, 1686,  524};  // HAIER_AC
uint8_t state[9] = {0xA5, 0x80, 0x34, 0x1C, 0x0C, 0x80, 0x20, 0x00, 0x21};

If I just do:
irsend.sendRaw(rawData, 149, 38);
nothing happens. How do I send out the state array? Or shall I use the ir_haier class instead of raw codes? Would somebody able to give me a quick sample code?

crankyoldgit, 2018-03-17:
Before you try the following, your `irsend.sendRaw(rawData, 149, 38);` example _should_ work. Have you confirmed your IR sending setup works as expected/with other devices yet?

Sending Haier via 'state': https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Haier.cpp#L45
```
uint8_t state[9] = {0xA5, 0x80, 0x34, 0x1C, 0x0C, 0x80, 0x20, 0x00, 0x21};
irsend.sendHaierAC(state);  // Or irsend.sendHaierAC(state, 9);
```

or via the 'class' (which you can mutate as needed. Not recommended, just here for completeness):
```
#include <ir_Haier.h>

uint8_t state[9] = {0xA5, 0x80, 0x34, 0x1C, 0x0C, 0x80, 0x20, 0x00, 0x21};
IRHaierAC ac(ir_transmitter_gpio_pin);
ac.setRaw(state);
ac.send();
```
FYI, if you are still having problems, please open a new issue for the problem.

mustafamq, 2024-02-19:
Cool