crankyoldgit/IRremoteESP8266 #406: Question . Generic protocol decode

gnkarn, 2018-01-29:
Hi David , me again now with different question , I need to decode an alarm system protocol , I already have this mapped so I do know how it works , and I’m considering to use the library to decode it .

It is 16 bits with a mark space code very similar to the NEC , a bit one is aprox 3000 us mark and a bit zero is aprox 1400 us mark.

My question what is the best example and decode function  in your opinion that i could base my modifications ( if needed as I see the functions are very flexible ) ,
Thanks in advance 
Gus

crankyoldgit, 2018-01-29:
Good question.

At quick glance, https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Nikai.cpp is probably a good place to start/copy from.

You don't need to do the TICK stuff, that's just to try to auto-calibrate the decode signal. It's a bonus thing, not a requirement.

The https://github.com/markszabo/IRremoteESP8266/commit/4d8f74ec9821964e656f42b69bf4eefb5c77adb0 commit which introduced it is a good "example" of what needs to be changed etc in other parts of the library. Obviously, the testing stuff is not required, but I do recommend it. It really helps me when I'm debugging & developing stuff for a new protocol.

Also, for the LARGE banner text, I use: http://asciiset.com/figletserver.html with the "letters" font. ;-)

Hope that helps.

gnkarn, 2018-01-29:
thanks for the hints , I'm on the way to add my mpx_decode : , 
here  i have modified the ir_nikai.cpp for this purpose 
[mpx_decode][https://github.com/gnkarn/IRremoteESP8266-mpx/blob/master/src/mpx_decode.cpp](url)

the define is MPX_DECODE
the decode type is MPX
i know this is far from finished , but by using the IR_receiveDumpV2 example i get lots of errors ..

and this class of errors that i don't know how to solve because they have no line numbers , etc .

for example 
`pioenvs/nodemcuv2/src/IRrecvDumpV2.ino.o:(.text.setup+0x8): undefined reference to `IRrecv::setUnknownThreshold(unsigned short)
'
.pioenvs/nodemcuv2/src/IRrecvDumpV2.ino.o:(.text.setup+0xc): undefined reference to `IRrecv::enableIRIn()'
.pioenvs/nodemcuv2/src/IRrecvDumpV2.ino.o: In function `setup':
IRrecvDumpV2.ino.cpp:(.text.setup+0x36): undefined reference to `IRrecv::setUnknownThreshold(unsigned short)'
IRrecvDumpV2.ino.cpp:(.text.setup+0x3e): undefined reference to `IRrecv::enableIRIn()'
.pioenvs/nodemcuv2/src/IRrecvDumpV2.ino.o:(.text.loop+0x14): undefined reference to `IRrecv::decode(decode_results*, irparams_t*
)'`

so , my question is , how should i organize the files , to be able to modify and make use of the library , without rearranging  all the files .? 
How to make sure the compiler-linker is using the files on my folders and not other version of the library ?

---------
just to help on what I'm trying to accomplish and may help for clarity , this is the signal to decode
[image]

are 16 bits , starting with a space 1400 us , 
a ONE digit represented by a 2800us mark and a 1400 us space

a zero = 1400 us mark and 2800 us space 
so always the corresponding mark + space time remains constant  to 4200 us .

thank you ,.

crankyoldgit, 2018-01-29:
I'll try to clone your repository and see if I can work out what is going on. Those errors are typically something failing in the linking phase. Something likely didn't get compiled and/or referenced correctly. That is the _linker_ doesn't seem to be able to find the IRrecv object definitions/code etc.

Going by the error messages, I'm assuming you are using PlatformIO. That's not a problem. I'm just basing my answers/assumptions around that.

Oh, given you've got nice round numbers for the marks/spaces etc at present, I'd suggest using a tick size of 100 instead of 1, and reduce accordingly etc.

crankyoldgit, 2018-01-29:
Oh, another massive tip. You are MUCH better off doing this as a *branch* in your existing fork of [IRremoteESP8266](https://github.com/gnkarn/IRremoteESP8266).

This will save you a heap of effort when it comes to merging the code back upstream as well as any other updates from the upstream library down to yours.

You can do this by:
```
cd To_the_directory_you're_using_for_your_clone_of_your_fork_of_IRremoteESP8266
git remote add upstream https://github.com/markszabo/IRremoteESP8266.git
git fetch upstream
git merge upstream/master
git checkout master  # Switch to the 'master' branch in IRremoteESP8266
git pull  # Pick up any updates
git checkout -b mpx_testing  # This will create a 'mpx_testing' branch
```

That will maintain all the change/commit history and allow you to do a simple `git diff` etc to see what you've done to the library.

See https://help.github.com/articles/configuring-a-remote-for-a-fork/ & https://help.github.com/articles/syncing-a-fork/

crankyoldgit, 2018-01-29:
Here is a `git diff` of what I had to do to get it to _just_ compile:
```
diff --git a/examples/IRrecvDumpV2/IRrecvDumpV2.ino b/examples/IRrecvDumpV2/IRrecvDumpV2.ino
index 15e8d13..4be8b1e 100755
--- a/examples/IRrecvDumpV2/IRrecvDumpV2.ino
+++ b/examples/IRrecvDumpV2/IRrecvDumpV2.ino
@@ -21,9 +21,9 @@
 #ifndef UNIT_TEST
 #include <Arduino.h>
 #endif
-#include </Volumes/D5-INFO1/IRremoteESP8266-mpx/src/IRremoteESP8266.h>
-#include </Volumes/D5-INFO1/IRremoteESP8266-mpx/src/IRrecv.h>
-#include </Volumes/D5-INFO1/IRremoteESP8266-mpx/src/IRutils.h>
+#include <IRremoteESP8266.h>
+#include <IRrecv.h>
+#include <IRutils.h>
 
 #define DECODE_AC false
 
diff --git a/examples/IRrecvDumpV2/platformio.ini b/examples/IRrecvDumpV2/platformio.ini
index f5044a7..eeb8d1f 100755
--- a/examples/IRrecvDumpV2/platformio.ini
+++ b/examples/IRrecvDumpV2/platformio.ini
@@ -1,3 +1,6 @@
+[platformio]
+lib_extra_dirs = ../../
+src_dir=.
 
 [common]
 build_flags =
diff --git a/platformio.ini b/platformio.ini
index 6e5b75a..5878568 100755
--- a/platformio.ini
+++ b/platformio.ini
@@ -1,5 +1,5 @@
 [platformio]
-;lib_extra_dirs = .
+lib_extra_dirs = .
 src_dir = examples/IRrecvDumpV2
 
 [common]
diff --git a/src/IRrecv.h b/src/IRrecv.h
index 18bee3d..124731e 100755
--- a/src/IRrecv.h
+++ b/src/IRrecv.h
@@ -269,6 +269,10 @@ class IRrecv {
                        uint16_t nbits = CARRIER_AC_BITS,
                        bool strict = true);
 #endif
+#if DECODE_MPX
+  bool decodeMPX(decode_results *results, uint16_t nbits = MPX_BITS,
+                 bool strict = true);
+#endif
 };
 
 #endif  // IRRECV_H_
diff --git a/src/IRremoteESP8266.h b/src/IRremoteESP8266.h
index 51c683d..6995a18 100755
--- a/src/IRremoteESP8266.h
+++ b/src/IRremoteESP8266.h
@@ -59,7 +59,7 @@
 #define SEND_RAW             false
 
 #define DECODE_NEC           false
-#define SEND_NEC
+#define SEND_NEC             false
 
 #define DECODE_SHERWOOD      false  // Doesn't exist. Actually is DECODE_NEC
 #define SEND_SHERWOOD        false
@@ -139,7 +139,7 @@
 #define DECODE_TROTEC        false  // Not implemented.
 #define SEND_TROTEC          false
 
-#define MPX_DECODE           true 
+#define DECODE_NIKAI         false
 #define SEND_NIKAI           false
 
 #define DECODE_TOSHIBA_AC    false
@@ -157,6 +157,9 @@
 #define DECODE_CARRIER_AC    false
 #define SEND_CARRIER_AC      false
 
+#define DECODE_MPX           true
+#define SEND_MPX             true
+
 
 #if (DECODE_ARGO || DECODE_DAIKIN || DECODE_FUJITSU_AC || DECODE_GREE || \
      DECODE_KELVINATOR || DECODE_MITSUBISHI_AC || DECODE_TOSHIBA_AC || \
@@ -202,7 +205,7 @@ enum decode_type_t {
   NEC_LIKE,
   ARGO,
   TROTEC,
-  MPX,
+  NIKAI,
   RAW,  // Technically not a protocol, but an encoding.
   GLOBALCACHE,  // Technically not a protocol, but an encoding.
   TOSHIBA_AC,
@@ -210,7 +213,8 @@ enum decode_type_t {
   MIDEA,
   MAGIQUEST,
   LASERTAG,
-  CARRIER_AC
+  CARRIER_AC,
+  MPX
 };
 
 // Message lengths & required repeat values
@@ -277,7 +281,7 @@ enum decode_type_t {
 #define TROTEC_COMMAND_LENGTH        9U
 #define WHYNTER_BITS                32U
 #define ARGO_COMMAND_LENGTH         12U
-#define MPX_BITS                  16U
+#define MPX_BITS                    16U
 #define MAGIQUEST_BITS              56U
 #define MIDEA_BITS                  48U
 #define MIDEA_MIN_REPEAT             0U
diff --git a/src/IRsend.h b/src/IRsend.h
index ad53938..b905143 100755
--- a/src/IRsend.h
+++ b/src/IRsend.h
@@ -229,6 +229,9 @@ void send(uint16_t type, uint64_t data, uint16_t nbits);
   void sendCarrierAC(uint64_t data, uint16_t nbits = CARRIER_AC_BITS,
                      uint16_t repeat = CARRIER_AC_MIN_REPEAT);
 #endif
+#if SEND_MPX
+  void sendMPX(uint64_t data, uint16_t nbits = MPX_BITS, uint16_t repeat = 0);
+#endif
 
  protected:
 #ifdef UNIT_TEST
diff --git a/src/mpx_decode.cpp b/src/mpx_decode.cpp
index db6c5bf..0885f3f 100644
--- a/src/mpx_decode.cpp
+++ b/src/mpx_decode.cpp
@@ -14,7 +14,7 @@
 // Ref:
 //   https://github.com/markszabo/IRremoteESP8266/issues/309
 #define MPX_TICK             1U
-#define _HDR_MARK_TICKS     4000U
+#define MPX_HDR_MARK_TICKS     4000U
 #define MPX_HDR_MARK         (MPX_HDR_MARK_TICKS * MPX_TICK)
 #define MPX_HDR_SPACE_TICKS    1400U
 #define MPX_HDR_SPACE        (MPX_HDR_SPACE_TICKS * MPX_TICK)
@@ -50,7 +50,7 @@ void IRsend::sendMPX(uint64_t data, uint16_t nbits, uint16_t repeat) {
 }
 #endif
 
-#if MPX_DECODE
+#if DECODE_MPX
 // Decode the supplied MPX message.
 //
 // Args:
```

You should be able to apply the patch using:
`git apply mpx_compile_fix.patch.txt`
[mpx_compile_fix.patch.txt](https://github.com/markszabo/IRremoteESP8266/files/1675589/mpx_compile_fix.patch.txt)
`

gnkarn, 2018-01-30:
i have corrected the differences , but on a separate repo  installing all the  relevant files in one directory ,  but still errors

> src/mpx_decode.cpp:66:76: error: no 'bool IRrecv::decodeMPX(decode_results*, uint16_t, bool)' member function declared in class 
'IRrecv'
bool IRrecv::decodeMPX(decode_results *results, uint16_t nbits, bool strict) {
^
*** [.pioenvs/nodemcuv2/src/mpx_decode.o] Error 1

i don't get the error, by looking at the function on IRrecv.cpp and .h , it seams ok .

crankyoldgit, 2018-01-30:
Answering some of your questions. i.e.
> so , my question is , how should i organize the files , to be able to modify and make use of the library , without rearranging all the files .?
How to make sure the compiler-linker is using the files on my folders and not other version of the library ?

Use/modify the following lines in the `platformio.ini` files:
```
[platformio]
lib_extra_dirs =
src_dir =
```

for example examples/IRrecvDumpV2/platformio.ini:
```
[platformio]
lib_extra_dirs = ../../
src_dir = .
```
Tells PlatformIO to search for the libraries it needs (e.g. IRremoteESP8266) in the [lib_extra_dirs](http://docs.platformio.org/en/latest/projectconf/section_platformio.html#lib-extra-dirs) "../../" directory which is two directories up from where the file is. ".." means the parent directory.
It also looks in the [src_dir](http://docs.platformio.org/en/latest/projectconf/section_platformio.html#src-dir) of "." as well. i.e. where is the source code for this project. "." means the _current directory_ the _platformio.ini_ file is in. Effectively [src_dir](http://docs.platformio.org/en/latest/projectconf/section_platformio.html#src-dir) is what tells platformio which program to compile into a firmware etc.

crankyoldgit, 2018-01-30:
re:
> src/mpx_decode.cpp:66:76: error: no 'bool IRrecv::decodeMPX(decode_results*, uint16_t, bool)' member function declared in class

1) Upload (git push) your code to your repository so I can see. ;-)
2) Taking a *guess*, because I can't see the code ... If you followed what I had in the patch, there is a `#ifdef DECODE_MPX` surrounding the line in `IRrecv.h`. Check that `#define DECODE_MPX true` is set in `IRremoteESP8266.h`. You previously had "MPX_DECODE".

crankyoldgit, 2018-01-30:
Apologies. You have updated github. My bad.

You didn't include the changes from the patch file for `IRrecv.h`. Thus the `decodeMPX()` function in the class hasn't been defined.
e.g. There is no `bool decodeMPX(decode_results *results, uint16_t nbits = MPX_BITS, bool strict = true);` line from what I can see in your github repository.

gnkarn, 2018-01-30:
getting better , but not there yet 

this is the error i have now 

test/IRsend_test.cpp:5:25: fatal error: gtest/gtest.h: No such file or directory
should i disable all test files ?

gnkarn, 2018-01-30:
yes, i was working with two file sets at the same time, one for the example only, and one with the complete library , so some of your finding were included on the other ...
never mind , now everything is on my library fork , but still getting some errors ,

crankyoldgit, 2018-01-30:
re:
> gtest/gtest.h: No such file or directory

Try:
```
cd test
make install-googletest
```

You only need to worry about _gtest_ etc if you want to run the unit test suite under `test/`.
I assume you got the error trying to compile/run the unit tests.

If you are getting it in "PlatformIO" when editing/saving the file(s) it's safe to ignore. I can't seem work out how to get it to ignore/fix that issue. It doesn't affect the unit tests or the rest of the library/code etc. It's a PlatformIO-only issue as far as I can tell.

As for your `src/mpx_decode.cpp:66:76: error: no 'bool IRrecv::decodeMPX(decode_results*, uint16_t, bool)' member function declared in class` error. I've sent two obvious PRs for your _mpx_testing_ branch which should fix some of your problems. I didn't try them, but they are part of what is wrong.

gnkarn, 2018-01-30:
Yes I’m using platformio, I like it but is complicated to me understand what is doing some times .

I have not used test , and I have not selected the option in platformio , those files were just included on the compile list , but no idea why .

And for the error , yes I have included all fixes , but still same error , for sure I’m still missing something , probably send mpx but the error is on the recev...
Thanks

crankyoldgit, 2018-01-30:
I hear you, platformio is very complicated. I only understand about 10% of it, and have to google a lot to work it out. I find it a lot more powerful than the Arduino IDE, and makes life easier maintaining this large library.

I too, have no idea why it (platformio) is trying to compile the unit test files. I've never tried to use them with platformio. I just edit them with platformio. In theory it's possible etc.

As for your error(s), I re-cloned and put the latest _mpx_testing_ branch into my platformio config, and here are the minimum required changes I needed to do to get it to successfully compile and link a firmware for IRrecvDumpV2.ino

[patch.txt](https://github.com/markszabo/IRremoteESP8266/files/1677973/patch.txt)

```
diff --git a/platformio.ini b/platformio.ini
index 704268e..5878568 100755
--- a/platformio.ini
+++ b/platformio.ini
@@ -1,6 +1,6 @@
 [platformio]
 lib_extra_dirs = .
-src_dir = .
+src_dir = examples/IRrecvDumpV2
 
 [common]
 build_flags =
diff --git a/src/IRrecv.cpp b/src/IRrecv.cpp
index d0c64d9..429f6cb 100755
--- a/src/IRrecv.cpp
+++ b/src/IRrecv.cpp
@@ -271,7 +271,7 @@ bool IRrecv::decode(decode_results *results, irparams_t *save) {
   #if DECODE_MPX
       DPRINTLN("Attempting MPX decode");
       // Try decodeMPX
-      if (decodeMpx(results,MPX_BITS, false))
+      if (decodeMPX(results,MPX_BITS, false))
         return true;
     #endif
 
diff --git a/src/mpx_decode.cpp b/src/mpx_decode.cpp
index e453779..1fed2fe 100644
--- a/src/mpx_decode.cpp
+++ b/src/mpx_decode.cpp
@@ -14,7 +14,7 @@
 // Ref:
 //   https://github.com/markszabo/IRremoteESP8266/issues/309
 #define MPX_TICK             100U
-#define _HDR_MARK_TICKS     40U
+#define MPX_HDR_MARK_TICKS     40U
 #define MPX_HDR_MARK         (MPX_HDR_MARK_TICKS * MPX_TICK)
 #define MPX_HDR_SPACE_TICKS    14U
 #define MPX_HDR_SPACE        (MPX_HDR_SPACE_TICKS * MPX_TICK)
```

gnkarn, 2018-01-30:
i can't make it to compile, 
still with this error , and with no ideas on what else to try ...
-----------------

> src/IRrecv.cpp: In member function 'bool IRrecv::decode(decode_results*, irparams_t*)':
src/IRrecv.cpp:274:44: error: 'decodeMpx' was not declared in this scope
if (decodeMpx(results,MPX_BITS, false))
^
*** [.pioenvs/nodemcuv2/lib851/src/IRrecv.o] Error 1

crankyoldgit, 2018-01-30:
I've said this multiple times previously in the patches I've suggested.

You need to change the line it is referring to from "decodeMpx" to
"decodeMPX"

gnkarn, 2018-01-30:
men, i will need to change my glasses , or pay more attention , haven't realized the caps difference ,
thanks you !

gnkarn, 2018-01-31:
DAvid , it is working now .
i have to adjust and and fine tune parameters to be able to decode the protocol correctly .

first question , is about header : In this case i don't have a defined "header" , it is just a long mark between each data , as the data begins on the first high to low edge , then this first space ( 1400 us) detection is what indicates the "frame" beginning .
Should i define ? : 

> #define MPX_TICK             100U
#define MPX_HDR_MARK_TICKS     0U
#define MPX_HDR_MARK         (MPX_HDR_MARK_TICKS * MPX_TICK)
#define MPX_HDR_SPACE_TICKS    14U
#define MPX_HDR_SPACE        (MPX_HDR_SPACE_TICKS * MPX_TICK)

gnkarn, 2018-01-31:
this is what I'm getting as raw timing so far
`Raw Timing[63]:
   +  4672, -  1578,    +    10, -    10,    +  1430, -  3120,    +    12, -  1436,
   +  3114, -    14,    +  2964, -  1592,    +    14, -    14,    +  2944, -  1592,
   +    12, -    12,    +  1428, -  3108,    +    12, -    10,    +  1442, -  3114,
   +    12, -  1438,    +  3114, -    10,    +    12, -  1438,    +  3118, -    14,
   +    14, -  1422,    +  3106, -    14,    +    14, -  1422,    +  3108, -    12,
   +  1440, -  3110,    +    12, -  1438,    +  3110, -    10,    +    10, -  1432,
   +  3102, -    12,    +    10, -    10,    +  1416, -  3112,    +    12, -  2954,
   +  1582, -    10,    +  2984, -  1588,    +    10, -    10,    +   540`

i think i need to fix those , very short pulses  ( two digits) , and it will detect the 16 bits ( 32 timings)  the rest looks fine  ..

gnkarn, 2018-01-31:
i think the problem may be on the m_tick calculation  as my header is not the way the original decode is looking for, will try making m_tick = s_tick ..

gnkarn, 2018-01-31:
is not detecting the signal in compliance with the MPX decode definition , so i may have something wrong there ..

crankyoldgit, 2018-01-31:
That raw data doesn't seem to marry up to what you showed on your oscilloscope earlier. i.e. There appears to be 17 pulses on that image, which should equate to ~33 entries in a raw capture.
I've not yet experienced IR pulses from an IR module as short as 10-14 usecs that were part of a "valid" IR protocol. They are just too short. That's the sort of time frame/pulse size of the modulation _within_ a "mark" signal. e.g. during a "mark" signal, the actual LED is toggled on/off at the modulation freq. with a period of around 10-24usecs.

Just to be sure, if you are capturing that "raw" signal with a clean copy of the library (or an arduino etc) then I'd *strongly* suspect that the modulation frequency the remote is transmitting at is incompatible with the modulation freq of the IR detector module.
For example, the IR remote may be transmitting at 36kHz, but you have a 38kHz IR module connected to the ESP.

In short, in my experience, if a raw capture has values in the teens, then it's a bad capture. Something is wrong. If it happens frequently (i.e. in most captures etc) *and* you have the remote about 1+ meter away, it's probably a frequency mis-match problem. Try a different frequency IR module. As you have access to an oscilloscope, you should be able to work out what that freq is if you measure the voltage/current etc across the transmitting IR LED in the remote. i.e. Calculate the period of each modulation during a "mark" signal.

gnkarn, 2018-01-31:
David, probably i wasn't clear enough on my initial post , the signal i posted "is not" the result of any IR protocol , it is just plain direct electrical signal going to  the Node mcc Pin D2 .

The shape of the signal is exactly what i showed , and yes the results I'm getting are far from being equal to the the received signal .

so those sort pulses do not exists on the signal, they are introduced by the decoder , but at present i was unable for the MPX decode to recognize the signal, so it is being decoded as the "unknown" , and  then it is very different from the real one .

i need to find the proper combination of parameters , for the irRecv recognize the protocol ( my signal has no header , and the footer is just a regular space )   ... 
Still trying to understand  the code and how to do it ,.

makes sense?
if you have any suggestion on how to find tune parameters , to make it decode the signal  that   stays high between frames , then changes to low to start the frame, and has 16 consecutive bits , where one is represented by 2800 us pulse followed by a 1400us space , and a zero is 1400 us high pulse followed by a 2800us space , then the total time for a single bit is always ( almost ) the same = 4200us
the end of the frame is again  the long high between frames .

gnkarn, 2018-01-31:
I'm trying to make this parallel : 
in this protocol case , mark and space are same duration ( 1400 us)
a one is represented by a 2 mark time follow by one space time
a zero is the opposite a 1 mark time followed by a 2 space time .

crankyoldgit, 2018-02-01:
I get what you are saying about how the protocol works. I've seen similar protocols. I am however, confused in a lot of other areas with what you are saying. To get myself on the same page as toy, I have a bunch of questions I'd like you to answer if you can. I'll explain why I'm asking after them. If you could answer them all, that'd be great.

1) What is the part number/model number of the IR module you have connected to the ESP8266?
2) What is the circuit you are using to read from that part?
3) Are you able to receive/decode signals from other remote controls etc correctly with this setup?
4) Can you list the "raw" output from a clean copy of IRrecvDumpV2 from a couple of messages? (i.e. everything it lists)

Looking at your oscilloscope output **and** your description, it seems that what ever circuit you are using, it sounds like the module/connection to the ESP's GPIO is being dragged LOW when it's detecting a pulse (mark) and HIGH when its detecting nothing (space/gap/idle). If that is the case, the library is not designed to really operate under that condition. It's expecting the opposite. It may work. It may not, I don't know. I haven't tested it in that situation. I _think_ it could be part of your problem. Every IR receiver I've come across so far creates a HIGH signal when it detects something, and LOW the rest of the time. If I understand what you are saying, your circuit/module seems to be behaving opposite to what's expected.

I'd also like to eliminate any of your changes/tinkering with the library to make sure, mainly due to your comments/feedback. The changes I think you've made should NOT change the "raw" output, but I just want to rule it out.

Now, answering some of your other question(s):

You don't need to match a header if there is none. There are other protocols that don't have a header, but it sounds like from what you described earlier it does have a header. I can't really comment until I can see a clean raw dump from something we can trust. If I can see a clean capture, I can tell with a lot more certainty what the protocol looks like from the library's perspective.

There are other _equal total data bit time_ protocols supported by the library, so what you are trying to do isn't something totally new.

You should be able to get rid of all the TICK stuff if you want, just use the totals. i.e. Don't use the _s_tick_ & _m_tick_ stuff. Just the final values, not the dynamic/calculated ones. e.g. just MPX_HDR_MARK, MPX_ZERO_SPACE, etc...

> so those sort pulses do not exists on the signal, they are introduced by the decoder 

Can you explain in gory detail what you mean here?
Which signal? (the one before it gets to the IR detector, the electrical one between the receiver module and the ESP, or the "digital" signal as seen by the library?)
Which decoder? (The IR receiver module i.e. electronics component, or your `decodeMPX()` routine?

gnkarn, 2018-02-01:
thank you for your comments .

will try to bring more detail : the signal do not come from IR , it comes directly from my alarm system digital data bus .

I'm quite close to have the decoder working  
this the actual output : 
`Attempting MPX decode
Matching MARK 1578 vs 8000 + 50. Matching: 6037 <= 1578 <= 10063
Timestamp : 001136.250
Encoding  : UNKNOWN
Code      : 4C18BBC (17 bits)
Library   : v2.3.2

Raw Timing[33]:
   +  1578, -  2986,    +  1596, -  1454,    +  3124, -  2990,    +  1600, -  2976,
   +  1596, -  1460,    +  3130, -  1444,    +  3126, -  1458,    +  3124, -  2978,
   +  1590, -  1452,    +  3118, -  1454,    +  3124, -  1448,    +  3126, -  1454,
   +  3124, -  2988,    +  1590, -  1464,    +  3128, -  1448,    +  3126, -  2974,
`
which is close to be correct ,  as it is not detecting correctly the header ( just a low space of 1400 us) , the frame is misaligned , . so will focus on that first header detection , before . 
I agree with what you are saying respect of this type of code being the opposite of a normal IR code , in this case it stays always ON , ant the beginning of the frame is a high to low transition , so i will focus on that part and see ..

gnkarn, 2018-02-01:
what haven't made myself clear is why I'm using  your library for this purpose ( non IR decoding)  ??
it is because is the best decoding library  for ESP i found to be flexible to be adapted  , and tested to be integrated with the tasmota code later ...

gnkarn, 2018-02-01:
this is a raw out , with my basic decoder , and the signal is ok here.

[image]

gnkarn, 2018-02-01:
the yellow are the raw timings for the 16 bits low and high states in between tho long high time ( just middle between frames) ,and the the last low space before idle

gnkarn, 2018-02-01:
I think i found the problem : 
on the function :  

match_result_t IRrecv::matchData(
it is expecting to be  either ,
if (onemark == zeromark) {  // Is this space encoded data format?   or

else if (onespace == zerospace) {  // Is this mark encoded data format?

and in this protocol non of them matches , it is the type of mark1+space1 = mark0+space0

i was expecting  to concentrate the peculiarities of the protocol inside decodeMpx , but it not be enough ...

gnkarn, 2018-02-01:
will include a special match data function inside decodeMpx

crankyoldgit, 2018-02-02:
@gnkarn I had forgotten about some optimisations I had put in `matchData()`. You are correct, they would not work for a fixed total time per bit protocol.
I've submitted a PR (#408) to make it even more generic, and it should handle your use case now.
If you can't work out how to merge in that commit in _git_ you could probably just replace the routine by hand by a cut & paste from https://github.com/markszabo/IRremoteESP8266/blob/equal_total_data_bit_time/src/IRrecv.cpp#L667

gnkarn, 2018-02-02:
thank you , i will try that !

gnkarn, 2018-02-02:
i understood your mod, and it looks good , better than mine .
but i was not able to make it work right .. yet.

my last modification  throws a decode , that is "almost right" , but instead of signing the 1 or zero , by looking at the mark , it is decoding by looking at the space ( even when in the code of course that is not the intention).

then the output , is for example 
`Raw Timing[33]:
   +  1580, -  2994,    +  1588, -  2992,    +  1594, -  1454,    +  3128, -  1462,

   +  3132, -  2982,    +  1586, -  2994,    +  1594, -  1460,    +  3124, -  1454,
   +  3126, -  1462,    +  3120, -  1464,    +  3124, -  1452,    +  3134, -  1448,
   +  3124, -  1458,    +  3124, -  1464,    +  3120, -  2986,    +  1598, -  2998,
   +  1598

uint16_t rawData[33] = {1580, 2994,  1588, 2992,  1594, 1454,  3128, 1462,  3132, 2982,  1586, 2994,  1594,
1460,  3124, 1454,  3126, 1462,  3120, 1464,  3124, 1452,  3134, 1448,  3124, 1458,  3124, 1464,  3120, 2986
,  1598, 2998,  1598};  // MPX 19FE
uint64_t data = 0x19FE;`

THE RawData is OK
but the 0x19FE  corresponds to a match with the space length instead of the mark length .

may be i have some "offset" problem on top of everything ?

i will look into it with more fresh head tomorrow , 1AM here now , tks

crankyoldgit, 2018-02-02:
I concur. The data looks like it shifted over one slot.

First comment out:
https://github.com/gnkarn/IRremoteESP8266-mpx/blob/mpx_testing/src/decodeMpx.cpp#L86
`uint32_t s_tick = results->rawbuf[offset++] * RAWTICK /  MPX_HDR_SPACE_TICKS;` as it is incrementing the value of `offset`.

Try that, first.

Then I'd suggest tweaking this line:
https://github.com/gnkarn/IRremoteESP8266-mpx/blob/master/src/mpx_decode.cpp
`uint16_t offset = OFFSET_START;`

OFFSET_START is usually 1. So try values of `0` or `2` instead. i.e. Change the value here, **don't** change what OFFSET_START is defined as.

You might want to comment out the `matchAtLeast()` line in your decoder too, till you have it sort of working. I'm assuming you've worked out how to set `DEBUG` on etc etc.

crankyoldgit, 2018-02-09:
Friendly ping. How are you going on this? Any progress?

gnkarn, 2018-02-09:
Very well , almost there , I already have a working version included on my tasmota fork , will add here the link later . Still need to complete some checks , and of course lots of cleanup and some doc .
Thank you for asking ¡

crankyoldgit, 2018-02-28:
Friendly ping.

gnkarn, 2018-03-01:
Have started to do some homework, 
Doc Is still in spanish , 
And code is under mpx branch , 

https://github.com/gnkarn/Sonoff-Tasmota/wiki/Organization-del-codigo-para-decodificador-MPX

Will get there , ASAP ( before I forget ...) ..

crankyoldgit, 2018-03-16:
How is it going?

crankyoldgit, 2018-05-04:
Marking this closed for now, due to inactivity. Feel free to reply/reopen etc.

closed 2018-05-04 by crankyoldgit