crankyoldgit/IRremoteESP8266 #411: Odd error when trying to run sample programs? I've tried various boards / latest libs

frownbreaker, 2018-02-05:
### Version/revision of the library used
v2.3.2 (also tried earlier versions eg 2.3.0)

### Expected behaviour
Tried to run the IRrecvDumpV2 program unchanged also IRrecvDemo
1. Compile and run the program with IR receiver connected
2. App gets flashed no errors reported 
3. Expect code to appear on serial output when IR remote is transmitting near receiver

### Actual behaviour
1. Compile and run the program with IR receiver connected (or not connected the same issue).
2. App gets flashed no errors reported and then output via serial looks like watchdog error
3. Blink and other apps (ones that use WiFi work fine) on the same setup. I've tried this on 2 different Wemos D1 minis and a Wemos D1 (large format like Uno) all are ESP8266 based boards that work fine with other apps

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
```
### Steps to reproduce the behaviour
Seems to be consistent. I complile other apps work fine. But the IRremoteESP8266 receiver demos all crash even if I change the board, USB cable, etc.. I've tried different ESP boards and USB cables. Deleted and reinstalled the libraries, flash wiped the ESPs etc. Blink and other apps compile, flash and run without issue. Problem is SAME even when no IR receiver connected (e.g. bare board)

#### Example code used
IRrecvDumpV2 or IRrecvDemo unchanged target board Wemos D1 r2 & Mini

#### Circuit diagram and hardware used (if applicable)
Stock setup as shown on this GitHub. Check the pinout Vs datsheet problem happens even with reciver not connect to board: Bare board.

### I have followed the steps in the [Troubleshooting Guide] Yes
Read the [FAQ] Arduino core for ESP8266 latest version, which is currently 2.4.0. Confirmed

### Other useful information
I've been using the ESPbase boards for a while no issues with my PC, and setup. As I can build and flash all apps except the IRremoteESP8266 perhaps there are some IR libraries I need to delete reinstall? I uninstalled all the other IR libraries that I had installed e.g. _IRremote by shirriff_. I've restart the IDE etc but not reinstalled the Arduino IDE. Windows 8.1 PC 64 bit patched to current update level

frownbreaker, 2018-02-05:
Renamed my user\documents\arduino folder to backup
Uninstalled Arduino IDE
Downloaded IDE
Reinstalled  Arduino IDE 
Compiled demo no errors :)

frownbreaker, 2018-02-05:
Reinstalled the IDE just produces a serial window with no errors but pressing reset on the board gets the same error.

frownbreaker, 2018-02-05:
My bad the program output appears when Ir codes are decoded so no output. 
Once I used the documented baud rate 115200 and provided an IR source the codes appeared.
My bad! Perhaps the demo program could display a line of text like "IR Receive demo running waiting for IR code? Just for daft people like me!

crankyoldgit, 2018-02-05:
Thanks for the detailed report, and for getting back to us that it wasn't a "real" issue per se.

Your suggestion sounds like a reasonable thing to add. I'll do so.

closed 2018-02-06 by crankyoldgit

crankyoldgit, 2018-03-02:
This is now live in the v2.3.3 release

frownbreaker, 2018-03-02:
Many thanks I'm sure it will help the intellectually undercapitalised like myself. Seems like a silly mistake I made but I'd used the library before, so I'm sure the change will help, thanks again.

PR #412 (2018-02-06): Add startup text to IRrecvDumpV2 and IRrecvDemo
* Add a simple text message to let the user know the code is running.
* Add boilerplate code to wait for serial connection to be establised.

Fixes #411
commit 8bebeeb2 (2018-02-06): Add startup text to IRrecvDumpV2 and IRrecvDemo (#412)

* Add a simple text message to let the user know the code is running.
* Add boilerplate code to wait for serial connection to be establised.

Fixes #411