crankyoldgit/IRremoteESP8266 #413: Question: how to repeat data from un unsupported SANYO Aircon remote

dolfandringa, 2018-02-09:
I am trying to turn a SANYO aircon that I control with a universal remote on an off through an esp8266. So far without luck. I have trouble understanding the library and am going through the source code to try to understand what's happening. I'd be happy to contribute docs back when I do understand it better with some help (hopefully).

When I take the example from the most recent version of the library
ircv.decode(&results) returns true, and I get results.value=0xf6f2640b when I press the remote's on/off button. Depending on the settings of the aircon (temp and such) the code is consistent (which makes sense since I understand AC remotes send all the settings at once, not just an on/off command). But when I irsend.sendSanyo(0xf6f2640b) the aircon doesn't turn off. 
Also, if I check results.decode_type (after irrecv.decode(&results)), it's value is -1 (UNKOWN) and the command and address are 0. So although my aircon is a Sanyo, but apparently different from the SanyoLC7461 protocol so the library is unable to decode the data. 
So I end up with basically three questions:

1. Where can I find more docs on the process of receiving and decoding an ir signal. Specifically how results.value, results.command and results.address relate to each other and what commands and addresses are in this case.
2. Is results.value the raw value that was received by the IR receiver? Or is that value already a derivative of the raw value that is the result of some kind of decoding?
3. If results.value indeed is the raw value, should irsed.sendRaw(results.value 32, 38), where results.value=0xf6f2640b and was received by pressing the button on the actual remote, result in the aircon doing exactly the same as what would happen if I press the button on the remote directly?

For now, just being able to turn the aircon on and off, even if that results in it always being on the same settings, would be enough. Figuring out more of the protocol can wait until later.

crankyoldgit, 2018-02-09:
First off, please read the FAQ. Most of your question(s) are answered there. e.g.
https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code

Re: Q1
If and only if `decode()` successfully identifies the protocol (i.e. not UNKNOWN) the `values`, `command`, `address`, & `state` will be set accordingly to something useful for the given protocol the message matched.

Re: Q2
No. Yes, it is a derivative.

Re: Q3
No, it's not. Per the FAQ, use the `IRrecvDumpV2` example code to get the `rawData` you're looking for. There is example code for how to use `sendRaw()`.

dolfandringa, 2018-02-09:
Thanks @crankyoldgit. I missed that part of the FAQ indeed. Sorry. I never understood the rawData in the IRsenddemo, but now finally understand that the resultToSourceCode will output the raw data. I was searching forever to find where I get the raw data and couldn't find it. I missed the IRUtils module. Thanks a lot for the help. I'll be checking it out later.

scropion86, 2018-02-27:
is firing the IR signal to an IR receiving circuit and in same time collecting the Data using Logic analyzer
will be helpful.
maybe message length and and frequency will easier or just add more complication to already complected equation. @crankyoldgit

crankyoldgit, 2018-02-27:
Attaching a IR receiver module to a Oscilloscope or Logic Analyser will give you the various "mark"s & "space"s timings for the given protocol and eliminate any potential issues with the ESP8266 and/or library. So, in almost all cases, it's better information than from the ESP8266 itself.

Note however that the results of what you described will NOT indicate the modulation frequency of the IR signal, not the carrier frequency. Just how long each "mark" (pin high) and each "space" (pin low) is. To get those, you'd need a "dumb" IR detecting diode, not an IR module, and you'd have to do a lot more measurements/calculations etc.

The advantage of using the library/esp to do it is it will break it down easily and give you individual timings for each mark/space without requiring the extra hardware, and saving you having to do all the measurements/calculations on the oscilloscope/logic analyser.

crankyoldgit, 2018-03-16:
@dolfandringa How is it going? Is this open issue still valid?

dolfandringa, 2018-04-15:
Sorry, this went very well. I should close it. Thanks for the help.

closed 2018-04-15 by dolfandringa