crankyoldgit/IRremoteESP8266 #414: Trying to power on ASYG24LGCC (Fujitsu) not working

arpmota, 2018-02-16:
Hello,
I'm trying to power on ASYG24LGCC (Fujitsu indoor unit) not working with example "TurnOnFujitsuAC.ino".

Also tryied receiving the code with "IRrecvDumpV2.ino" from the fujitsu remote control and the replicate using "IRsendDemo.ino" and the result is the same nothing happen to the AC unit.

I'm using a ESP8266 (WROOM-02) and I know that I can send IR code because I receive using another board. It acknowledges with the code below.

Encoding  : FUJITSU_AC
Code      : 1463001010FE0930810400000000202B (128 bits)
Mesg Desc.: Power: On, Mode: 4 (HEAT), Temp: 24C, Fan: 0 (AUTO), Swing: Off, Command: N/A
Library   : v2.3.2

Raw Timing[259]:
   +  3352, -  1574,    +   480, -   350,    +   480, -   346,    +   480, -  1190, 
   +   458, -   346,    +   508, -  1140,    +   480, -   346,    +   506, -   346, 
   +   458, -   346,    +   480, -  1168,    +   480, -  1192,    +   452, -   374, 
   +   458, -   346,    +   480, -   346,    +   508, -  1168,    +   480, -  1140, 
   +   480, -   346,    +   506, -   346,    +   458, -   346,    +   480, -   346, 
   +   480, -   346,    +   480, -   346,    +   484, -   372,    +   454, -   374, 
   +   456, -   346,    +   508, -   318,    +   480, -   374,    +   458, -   374, 
   +   480, -   318,    +   480, -  1196,    +   452, -   346,    +   480, -   346, 
   +   484, -   342,    +   484, -   346,    +   480, -   374,    +   458, -   346, 
   +   506, -   318,    +   508, -  1170,    +   452, -   346,    +   480, -   374, 
   +   458, -   346,    +   506, -   318,    +   480, -  1196,    +   452, -  1190, 
   +   458, -  1162,    +   480, -  1196,    +   452, -  1170,    +   480, -  1190, 
   +   458, -  1164,    +   480, -  1196,    +   480, -   318,    +   508, -   346, 
   +   456, -  1192,    +   480, -   346,    +   456, -   374,    +   452, -   346, 
   +   480, -   374,    +   458, -   342,    +   484, -   346,    +   508, -   346, 
   +   456, -   342,    +   512, -  1164,    +   458, -  1164,    +   508, -   346, 
   +   456, -   346,    +   480, -  1190,    +   456, -   342,    +   484, -   346, 
   +   506, -   346,    +   456, -   374,    +   452, -   346,    +   508, -   346, 
   +   458, -  1164,    +   508, -   346,    +   458, -   374,    +   452, -  1168, 
   +   480, -   374,    +   480, -   318,    +   480, -   374,    +   456, -   346, 
   +   508, -   318,    +   480, -   346,    +   484, -   374,    +   480, -   318, 
   +   484, -   342,    +   484, -   374,    +   480, -   318,    +   484, -   342, 
   +   484, -   346,    +   508, -   318,    +   508, -   346,    +   458, -   346, 
   +   506, -   318,    +   480, -   374,    +   458, -   346,    +   506, -   318, 
   +   480, -   346,    +   484, -   374,    +   480, -   318,    +   482, -   372, 
   +   456, -   346,    +   508, -   318,    +   506, -   348,    +   456, -   342, 
   +   484, -   346,    +   508, -   318,    +   484, -   374,    +   480, -   318, 
   +   508, -   318,    +   484, -   346,    +   508, -   318,    +   480, -   374, 
   +   456, -   346,    +   508, -   346,    +   480, -   318,    +   480, -   346, 
   +   484, -   374,    +   480, -   320,    +   484, -  1164,    +   508, -   346, 
   +   458, -   342,    +   512, -  1164,    +   458, -  1190,    +   454, -   346, 
   +   484, -  1164,    +   508, -   346,    +   458, -  1164,    +   480, -   350, 
   +   480, -   374,    +   480

uint16_t rawData[259] = {3352, 1574,  480, 350,  480, 346,  480, 1190,  458, 346,  508, 1140,  480, 346,  506, 346,  458, 346,  480, 1168,  480, 1192,  452, 374,  458, 346,  480, 346,  508, 1168,  480, 1140,  480, 346,  506, 346,  458, 346,  480, 346,  480, 346,  480, 346,  484, 372,  454, 374,  456, 346,  508, 318,  480, 374,  458, 374,  480, 318,  480, 1196,  452, 346,  480, 346,  484, 342,  484, 346,  480, 374,  458, 346,  506, 318,  508, 1170,  452, 346,  480, 374,  458, 346,  506, 318,  480, 1196,  452, 1190,  458, 1162,  480, 1196,  452, 1170,  480, 1190,  458, 1164,  480, 1196,  480, 318,  508, 346,  456, 1192,  480, 346,  456, 374,  452, 346,  480, 374,  458, 342,  484, 346,  508, 346,  456, 342,  512, 1164,  458, 1164,  508, 346,  456, 346,  480, 1190,  456, 342,  484, 346,  506, 346,  456, 374,  452, 346,  508, 346,  458, 1164,  508, 346,  458, 374,  452, 1168,  480, 374,  480, 318,  480, 374,  456, 346,  508, 318,  480, 346,  484, 374,  480, 318,  484, 342,  484, 374,  480, 318,  484, 342,  484, 346,  508, 318,  508, 346,  458, 346,  506, 318,  480, 374,  458, 346,  506, 318,  480, 346,  484, 374,  480, 318,  482, 372,  456, 346,  508, 318,  506, 348,  456, 342,  484, 346,  508, 318,  484, 374,  480, 318,  508, 318,  484, 346,  508, 318,  480, 374,  456, 346,  508, 346,  480, 318,  480, 346,  484, 374,  480, 320,  484, 1164,  508, 346,  458, 342,  512, 1164,  458, 1190,  454, 346,  484, 1164,  508, 346,  458, 1164,  480, 350,  480, 374,  480};  // FUJITSU_AC
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0x81, 0x04, 0x00, 0x00, 0x00, 0x00, 0x20, 0x2B};

If I use the remote control of the unit I receive the same code. But by some reason the device doesn't operate. 

Can someone help me sort this out?

Thank you very much.

crankyoldgit, 2018-02-16:
Can you please show us the code you are using to send with?
Are you able to control other IR device with your setup? (Note: That's different to another arduino etc seeing the signal. e.g. Signal strength for a real device)

Have you tried using `sendRaw()` instead after you've captured the code via IRrecvDumpV2?

crankyoldgit, 2018-02-16:
Also, note that there are two differing types of Fujitsu remotes supported by this library.
See: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Fujitsu.cpp#L61 & https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Fujitsu.h#L48
The library defaults to ARRAH2E, so you might want to try ARDB1.

I believe the ARDB1 codes are decoded as 120 bit messages, and the ARRAH2E as 128 bit.
So, it's vital you give us a dump from the real remote to help us aid you.

arpmota, 2018-02-16:
The code is the one for the example. Another try is also sending with sendraw
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

IRsend irsend(4);  // An IR LED is controlled by GPIO pin 4 (D2)

// Example of data captured by IRrecvDumpV2.ino
uint16_t rawData[115] = {3372, 1552,  502, 324,  480, 352,  502, 1140,  502, 324,  508, 1140,  480, 346,  508, 324,  502, 324,  480, 1168,  502, 1146,  502, 324,  502, 324,  502, 324,  508, 1140,  480, 1168,  502, 324,  508, 322,  502, 324,  502, 324,  506, 324,  502, 324,  502, 324,  506, 320,  506, 324,  502, 324,  480, 350,  502, 324,  502, 324,  506, 1142,  480, 346,  480, 350,  502, 324,  502, 322,  508, 322,  502, 324,  506, 318,  508, 1142,  506, 324,  502, 324,  502, 324,  506, 318,  508, 1142,  480, 346,  508, 324,  502, 324,  480, 350,  502, 324,  502, 324,  508, 1140,  502, 324,  508, 1140,  480, 1164,  506, 1142,  506, 1142,  502, 1146,  502, 1142,  506};

void setup() {
  irsend.begin();
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {
  irsend.sendRaw(rawData, 115, 38);  // Send a raw data capture at 38kHz.
  delay(2000);
}

The dump that I sent is from the real remote controller.

I can send sony codes with this same device. 
So hardware is proven to work, and the example works but with different "codes".

It should work because I'm sending the same raw content that I receive from the real remote controller.
I can't really understand what could cause this?

crankyoldgit, 2018-02-16:
As far as I can tell, the data you provided looks good and is as expected. (I assume you set the remote to the same as in the example code?!)

Based on your capture data, it seems some of the timing values may need to be tweaked for your unit.
e.g. Try changing the following defines:
```
#define FUJITSU_AC_HDR_MARK    3352U
#define FUJITSU_AC_HDR_SPACE   1574U
#define FUJITSU_AC_BIT_MARK    480U // or 450
#define FUJITSU_AC_ONE_SPACE   1160U
#define FUJITSU_AC_ZERO_SPACE  346U  // Or 320
```

Let us know if that works, or if you other values worked for you.

FYI, to send the raw code, you need to use something like:
```
// Capture as supplied by arpmota
  uint16_t rawData[259] = {3352, 1574, 480, 350, 480, 346, 480, 1190, 458, 346,
      508, 1140, 480, 346, 506, 346, 458, 346, 480, 1168, 480, 1192, 452, 374,
      458, 346, 480, 346, 508, 1168, 480, 1140, 480, 346, 506, 346, 458, 346,
      480, 346, 480, 346, 480, 346, 484, 372, 454, 374, 456, 346, 508, 318,
      480, 374, 458, 374, 480, 318, 480, 1196, 452, 346, 480, 346, 484, 342,
      484, 346, 480, 374, 458, 346, 506, 318, 508, 1170, 452, 346, 480, 374,
      458, 346, 506, 318, 480, 1196, 452, 1190, 458, 1162, 480, 1196, 452,
      1170, 480, 1190, 458, 1164, 480, 1196, 480, 318, 508, 346, 456, 1192,
      480, 346, 456, 374, 452, 346, 480, 374, 458, 342, 484, 346, 508, 346,
      456, 342, 512, 1164, 458, 1164, 508, 346, 456, 346, 480, 1190, 456, 342,
      484, 346, 506, 346, 456, 374, 452, 346, 508, 346, 458, 1164, 508, 346,
      458, 374, 452, 1168, 480, 374, 480, 318, 480, 374, 456, 346, 508, 318,
      480, 346, 484, 374, 480, 318, 484, 342, 484, 374, 480, 318, 484, 342,
      484, 346, 508, 318, 508, 346, 458, 346, 506, 318, 480, 374, 458, 346,
      506, 318, 480, 346, 484, 374, 480, 318, 482, 372, 456, 346, 508, 318,
      506, 348, 456, 342, 484, 346, 508, 318, 484, 374, 480, 318, 508, 318,
      484, 346, 508, 318, 480, 374, 456, 346, 508, 346, 480, 318, 480, 346,
      484, 374, 480, 320, 484, 1164, 508, 346, 458, 342, 512, 1164, 458, 1190,
      454, 346, 484, 1164, 508, 346, 458, 1164, 480, 350, 480, 374, 480};
  irsend.sendRaw(rawData, 259, 38000);
```
or
```
uint8_t state[16] = {
      0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0x81, 0x04, 0x00, 0x00,
      0x00, 0x00, 0x20, 0x2B};
irsend.sendFujitsuAC(state, 16);
```

arpmota, 2018-02-16:
I changed the definitions and they solved the issue.

Just de HDR MARK definition solved it.
#define FUJITSU_AC_HDR_MARK    3352U
//#define FUJITSU_AC_HDR_SPACE   1574U
//#define FUJITSU_AC_BIT_MARK    480U // or 450
//#define FUJITSU_AC_ONE_SPACE   1160U
//#define FUJITSU_AC_ZERO_SPACE  346U  // Or 320

using   irsend.sendRaw(rawData, 259, 38000);

Thank you very much.

closed 2018-02-16 by arpmota

crankyoldgit, 2018-03-02:
This is now live in the v2.3.3 release