crankyoldgit/IRremoteESP8266 #415: Garbege output

B0R1K, 2018-02-16:
Hi Team

I am using IRrecvDumpV2 0.3 and when i press any button on Samsung Remote I get following:

It seems like garbage to me, or am i doing something wrong.

⸮⸮U*U(⸮⸮⸮⸮⸮
⸮Y⸮*U(⸮⸮⸮(⸮
⸮⸮U*U(⸮⸮⸮⸮⸮
⸮Y⸮*U(⸮⸮⸮⸮⸮
⸮⸮⸮*U(⸮⸮⸮(⸮U(⸮⸮⸮⸮B⸮U⸮*⸮⸮U(⸮⸮⸮U(⸮⸮⸮⸮⸮
⸮⸮⸮*U(⸮⸮⸮⸮⸮
⸮Ye*U(⸮⸮⸮⸮⸮
⸮⸮UB*⸮⸮⸮*⸮⸮U(⸮Q⸮⸮B(⸮⸮*⸮⸮
⸮Y⸮B*⸮UU⸮⸮⸮
⸮⸮⸮*U(UU⸮⸮⸮U(⸮U⸮⸮⸮⸮⸮*(⸮
etc...

Thank you

crankyoldgit, 2018-02-16:
Use the correct baud rate. I.e. 115200

B0R1K, 2018-02-16:
i am using it, just tried remote from Fujitsu same thing,..

[image]

crankyoldgit, 2018-02-16:
Try reflashing the example code, and check what does is set in the code.

B0R1K, 2018-02-16:
Not sure what Refreshing example code means, 
but i see this line #define BAUD_RATE 115200

darshkpatel, 2018-02-16:
Can you try uploading the code again ?
Maybe try it with another cable ?

crankyoldgit, 2018-02-16:
Reflashing means, recompile and upload the example code again. Try powering
off and on the esp8266. And restart the Arduino IDE on your computer. Maybe
unplug the USB cable too

B0R1K, 2018-02-16:
Just tried all of these
New Cable, new Upload, restarted, but same gibberish...

Thank you for taking time to help out.

crankyoldgit, 2018-02-16:
It works for pretty much everyone else, so that leaves something that is odd with your computer. Try it on a different PC perhaps? or reboot the PC? Or try changing the BAUD rate to 9600 on both the terminal program and the code and re-upload.

Are you able to talk at that BAUD rate with any other arduino sketches on this ESP8266 or other devices?

crankyoldgit, 2018-02-16:
or on a different ESP8266 module.

B0R1K, 2018-02-17:
ok i got it to work
i had to compile it with BAUD_RATE 74880
and run Serial Monitor at 115200

B0R1K, 2018-02-17:
but i Get 
Encoding  : UNKNOWN for Samsung Remote, please advise...

crankyoldgit, 2018-02-17:
I suspect something is odd with your esp8266 module

crankyoldgit, 2018-02-18:
WIthout the full data from the capture from IRrecvDumpV2, we can't help you further.
Again, the changes you've had to do to get it to work, I seriously think something is wrong with your ESP8266 module, or your serial/usb setup.

B0R1K, 2018-02-19:
i am going from USB Type C to micro-B, Do u think that might have something to do with it?

crankyoldgit, 2018-02-19:
Not sure. It might affect the baud rate issue. ie. usb drivers etc. It should _not_ affect the ESP's ability to detect IR codes.

B0R1K, 2018-02-19:
I will try doing it on Different PC, 
but in mean type is there a specific ESP Brand/model you would recommend?

crankyoldgit, 2018-02-19:
Doing some research, it looks like you have a "bad" esp8266 board.
e.g. Read all of this issue thread https://github.com/esp8266/Arduino/issues/4005
I saw your other issue here https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/148 where you specify the board etc. that eventually lead me to the above thread.

The TL;DR: is if you have to use a baud rate of 74880 then it's a bad module. The clock/crystal on it is wrong and it won't work for wifi etc. All the timings are off.

I don't want to recommend specific vendors etc, but I've had great luck with purchasing NodeMCU and Wemos D1 mini modules via Aliexpress or ebay.
The official wemos store is https://www.aliexpress.com/store/1331105 (via http://www.wemos.cc/)

crankyoldgit, 2018-02-28:
Closing this issue. Feel free to re-open the issue if you "good" ESP module produces similar text.

closed 2018-02-28 by crankyoldgit

B0R1K, 2018-03-06:
Thank you for your help, Just want to update, that after receiving new ESP, 
everything seems to be working... Will be playing with it for next few days...