crankyoldgit/IRremoteESP8266 #416: Multiple compile errors with example IRsendGCDemo

s0170071, 2018-02-17:
Just tried to run the IRsendGCDemo with Arduino IDE 1.8.5 / ESP core version 2.4.0

```
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:0:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:24:0: warning: "MARK_EXCESS" redefined [enabled by default]
 #define MARK_EXCESS   50U
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:35:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteInt.h:216:0: note: this is the location of the previous definition
 #define MARK_EXCESS 100
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:0:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:25:0: warning: "RAWBUF" redefined [enabled by default]
 #define RAWBUF       100U  // Default length of raw capture buffer
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:35:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteInt.h:229:0: note: this is the location of the previous definition
 #define RAWBUF 100 // Length of raw duration buffer
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:0:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:26:0: warning: "REPEAT" redefined [enabled by default]
 #define REPEAT UINT64_MAX
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:93:0: note: this is the location of the previous definition
 #define REPEAT 0xffffffff
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:0:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:29:0: warning: "STATE_IDLE" redefined [enabled by default]
 #define STATE_IDLE     2U
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:35:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteInt.h:224:0: note: this is the location of the previous definition
 #define STATE_IDLE     2
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:0:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:30:0: warning: "STATE_MARK" redefined [enabled by default]
 #define STATE_MARK     3U
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:35:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteInt.h:225:0: note: this is the location of the previous definition
 #define STATE_MARK     3
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:0:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:31:0: warning: "STATE_SPACE" redefined [enabled by default]
 #define STATE_SPACE    4U
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:35:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteInt.h:226:0: note: this is the location of the previous definition
 #define STATE_SPACE    4
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:0:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:32:0: warning: "STATE_STOP" redefined [enabled by default]
 #define STATE_STOP     5U
 ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:35:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteInt.h:227:0: note: this is the location of the previous definition
[... 132 lines cut ...]
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:146:8: error: candidate is: bool IRrecv::matchMark(uint32_t, uint32_t, uint8_t, int)
   bool matchMark(uint32_t measured_ticks, uint32_t desired_us,
        ^
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:592:6: error: prototype for 'bool IRrecv::matchSpace(uint32_t, uint32_t, uint8_t, int16_t)' does not match any in class 'IRrecv'
 bool IRrecv::matchSpace(uint32_t measured, uint32_t desired,
      ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:148:8: error: candidate is: bool IRrecv::matchSpace(uint32_t, uint32_t, uint8_t, int)
   bool matchSpace(uint32_t measured_ticks, uint32_t desired_us,
        ^
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:621:9: error: prototype for 'int16_t IRrecv::compare(uint16_t, uint16_t)' does not match any in class 'IRrecv'
 int16_t IRrecv::compare(uint16_t oldval, uint16_t newval) {
         ^
In file included from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.h:15:0,
                 from /home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:6:
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteESP8266.h:141:7: error: candidate is: int IRrecv::compare(unsigned int, unsigned int)
   int compare(unsigned int oldval, unsigned int newval);
       ^
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRrecv.cpp:677:57: error: no 'match_result_t IRrecv::matchData(volatile uint16_t*, uint16_t, uint16_t, uint32_t, uint16_t, uint32_t, uint8_t)' member function declared in class 'IRrecv'
                                  const uint8_t tolerance) {
                                                         ^
```

crankyoldgit, 2018-02-18:
I believe you have a corrupt install of the library.
E.g "
/home/john/Arduino/scetchbooks/libraries/IRremoteESP8266/IRremoteInt.h"
IRremoteInt.h hasn't been in the library since v2.0.
Delete the contents of that directory and reinstall the lib.

crankyoldgit, 2018-02-21:
friendly ping.

s0170071, 2018-02-21:
haven't tried yet, maybe next weekend. It may be that there are multiple versions of the library (due to ESPEasy project), however, the Arduino IDE clearly states that it is using the 'right' one. But that may be an occasional lie...

crankyoldgit, 2018-02-28:
Friendly ping. How did it go?

crankyoldgit, 2018-03-16:
Marking this closed due to inactivity. Please re-open if you are able to get back to investigating the problem etc.

closed 2018-03-16 by crankyoldgit