crankyoldgit/IRremoteESP8266 #417: Help Decoding HITACHI AC Remote

matkappert, 2018-02-17:
Hello all,
I've been playing around with the library trying to decode my Hitachi RAS-35THA6 remote but i'm not getting anywhere with it, any help at this point would be awesome! 
Here's what I have as far:

```
HDR_MARK           3436U
HDR_SPACE          1640U
BIT_MARK            420U
ONE_SPACE          1250U
ZERO_SPACE          500U
```

Here's some raw data:

Remote: 'On' '16c' 'auto fan' 'cooling mode'
` rawData[451] = {3318, 1720,  400, 1276,  400, 432,  398, 434,  398, 434,  400, 432,  398, 432,  398, 432,  398, 434,  398, 432,  398, 434,  400, 432,  398, 434,  398, 1278,  398, 434,  398, 434,  396, 434,  398, 434,  398, 432,  398, 434,  398, 432,  398, 1276,  426, 1252,  424, 408,  424, 406,  424, 408,  426, 406,  398, 432,  398, 434,  398, 432,  400, 432,  398, 1276,  426, 408,  424, 1252,  426, 1252,  424, 1250,  426, 1252,  428, 1250,  426, 1252,  424, 406,  426, 1248,  428, 1252,  426, 406,  426, 406,  424, 408,  400, 432,  400, 430,  400, 432,  400, 430,  400, 432,  400, 1276,  400, 1276,  402, 1276,  400, 1276,  400, 1276,  400, 1278,  400, 1276,  402, 1276,  402, 428,  402, 430,  400, 430,  402, 1276,  400, 430,  402, 430,  400, 432,  402, 428,  402, 1278,  400, 430,  402, 430,  402, 1276,  402, 428,  402, 430,  402, 430,  400, 1276,  402, 430,  402, 430,  402, 430,  402, 430,  402, 428,  402, 430,  404, 430,  402, 428,  402, 430,  402, 1276,  402, 430,  402, 428,  402, 430,  400, 428,  402, 430,  402, 430,  402, 430,  402, 430,  402, 428,  402, 430,  402, 1274,  402, 428,  402, 430,  402, 430,  402, 430,  402, 430,  402, 428,  402, 428,  402, 428,  404, 428,  404, 428,  402, 1276,  400, 430,  402, 430,  400, 432,  400, 456,  374, 432,  400, 456,  404, 428,  404, 426,  404, 428,  402, 428,  402, 430,  402, 430,  400, 432,  398, 434,  376, 454,  378, 454,  380, 452,  378, 452,  404, 428,  406, 424,  432, 398,  406, 426,  430, 402,  404, 428,  428, 402,  400, 430,  400, 432,  398, 434,  398, 432,  398, 434,  372, 460,  374, 434,  398, 432,  398, 434,  396, 434,  376, 456,  376, 456,  376, 456,  376, 1300,  378, 454,  378, 452,  378, 454,  378, 454,  378, 454,  378, 452,  378, 454,  400, 432,  402, 430,  402, 430,  402, 430,  402, 428,  402, 430,  402, 430,  400, 430,  402, 430,  400, 432,  400, 430,  400, 432,  400, 430,  402, 430,  400, 432,  398, 432,  400, 430,  400, 432,  398, 432,  398, 434,  398, 432,  398, 432,  400, 434,  398, 432,  398, 432,  398, 434,  398, 434,  396, 434,  398, 434,  398, 432,  398, 434,  398, 432,  398, 456,  376, 454,  376, 436,  396, 454,  376, 454,  378, 454,  376, 454,  376, 456,  374, 458,  374, 1302,  374, 456,  374, 458,  374, 458,  376, 456,  374, 456,  374, 456,  374, 456,  376, 456,  376, 456,  374, 458,  374, 458,  374, 458,  372, 458,  372, 460,  374, 458,  372, 460,  372, 460,  372, 460,  370, 462,  372, 460,  372, 460,  370, 462,  370, 462,  356, 1320,  368, 464,  346, 1332,  356, 476,  368, 1310,  366, 1312,  366, 464,  366, 466,  366};  // UNKNOWN 3FFBF9FE813FC0`

Remote: 'Off' '16c' 'auto fan' 'cooling mode'
`rawData[451] = {3298, 1736,  382, 1296,  382, 428,  402, 430,  402, 430,  400, 430,  402, 428,  402, 430,  400, 430,  402, 430,  402, 428,  402, 430,  402, 430,  402, 1296,  382, 430,  402, 428,  402, 430,  402, 430,  402, 430,  402, 430,  402, 430,  402, 1296,  382, 1294,  382, 430,  402, 430,  402, 430,  402, 428,  402, 430,  402, 430,  404, 428,  402, 428,  402, 1296,  382, 428,  402, 1274,  404, 1294,  384, 1292,  384, 1294,  382, 1296,  382, 1296,  382, 430,  402, 1294,  382, 1294,  382, 428,  402, 430,  402, 430,  402, 430,  402, 430,  402, 430,  402, 430,  400, 432,  400, 1296,  380, 1296,  382, 1294,  382, 1294,  382, 1296,  380, 1296,  382, 1294,  382, 1296,  382, 428,  404, 428,  402, 430,  402, 1296,  382, 430,  402, 428,  402, 430,  402, 430,  402, 1296,  382, 430,  400, 430,  402, 1294,  380, 430,  400, 430,  402, 430,  402, 428,  404, 1272,  402, 430,  402, 430,  402, 428,  404, 428,  402, 430,  402, 430,  402, 428,  402, 430,  404, 1272,  404, 428,  404, 426,  404, 428,  402, 428,  402, 430,  402, 428,  402, 430,  402, 428,  404, 428,  402, 430,  402, 1274,  404, 428,  404, 428,  402, 430,  402, 428,  402, 428,  404, 428,  404, 428,  402, 430,  404, 428,  402, 428,  402, 1274,  404, 426,  404, 428,  402, 430,  402, 428,  402, 428,  404, 428,  402, 428,  404, 428,  404, 428,  402, 430,  402, 430,  402, 428,  402, 430,  402, 430,  402, 428,  404, 428,  402, 432,  402, 428,  402, 428,  402, 430,  402, 430,  402, 430,  402, 428,  404, 430,  402, 430,  402, 430,  402, 428,  404, 428,  402, 430,  402, 430,  402, 430,  402, 430,  402, 428,  402, 430,  402, 430,  402, 430,  400, 430,  402, 430,  402, 430,  402, 428,  404, 428,  404, 428,  404, 428,  402, 430,  404, 428,  402, 430,  402, 428,  404, 428,  404, 430,  400, 430,  402, 428,  402, 432,  400, 428,  402, 454,  376, 454,  378, 454,  378, 452,  378, 454,  404, 426,  380, 452,  378, 456,  402, 428,  404, 428,  380, 452,  380, 450,  380, 452,  380, 452,  380, 450,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  378, 452,  404, 428,  402, 428,  378, 454,  376, 456,  400, 430,  402, 430,  376, 454,  378, 454,  376, 456,  398, 434,  376, 456,  376, 458,  372, 1278,  398, 458,  374, 458,  372, 460,  376, 454,  372, 460,  352, 480,  352, 478,  376, 456,  352, 480,  352, 480,  352, 478,  376, 454,  356, 478,  378, 452,  380, 430,  402, 428,  404, 428,  404, 428,  406, 424,  434, 400,  430, 400,  434, 396,  436, 396,  434, 400,  434, 394,  436, 1240,  438, 396,  434, 1240,  436, 1238,  436, 396,  436, 1240,  436};  // UNKNOWN 3FFBF9FE813FC0 `

Remote: 'On' '32c' 'auto fan' 'cooling mode'
`rawData[451] = {3320, 1714,  404, 1274,  404, 428,  404, 428,  402, 430,  402, 428,  402, 428,  402, 430,  402, 428,  402, 430,  402, 428,  402, 430,  404, 428,  402, 1276,  400, 428,  402, 430,  402, 428,  404, 430,  402, 428,  402, 430,  402, 430,  404, 1274,  402, 1272,  402, 430,  402, 428,  402, 430,  402, 428,  402, 430,  402, 430,  402, 428,  402, 430,  402, 1274,  402, 428,  404, 1272,  404, 1274,  404, 1274,  402, 1274,  402, 1274,  404, 1274,  402, 428,  402, 1276,  402, 1276,  402, 428,  402, 430,  402, 430,  402, 428,  402, 430,  402, 430,  402, 428,  402, 430,  402, 1272,  404, 1274,  404, 1274,  402, 1276,  402, 1296,  380, 1274,  404, 1274,  402, 1274,  404, 428,  404, 428,  402, 430,  402, 1272,  404, 430,  404, 428,  402, 428,  404, 428,  402, 1274,  402, 430,  404, 428,  404, 1274,  402, 430,  402, 430,  402, 430,  404, 1272,  404, 428,  404, 428,  402, 430,  402, 430,  402, 428,  402, 428,  402, 430,  402, 428,  402, 430,  404, 1270,  406, 428,  402, 430,  402, 430,  402, 428,  402, 428,  404, 428,  402, 430,  402, 428,  402, 428,  404, 428,  404, 428,  404, 1276,  400, 428,  402, 428,  402, 430,  402, 428,  402, 428,  402, 428,  402, 432,  406, 422,  402, 430,  402, 1274,  402, 430,  402, 428,  404, 428,  402, 430,  402, 430,  402, 428,  404, 428,  402, 430,  402, 428,  404, 428,  402, 430,  402, 428,  404, 430,  404, 428,  402, 428,  404, 428,  404, 428,  402, 428,  404, 428,  402, 430,  402, 428,  402, 428,  402, 430,  402, 428,  404, 428,  402, 428,  402, 430,  402, 428,  404, 428,  402, 428,  402, 430,  404, 428,  404, 428,  404, 428,  402, 430,  404, 428,  402, 430,  402, 428,  404, 1296,  382, 428,  402, 428,  404, 428,  402, 430,  404, 428,  402, 428,  404, 428,  402, 430,  402, 430,  404, 428,  402, 430,  402, 430,  402, 452,  382, 428,  404, 428,  402, 450,  382, 450,  380, 450,  380, 430,  402, 450,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 450,  380, 452,  380, 450,  380, 452,  380, 452,  378, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 450,  380, 452,  380, 452,  380, 450,  404, 428,  404, 426,  380, 450,  380, 452,  380, 452,  380, 1272,  424, 432,  378, 454,  402, 428,  402, 430,  378, 454,  376, 454,  374, 456,  374, 458,  374, 456,  354, 478,  374, 458,  374, 456,  354, 478,  354, 478,  354, 478,  354, 478,  354, 478,  352, 478,  354, 478,  354, 478,  354, 476,  354, 478,  354, 478,  356, 1296,  382, 452,  380, 1294,  382, 452,  402, 1272,  406, 452,  380, 452,  382, 424,  406};  // UNKNOWN 3FFBF9FE813FC0 `

Remote: 'On' '32c' 'auto fan' 'heating mode'
`rawData[451] = {3322, 1718,  400, 1278,  398, 432,  402, 430,  400, 430,  402, 430,  402, 430,  400, 432,  400, 432,  402, 430,  400, 430,  402, 430,  400, 430,  402, 1276,  402, 430,  400, 430,  402, 430,  402, 430,  402, 428,  402, 430,  400, 430,  402, 1278,  4 [... cut ...] 78, 430,  404, 452,  376, 430,  404, 428,  406, 426,  406, 426,  404, 426,  406, 428,  406, 426,  430, 400,  430, 1244,  408, 424,  434, 398,  432, 400,  406, 424,  432, 400,  406, 426,  430, 400,  432, 400,  432, 400,  430, 402,  428, 402,  404, 428,  404, 426,  406, 426,  406, 426,  404, 428,  428, 402,  404, 428,  404, 428,  402, 428,  404, 428,  402, 428,  404, 428,  404, 1274,  402, 1274,  404, 428,  400, 1276,  402, 430,  402, 430,  402, 430,  402, 430,  400};  // UNKNOWN 3FFBF9FE813FC0
 `

crankyoldgit, 2018-02-18:
I suggest you use the [AutoAnalyseRawData.sh](https://github.com/markszabo/IRremoteESP8266/blob/master/tools/AutoAnalyseRawData.sh) program in the tools directory to try to decode it.

e.g. based on your first chunk of data, it's a 224 bit (28 bytes) code, most likely: 0x80080C02FD807F8848802004008000000001000000000000800000AC (MSB first order)

Use the the program above to decode each message, and you might be able to work out what bits do what.
See the FAQ for dealing with A/C remote messages. i.e. they are a LOT of work.
You'll basically need to work out via collecting lots of data where the mode bits are, the temp bits, the fan bits, checkums, etc in that 28 byte message if you want a full decoder/constructor.

For sending the code, you'll want something like:
```
void IRsend::sendXYZ(uint8_t data[], uint16_t nbytes, uint16_t repeat) {
  // nbytes should typically be 28
  // data should typically be of a type: uint8_t data[28] = {};
  // data[] will is assumed to be in MSB order.
  for (uint16_t r = 0; r <= repeat; r++) {
    sendGeneric(HDR_MARK, HDR_SPACE, BIT_MARK, ONE_SPACE, BIT_MARK, ZERO_SPACE, BIT_MARK
                100, data, nbytes, 38, true, 0, 50);
}
```

Per example above, your data would be:
```
uint8_t code_to_send[28] = {0x80, 0x08, 0x0C, 0x02, 0xFD, 0x80, 0x7F, 0x88, 0x48, 0x80, 0x20, 0x04, 0x00, 0x80, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00, 0xAC};
```

crankyoldgit, 2018-04-09:
FYI Prelim support for Hitachi A/C has been added in the new [v2.4.0](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.4.0) release

crankyoldgit, 2018-04-16:
@matkappert Have you had a chance to test the new changes?

darshkpatel, 2018-04-26:
@crankyoldgit I have a Hitachi A/C , I'd test the changes tonight

crankyoldgit, 2018-04-26:
@darshkpatel That'd be awesome! Thanks. Looking forward to the feedback.

darshkpatel, 2018-04-26:
Hey @crankyoldgit,
I tried decode using IRrecvDumpv2 sketch.
here are the results :

### ON - 32c cool (fan auto)
```
Timestamp : 000726.583
Encoding  : UNKNOWN
Code      : 19581AEF (426 bits)
Library   : v2.4.0

Raw Timing[851]:
   +  3432, -  1654,    +   492, -  1180,    +   488, -   360,    +   486, -   360, 
   +   486, -   360,    +   486, -   362,    +   486, -   362,    +   460, -   360, 
   +   486, -   362,    +   486, -   362,    +   486, -   360,    +   486, -   362, 
   +   486, -   362,    +   486, -  1184,    +   488, -   362,    +   486, -   360, 
   +   486, -   362,    +   486, -   362,    +   486, -   362,    +   460, -   362, 
   +   486, -   362,    +   486, -   362,    +   484, -   362,    +   486, -   362, 
   +   486, -   362,    +   486, -   362,    +   458, -   362,    +   484, -   362, 
   +   486, -   362,    +   484, -   364,    +   484, -   362,    +   486, -  1184, 
   +   486, -   362,    +   486, -  1210,    +   488, -  1210,    +   488, -  1184, 
   +   486, -  1210,    +   488, -  1210,    +   488, -  1182,    +   488, -   360, 
   +   488, -  1208,    +   488, -  1208,    +   488, -  1182,    +   488, -  1208, 
   +   488, -  1208,    +   488, -  1208,    +   462, -  1208,    +   490, -  1208, 
   +   488, -  1208,    +   490, -   332,    +   488, -   360,    +   488, -   360, 
   +   488, -   360,    +   486, -   360,    +   486, -   360,    +   486, -   362, 
   +   460, -   360,    +   486, -   362,    +   486, -   362,    +   486, -  1210, 
   +   488, -  1210,    +   460, -   362,    +   486, -   362,    +   486, -  1210, 
   +   488, -  1210,    +   488, -  1182,    +   488, -  1210,    +   488, -   362, 
   +   486, -   360,    +   488, -  1182,    +   488, -  1210,    +   488, -   362, 
   +   486, -   362,    +   486, -   360,    +   486, -  1182,    +   488, -   362, 
   +   486, -   362,    +   486, -  1210,    +   488, -   360,    +   486, -   336, 
   +   486, -  1210,    +   488, -  1210,    +   488, -   360,    +   486, -  1182, 
   +   488, -  1210,    +   488, -   360,    +   488, -  1208,    +   488, -  1182, 
   +   488, -   362,    +   486, -   362,    +   486, -   360,    +   486, -  1210, 
   +   460, -   362,    +   486, -   360,    +   486, -   362,    +   486, -  1210, 
   +   486, -   362,    +   486, -  1184,    +   488, -  1208,    +   488, -   360, 
   +   486, -  1210,    +   488, -  1184,    +   488, -  1208,    +   488, -   362, 
   +   486, -  1208,    +   462, -   360,    +   486, -   360,    +   486, -   362, 
   +   486, -   360,    +   486, -   362,    +   486, -   360,    +   486, -   334, 
   +   486, -  1210,    +   488, -  1208,    +   488, -  1210,    +   488, -  1182, 
   +   488, -  1208,    +   488, -  1210,    +   488, -  1182,    +   488, -  1208, 
   +   488, -   360,    +   488, -   360,    +   486, -   360,    +   486, -   334, 
   +   486, -   362,    +   486, -   362,    +   486, -   362,    +   486, -   362, 
   +   486, -   362,    +   486, -  1182,    +   488, -  1210,    +   486, -  1210, 
   +   488, -  1210,    +   460, -  1210,    +   488, -  1210,    +   488, -  1210, 
   +   488, -  1184,    +   488, -   360,    +   486, -   360,    +   486, -   362, 
   +   486, -   362,    +   486, -   360,    +   460, -   360,    +   486, -   362, 
   +   486, -   362,    +   484, -  1210,    +   488, -  1182,    +   488, -  1210, 
   +   488, -  1210,    +   488, -  1210,    +   488, -  1182,    +   488, -  1208, 
   +   488, -  1208,    +   488, -   360,    +   460, -   360,    +   486, -   360, 
   +   486, -   360,    +   488, -   360,    +   486, -   362,    +   486, -   362, 
   +   486, -   334,    +   486, -  1210,    +   488, -  1208,    +   488, -  1210, 
   +   460, -  1210,    +   488, -  1208,    +   488, -  1210,    +   488, -  1182, 
   +   488, -  1208,    +   488, -   360,    +   486, -   362,    +   486, -   360, 
   +   488, -   334,    +   486, -   360,    +   486, -   362,    +   486, -   362, 
   +   486, -   362,    +   486, -  1210,    +   460, -  1210,    +   488, -  1210, 
   +   488, -  1210,    +   488, -  1182,    +   488, -  1210,    +   488, -  1208, 
   +   488, -  1182,    +   488, -   360,    +   486, -   360,    +   486, -   362, 
   +   486, -   362,    +   486, -   360,    +   486, -   362,    +   460, -   360, 
   +   486, -   362,    +   486, -  1210,    +   486, -  1212,    +   486, -  1184, 
   +   488, -  1210,    +   488, -  1208,    +   488, -  1182,    +   488, -  1208, 
   +   488, -  1208,    +   488, -  1208,    +   462, -  1208,    +   488, -   360, 
   +   486, -   362,    +   486, -  1208,    +   488, -   334,    +   486, -  1208, 
   +   488, -   360,    +   486, -   360,    +   486, -   362,    +   486, -  1182, 
   +   488, -  1210,    +   488, -   360,    +   486, -  1210,    +   488, -   334, 
   +   486, -  1208,    +   488, -  1208,    +   488, -   360,    +   486, -   360, 
   +   486, -   334,    +   486, -  1210,    +   488, -  1210,    +   488, -  1210, 
   +   460, -  1210,    +   488, -   360,    +   486, -  1208,    +   488, -  1182, 
   +   490, -  1208,    +   488, -   360,    +   486, -   360,    +   486, -   360, 
[... 500 lines cut ...]
   +   492, -  1178,    +   492, -  1206,    +   490, -  1204,    +   492, -  1204, 
   +   466, -  1204,    +   492, -   358,    +   490, -   358,    +   488, -   312, 
   +   536, -   312,    +   534, -   312,    +   510, -   314,    +   534, -   316, 
[... cut ...]   492, 1206,  464, 306,  540, 312,  536, 310,  536, 310,  536, 318,  530, 320,  528, 264,  556, 318,  528, 1206,  492, 1206,  492, 1180,  490, 1206,  492, 1204,  492, 1204,  466, 1206,  492, 1204,  492, 358,  488, 308,  540, 306,  514, 310,  536, 360,  488, 312,  534, 314,  532, 320,  526, 1182,  488, 1206,  492, 1206,  492, 1206,  464, 1206,  492, 1204,  492, 1204,  492, 1178,  492, 306,  540, 312,  536, 310,  536, 314,  534, 360,  460, 314,  532, 314,  532, 360,  486};  // UNKNOWN 8DCDA59C
```

crankyoldgit, 2018-04-27:
@darshkpatel I think that's a different HITACHI protocol. Boy, these are huge!

Analysing the "ON - 32c cool (fan auto)" signal (first one), I get:
```
./AutoAnalyseRawData.sh < /tmp/t
Potential Mark Candidates (using a range of 200):
3432
492

Potential Space Candidates (using a range of 200):
1654
1212
364

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK = 3432
BIT_MARK = 492
HDR_SPACE = 1654
ONE_SPACE = 1212
ZERO_SPACE = 364

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+1000000000001000000000000000001011111101111111110000000000110011110011000100100110110110001000101101110100000001111111100000000011111111000000001111111100000000111111110000000011111111000000001111111111001010001101011000111101110000000000001111111100000000111111110000000111111110110000000011111110000000011111110001000111101110000000001111111100000000111111111111111100000000111111110000000011111111000000001111111100000000
  Bits: 424
  Hex:  0x80080002FDFF0033CC49B622DD01FE00FF00FF00FF00FF00FFCA358F7000FF00FF01\
FEC03F807F11EE00FF00FFFF00FF00FF00FF00 (MSB first)
        0xFF00FF00FF00FFFF00FF007788FE01FC037F80FF00FF000EF1AC53FF00FF00FF00FF\
00FF007F80BB446D9233CC00FFBF40001001 (LSB first)
  Dec:  21666770463250971033249250656885825194738684083025669751149395029707\
818003531088973511075669188805888302356754191519362576350976 (MSB first)
        16857184424372658669179408622645041920057617761062334415219526095024\
9734545225589859643087626664521656624065247901356434395137 (LSB first)
  Bin:  1000000000001000000000000000001011111101111111110000000000110011110011000100100110110110001000101101110100000001111111100000000011111111000000001111111100000000111111110000000011111111000000001111111111001010001101011000111101110000000000001111111100000000111111110000000111111110110000000011111110000000011111110001000111101110000000001111111100000000111111111111111100000000111111110000000011111111000000001111111100000000 (MSB first)
        0000000011111111000000001111111100000000111111110000000011111111111111110000000011111111000000000111011110001000111111100000000111111100000000110111111110000000111111110000000011111111000000000000111011110001101011000101001111111111000000001111111100000000111111110000000011111111000000001111111100000000011111111000000010111011010001000110110110010010001100111100110000000000111111111011111101000000000000000001000000000001 (LSB first)
```

That's 424 bits (53 bytes). Probably not all of it "real" data. i.e. lots of 0xFF and 0x00 etc.

I'm happy to look at adding support for _this_ protocol, but I don't think its the same as the newly added Hitachi one (which is 28 bytes/224 bits). I also checked to see if yours included a repeat. It doesn't appear to. 424 bits appears to be the correct length for this variant.

crankyoldgit, 2018-05-12:
For the record, it appears there are at least three different protocols used by Hitachi. The one we've coded, the one @darshkpatel has, and another "shorter" one reported by @DiggiD in #453

i.e. Three different sized codes.

crankyoldgit, 2018-09-14:
@darshkpatel Any update on this? If not, I think we should consider closing this issue for now.

crankyoldgit, 2018-11-23:
Hitachi 28 Byte A/C Protocol is has been included in the v2.5.3 release.