crankyoldgit/IRremoteESP8266 #420: IR Led on GPIO 4

aleonnet, 2018-02-20:
Hi David @crankyoldgit 
Keep up the good job! I don't know if it's a silly question but...
If IR_LED is defined by a parameter, why the code header states that "An IR LED circuit *MUST* be connected to ESP8266 pin 4 (D2). See IR_LED"?
What would be the effect of connection the IR_LED on a differente GPIO?

// Configuration parameters
#define IR_LED 4 // GPIO the IR LED is connected to/controlled by. GPIO 4 = D2.

Thank you.
Alessandro

crankyoldgit, 2018-02-21:
You are correct. It is a definable parameter. The wording was chosen to keep the example simple. I'll try to re-word it so that it clearer that it needs to be connected to what ever IR_LED is defined as.

crankyoldgit, 2018-02-21:
@aleonnet Can you please take a look at https://github.com/markszabo/IRremoteESP8266/commit/11b55f2df551d0dbafe997cf36d4209dbacd9404 and see if those new comments make it clearer for you/others?

closed 2018-02-28 by crankyoldgit