crankyoldgit/IRremoteESP8266 #421: ir_Gree.h: No such file or directory

dragouf, 2018-02-23:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revison of the library used
2.3.1

### Expected behavior
try to build IRrecvDumpV2 example

### Actual behavior
got this compilation error :  fatal error: ir_Gree.h: No such file or directory

### Steps to reproduce the behavior
install IRremoteESP8266 2.3.1 library on arduino ide

#### Circuit diagram and hardware used (if applicable)
receiver/emitter from electrodragon

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)

Yes

crankyoldgit, 2018-02-23:
Hi @dragouf thanks for the report.
The file `ir_Gree.h` certainly exists under the `src/` directory. e.g. https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Gree.h but only under the current `master` branch of the library. It is not yet in any of the official released versions, certainly not in v2.3.1 as per your report.

How did you install the library? Did you use the Arduino Library Manager?
I also note, you are using Version 2.3.1, and 2.3.2 has been out for almost 4 weeks. You should update, but I'm *not* convinced that will fix your problem. `ir_Gree.h` has been in the library only for the last 25 days, just after the 2.3.2 release. I suspect you've downloaded part of the library out of sync with the rest of the library.

You should either move to v2.3.2 and revert any "example code" you may have already saved. OR delete the version of the library you have, download the zip file or `git clone https://github.com/markszabo/IRremoteESP8266.git` again to get a copy that is fully in sync.

crankyoldgit, 2018-02-28:
Just a follow up. How did you go fixing your install?

dragouf, 2018-02-28:
in the end I removed reference of the new code to keep managing my library with Arduino IDE. but I guess just download the code from GitHub would have fix the problem like you explained me.

closed 2018-02-28 by dragouf