crankyoldgit/IRremoteESP8266 #423: difference between address, command and data

DESAWI, 2018-02-27:
what is the difference between :
uint32_t address = 0x9F00;
uint32_t command = 0x94;
uint64_t data = 0xF929D6;

address, command and data

crankyoldgit, 2018-02-27:
What protocol (`decode_type`) is it from?
`data`/`value` is typically the full "code" for the message for that given protocol. You typically only need this value to re-send the message later with the appropriate `sendBlah()` function.
`command` and `address` are protocol specific and normally derived from the `data`/`value` variable. They mean different things for different protocols.

DESAWI, 2018-02-27:
so,
uint32_t address = 0x9F00;
uint32_t command = 0x94;
uint64_t data = 0xF929D6; 
the "data = 0xF929D6" is the code if I send it again to device it will operate button function.
but, I am still confused because command and address are totally different from data.

crankyoldgit, 2018-02-27:
I repeat: "What protocol (decode_type) is it from?"

DESAWI, 2018-02-27:
Encoding  : NEC
Code      : 2FD5AA5 (32 bits)
Library   : v2.3.2

Raw Timing[67]:
   +  8962, -  4482,    +   564, -   544,    +   564, -   598,    +   536, -   596, 
   +   538, -   596,    +   538, -   596,    +   538, -   596,    +   538, -  1714, 
   +   538, -   594,    +   540, -  1714,    +   538, -  1690,    +   564, -  1712, 
   +   540, -  1686,    +   566, -  1712,    +   538, -  1714,    +   538, -   598, 
   +   538, -  1686,    +   566, -   572,    +   564, -  1712,    +   538, -   596, 
   +   538, -  1714,    +   540, -  1714,    +   538, -   570,    +   564, -  1714, 
   +   538, -   596,    +   538, -  1714,    +   540, -   596,    +   538, -  1712, 
   +   540, -   596,    +   538, -   596,    +   540, -  1712,    +   540, -   596, 
   +   538, -  1688,    +   566

uint16_t rawData[67] = {8962, 4482,  564, 544,  564, 598,  536, 596,  538, 596,  538, 596,  538, 596,  538, 1714,  538, 594,  540, 1714,  538, 1690,  564, 1712,  540, 1686,  566, 1712,  538, 1714,  538, 598,  538, 1686,  566, 572,  564, 1712,  538, 596,  538, 1714,  540, 1714,  538, 570,  564, 1714,  538, 596,  538, 1714,  540, 596,  538, 1712,  540, 596,  538, 596,  540, 1712,  540, 596,  538, 1688,  566};  // NEC 2FD5AA5
uint32_t address = 0x40;
uint32_t command = 0x5A;
uint64_t data = 0x2FD5AA5;

crankyoldgit, 2018-02-27:
See https://www.sbprojects.net/knowledge/ir/nec.php for some of the gory details.

In short, `data` is made up from the `address` then the inverted `address`, then the `command` and then the inverted `command`, and due to historical reasons, the library used to report the bits in reversed order for the data/value variable.

See https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_NEC.cpp#L75 & https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_NEC.cpp#L187 for even more details.

i.e. `address` and `command` are VERY protocol specific.

closed 2018-03-09 by crankyoldgit

feder-ico, 2018-06-28:
Hi @crankyoldgit , Let me ask you for a confirmation because I am a bit lost.

This is what a got with IRrecvDumpV2 when I pressed power button on a LG M2380A TV (Remote MKJ33981435)

IRrecvDumpV2 is now running and waiting for IR input on Pin 14
Timestamp : 000046.299
Encoding  : NEC
Code      : 20DF10EF (32 bits)
Library   : v2.4.1

Raw Timing[71]:
   +  9038, -  4456,    +   594, -   536,    +   594, -   536,    +   594, -  1654, 
   +   596, -   534,    +   596, -   536,    +   596, -   536,    +   594, -   536, 
   +   594, -   536,    +   594, -  1654,    +   594, -  1654,    +   596, -   534, 
   +   594, -  1654,    +   596, -  1652,    +   596, -  1654,    +   596, -  1654, 
   +   594, -  1654,    +   594, -   536,    +   594, -   536,    +   596, -   534, 
   +   596, -  1654,    +   594, -   536,    +   594, -   536,    +   594, -   534, 
   +   596, -   536,    +   594, -  1654,    +   596, -  1654,    +   594, -  1656, 
   +   594, -   536,    +   594, -  1654,    +   594, -  1654,    +   596, -  1652, 
   +   596, -  1654,    +   594, - 39728,    +  9036, -  2220,    +   594

uint16_t rawData[71] = {9038, 4456,  594, 536,  594, 536,  594, 1654,  596, 534,  596, 536,  596, 536,  594, 536,  594, 536,  594, 1654,  594, 1654,  596, 534,  594, 1654,  596, 1652,  596, 1654,  596, 1654,  594, 1654,  594, 536,  594, 536,  596, 534,  596, 1654,  594, 536,  594, 536,  594, 534,  596, 536,  594, 1654,  596, 1654,  594, 1656,  594, 536,  594, 1654,  594, 1654,  596, 1652,  596, 1654,  594, 39728,  9036, 2220,  594};  // NEC 20DF10EF
uint32_t address = 0x4;
uint32_t command = 0x8;
uint64_t data = 0x20DF10EF;

**So, If I do the following should power on (or off) my TV?**

```
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>
#define IR_LED 4  //I am using a D2 PIN on NODEMCU.
IRsend irsend(IR_LED); 

// This is what I captured with IRrecvDumpV2.ino
uint16_t rawData[67] = {9038, 4456,  594, 536,  594, 536,  594, 1654,  596, 534,  596, 536,  596, 536,  594, 536,  594, 536,  594, 1654,  594, 1654,  596, 534,  594, 1654,  596, 1652,  596, 1654,  596, 1654,  594, 1654,  594, 536,  594, 536,  596, 534,  596, 1654,  594, 536,  594, 536,  594, 534,  596, 536,  594, 1654,  596, 1654,  594, 1656,  594, 536,  594, 1654,  594, 1654,  596, 1652,  596, 1654,  594, 39728,  9036, 2220,  594};

void setup() {
  irsend.begin();
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {
#if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x20DF10EF, 32);
#endif  // SEND_NEC
  delay(2000);
#if SEND_RAW
  Serial.println("a rawData capture from IRrecvDumpV2");
  irsend.sendRaw(rawData, 67, 38);  // Send a raw data capture at 38kHz.
#endif  // SEND_RAW
  delay(2000);
}
```

**Is this ok?**
I know I should use only one of those (NEC or RAW). I prefer to know how to transmit raw for being able to use it with other devices different than NEC.
**How can I know the frecuency (38) from the IRrecvDumpV2?**
I see the led flashing with my smartphone camera but TV is doing nothing.

I much apreciate your help.

Thanks in advance.

Best regards.
Federico.-

crankyoldgit, 2018-06-28:
You only need to use either `irsend.sendNEC(0x20DF10EF);` **or** the `sendRaw()` method you used. Both are not needed, only one is needed. It looks like you are ALMOST using the sendRaw() correctly, so yes, that's how you do it. However, you've used the wrong length of the array. The DumpV2 rawData output is says it is **71** entries long, so you need to change all the 67's to 71's.

Also, upgrade to v2.4.2 of the library. It has a small but important timing change that may help you.

As for:
> How can I know the frecuency (38) from the IRrecvDumpV2?

You can't. Only the hardware IR module can see the frequency modulation. It's hidden from the ESP8266. You can make an educated guess though. If you are using a 38kHz IR module, and you're getting a reliable & repeatable signal. Then it is probably a 38kHz signal being received. 38kHz is also the most common freq. modulation used by protocols, and it is also the most common IR receiver module type.