crankyoldgit/IRremoteESP8266 #424: Help decoding Carrier AC - probably missunderstanding library

sebaretamozo, 2018-02-28:
Hello, I have the next code decoded by the library IRrecvDumpV2.ino for a remote controller like this:
http://www.ttinwest.com.pl/pl/p/file/8b41d748aa62eaa4652f97f92d83bdb7/KJR-12B.pdf

I think my circuit is bad, or I need a filter for 38KHz because I see very small times (arround 13 us wich is the 38Khz signal). 

The GPIO14 pin is connected through a resistor directly to the microcontroller pin of the remote control.

Here is my dump of the controller in AUTO - 24°C - FAN AUTO

Timestamp : 000583.786
Encoding  : UNKNOWN
Code      : 3A046F6C (2499 bits)
Library   : v2.3.2

Raw Timing[4997]:
   +    12, -    14,    +    12, -    12,    +    12, -    12,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    12,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    12,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    12,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    12,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    12,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    12,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    12, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    12, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    12, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    12, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -  4458,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    12,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -  1634,    +    12, -    14,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    14,    +    12, -    12, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    14,    +    12, -   536,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
[... 188 lines cut ...]
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -  1634,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -    14,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -   536,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    14,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    14, 
   +    12, -   538,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -   536,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    12,    +    12, -, , 
uint16_t rawData[4997] = {12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 12,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,   [... cut ...] , 14,  12, 14,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 1634,  12, 14,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, ,  12, 14,  12, ,  12, 12,  12, ,  12, 14,  12, ,  12, 1634,  12, 14,  12, 14,  , 12,  12, 14,  , 12,  12, 14,  , 14,  12, 14,  , 14,  12, 12,  , 14,  12, 14,  , 14,  12, 14,  , 14,  12, 14,  , 14,  12, 14,  , 14,  12, 14,  , 14,  12};

crankyoldgit, 2018-02-28:
@sebaretamozo I'd almost guarantee you are using an IR LED detecting diode (2 pins), rather than an IR receiving module (3 pins). This library requires/expects the latter. e.g. a [TSOP38238](https://www.sparkfun.com/products/10266) or similar. Previous Carrier A/C remotes have been reported to work with a 38kHz IR receiver module. This is the most common type.

crankyoldgit, 2018-02-28:
Sorry, I re-read your original message.

> The GPIO14 pin is connected through a resistor directly to the microcontroller pin of the remote control.

This library does not to handle that sort of operation/configuration. It expects to 'receive' via an IR receiver module that handles the de-modulation.

I suggest you attach a 940nm IR LED to the remote circuit, and then use a IR receiver module on the ESP to decode/capture it.

closed 2018-02-28 by crankyoldgit