crankyoldgit/IRremoteESP8266 #427: ESP8266 doesn't show any output in serial monitor

wakandan, 2018-03-03:
## Version/revison of the library used: master, as of March 3rd

### Expected behavior: the serial monitor should show something, anything
_What steps did you do and what should it have done?_
I compile and uploaded the stock code on 2 examples: IRrecvDemo & IRrecvDumpV2, both doesn't work

### Actual behavior: the serial monitor doesn't show anything
What I have try to diagnose the problem: 
- Test if some other port (besides 14) works by changing the data port to some other: serial monitor also doesn't show anything
- Test with another (new) ESP8266 (nodemcu module): serial monitor also doesn't show anything
- Test with another IR receiver module: same result
- Test with an Arduino board with nearly identical code (but different library): it was able to read the command off my tv remote
- Test with Arduino but with 3.3 volt input: it still can read the IR command

#### Output of raw data from IRrecvDumpV2.ino (if applicable): there's nothing. Is there a way I can read anything off the ir receiver to make sure that it works?

### Steps to reproduce the behavior
Just follow this guide https://github.com/markszabo/IRremoteESP8266/wiki#ir-receiving

#### Example code used: stock code of IRrecvDumpV2 and IRrecvDemo

#### Circuit diagram and hardware used (if applicable): https://github.com/markszabo/IRremoteESP8266/wiki#ir-receiving

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes

crankyoldgit, 2018-03-03:
Has you previously had this library working for you?

Some changes were made to the serial output in V2.3.3. Can you please
download and try the v2.3.2 release example code and see if that works?

crankyoldgit, 2018-03-03:
Also, check you are using the correct Baud rate on your serial terminal/monitor. i.e. Most of the examples use 115200/8-N-1. If you don't use the same Baud as the code is sending you may get no output or garbage output.

Re:
> Test with another (new) ESP8266 (nodemcu module): serial monitor also doesn't show anything

The example code _should_ produce at least some output via the USB serial port.
e.g.
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L170 & https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDemo/IRrecvDemo.ino#L40

wakandan, 2018-03-09:
turns out my wires are broken >_<, it's not a problem with the library. Lesson learned. Thanks @crankyoldgit

closed 2018-03-09 by wakandan

crankyoldgit, 2018-03-09:
Thanks for letting us know.

jmoellers, 2020-04-24:
<redface>This was veeery good! I had been frustrated about getting no response and this discussion got me checking the adapter which I use to make my ESP-01s more breadboard-friendly and ... lo and behold ... the connection for pin 1 was not good!