crankyoldgit/IRremoteESP8266 #428: *noob* External IP error

maxjosh, 2018-03-04:
Hi
I have reached till config screen of IR Blaster. I have an ADSL modem linked to my wifi router with internal IP 192.168.0.1, which I put as the static gateway address. Static IP of IR device- 192.168.0.67
The serial monitor shows the following issues-

_Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 9 ms
Warning, user_id appears to be in the wrong format, security check will most likely fail. Should start with amzn1.account.***
Epoch time from timeServer is unexpectedly old, probably failed connection to the time server. Check your network settings
14400_

If this is incorrect than what static IP should I put? 
I understand the part about the user_id format, but unsure why the external IP is not connecting.
Thanks for reading..

crankyoldgit, 2018-03-04:
Are you sure you have the right project? None of the code in this library
has anything to do with Amazon or external IP addresses.

crankyoldgit, 2018-03-04:
Looks like you're using https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster, please report your issue with them, not with us.

closed 2018-03-04 by crankyoldgit