crankyoldgit/IRremoteESP8266 #429: fatal error:gpio.h is missing

jay7583, 2018-03-09:
Hi,
When i was tried to compile example code i'm getting this error.
 /home/jay/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:10:20: fatal error: gpio.h: No such file or directory compilation terminated.

do i have to add gpio.h file?

crankyoldgit, 2018-03-09:
gpio.h is included as part of the [ESP8266 Arduino board support](https://github.com/esp8266/Arduino)
Ensure you've installed it, and have selected a ESP8266 board type in your Arduino IDE.

crankyoldgit, 2018-03-16:
Just following up, did you fix your problem?

crankyoldgit, 2018-03-29:
Closing this due to no response. (and likely user environment problem)

closed 2018-03-29 by crankyoldgit

Andr1961, 2019-04-08:
Hi,
ESP32 1.0.1, ide 1.8.8 
compilation error,
IRremoteESP8266-master\src\IRrecv.cpp:11:18: fatal error: gpio.h: No such file or directory
gpio.h is present

crankyoldgit, 2019-04-08:
> ESP32 1.0.1, ide 1.8.8

@Andr1961 Are you trying to compile this for an ESP32 board? If so, that's probably your problem. This is only for **_ESP8266_** boards & chips. You need to select a board of that type and install the ESP8266 core library from the Board Manager. (See: https://github.com/esp8266/Arduino#installing-with-boards-manager)

Andr1961, 2019-04-08:
yes, ESP32, I want to transfer the esp8266 project
on esp32,
it is a pity, it is necessary to hiccup replacement.

crankyoldgit, 2019-06-19:
@Andr1961 As of [v2.6.2](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.6.2) this library now supports the ESP32.