crankyoldgit/IRremoteESP8266 #430: ESP Crashing after upgrading to latest version 2.3

arihantdaga, 2018-03-13:
### Version/revison of the library used
2.3.3

### Expected behavior

Version 1 of library was working fine. Today i updated to v 2.3.3 and i checked ir firing and my esp is crashing everytime it fires a code (right after firing the code i think).
I am firing this with AsyncWebserver and not in the main loop. So i fire IR Code in the callback function after recieving http request. 
I know that there is a possibility that esp crashes if delay() is very long. But it was working fine in previous version(Version 1 i guess, in library.json no version is mentioned). 
And i also added this code for safety
```

ESP.wdtDisable();
irsend.sendLG((unsigned long)actCode,length);
ESP.wdtFeed();
ESP.wdtEnable(5000);
```

Can anybody point out to me whats wrong here.. ?

crankyoldgit, 2018-03-13:
From the readme.md for Async Webserver:

"You can not use yield or delay or any function that uses them inside the
callbacks"

As of V2, there are delay() calls in most protocols send() routines.

Depending on what you're trying to do, you've got a few workarounds. Either
change the LG end of message gap to zero, or change the threshold in
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L155
to stop delay() calls.

The better approach is to not do the send in the callback. Eg. Set a flag
or add to a queue for the send request, and do the send(s) in the main loop.

crankyoldgit, 2018-03-14:
Now that I can see the stack trace, this more like a language use error/problem.
i.e. A google search of "Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address" indicates some issues with users using references to objects that no-longer exist. i.e. They (object) were on the stack, and then destroyed, then subsequently tried to be used.

I can't suggest anything more without seeing the real code etc.

TL;DR: I think you've got a coding problem based on that stack trace, not a WDT error etc. The delay thing _might_ be a red-herring. Try reducing your code to as simple as possible that still causes the issue, and post that code here. That process may help you see the cause of the error along the way.

crankyoldgit, 2018-03-16:
How are you going with this?

arihantdaga, 2018-03-17:
@crankyoldgit Thank you for explaining this.. I dint get a chance to work on it again, so for now i just rolled back to the previous version, But i'll check this soon..

crankyoldgit, 2018-04-02:
Friendly ping. How did you go?

arihantdaga, 2018-04-03:
Still caught up in other things. Will get back to it as soon as possible.

arihantdaga, 2018-05-06:
> @arihantdaga <https://github.com/arihantdaga> Can you please try out the
> new branch
> https://github.com/markszabo/IRremoteESP8266/tree/asyncweb-support and
> let me know if it fixes your issue? I've just added what I think will
> support what you want to do w.r.t. AsyncWebserver etc.
>
> Per the new comments in IRremoteESP8266.h you'll need to set PREFER_DELAY
> to false for your situation.

Nice.. Thank you.

crankyoldgit, 2018-05-07:
Did the changes I made work for you?

arihantdaga, 2018-05-07:
Yeah. Working great.

crankyoldgit, 2018-05-07:
Excellent. I'll mark this closed when the PR is merged in.

crankyoldgit, 2018-05-07:
Merged in to the master branch. Closing this now.

closed 2018-05-07 by crankyoldgit