crankyoldgit/IRremoteESP8266 #431: Fujitsu AC comes in as Unknow

B0R1K, 2018-03-14:
46E4C1DD:UNKNOWN:75
Data
    46E4C1DD
Type
    UNKNOWN
Length
    75
Address
    0x0
Raw
3276,1606,416,426,384,428,384,1214,414,426,384,1214,412,428,384,428,384,428,384,1214,414,1212,414,428,384,428,384,428,384,1214,412,1212,412,428,384,428,384,428,384,428,384,428,382,430,384,428,384,428,384,428,384,428,384,428,384,428,384,430,382,1214,412,428,384,428,384,428,382,430,384,428,384,402,410,428,384,1242,386,428,384,428,384,428,384,430,382,1214,412,1214,412,1212,414,1212,412,1214,412,1214,412,1214,412,1212,412,428,384,428,384,1212,414,402,410,428,384,428,384,430,382,402,410,402,410,402,410,430,382,1214,412,1212,412,404,408,402,410,1214,410,430,384,402,410,402,410,1214,412,402,410,1214,412,1212,414,1212,412

And Sending as Raw Doesn't work

crankyoldgit, 2018-03-14:
What a/c unit and remote model is it from?

crankyoldgit, 2018-03-14:
Also, please paste the entire capture text (as is) from IRrecvDumpV2 inside a [Code Block](https://help.github.com/articles/creating-and-highlighting-code-blocks/).
It includes useful other information, like library version etc. and gives us details to help emulate your problem(s).

Also:
> Sending as Raw Doesn't work

Isn't as helpful as including the full `sendRaw()` call and data you are using. Please include that too.

B0R1K, 2018-03-14:
My Head Unit Model is: ASU24RLXQ, 
i will get full Dump later today...

Thank you for your help.

crankyoldgit, 2018-03-16:
Friendly reminder

B0R1K, 2018-03-17:
I got another Board so i can develop and test at the same time 
and i replace a receiver with a new one and latest IRrecvDumpV2 

and now i am think it working, at least the receiving part: 

Timestamp : 000275.030
Encoding  : FUJITSU_AC
Code      : 1463001010FE0930B1032400000028D0 (128 bits)
Mesg Desc.: Power: On, Mode: 3 (FAN), Temp: 27C, Fan: 4 (QUIET), Swing: Horiz, Command: N/A
Library   : v2.3.3

crankyoldgit, 2018-03-17:
Awesome. You might want to check it works with the "short" commands as well. Fuji A/C's have two lengths of messages. Short ones, like power on/off, tilt the fan vanes up/down etc. and long ones like "set temp to 25c" "switch to cool mode" etc.

Try both, but if they both work, I'll add your model to the list of working ones.

crankyoldgit, 2018-03-29:
This issue seems resolved. Marking it closed. If you disagree, please reply/re-open etc.

closed 2018-03-29 by crankyoldgit