crankyoldgit/IRremoteESP8266 #436: How to compile examples i want to use

xfce, 2018-03-24:
hi all,
I am a newbie, but i read all the Issues, i can't find how to compile the example i want to use.
i use the ATOM editor , use platformio, when i import the whole dir ,and use F7, I can build the project, and in the end got  bin and elf files named firmware in .pioenvs dir, but it seems it  allways compile the IRrecvDumpV2 example, if i want to compile other example, how can i do? thanks

crankyoldgit, 2018-03-24:
This is more an ATOM/PlatformIO question than for this library. I'm by far, not a platformio expert at all.

However, here are at least a couple of options for you that should work:

**1)** Change the [src_dir](http://docs.platformio.org/en/latest/projectconf/section_platformio.html#src-dir) definition in _platformio.ini_ file in the top directory.
e.g. Currently it is `src_dir = examples/IRrecvDumpV2`, thus it defaults to making the IRrecvDumpV2 example. Change the directory to the example code you want to compile. Think of this as the _default_ program for the entire library it will compile/build/upload etc.

**or:** _(This is what I personally have done so I can compile the various examples easily.)_

**2)** Create a new PlatfomrIO project for the specific example directory in Atom/PlatformIO. Each example directory has it's own _platformio.ini_ file that should work for it.
e.g. ATOM -> "File" -> "Add Project Folder...", then navigate to the example directory you want (it should have a platformio.ini file in it) e.g. "_<directory where you stored the unzip'ed/git clone copy of this library>_/examples/IRsendDemo/"
Once that has been added, you should click on the new project in Atom/PlatformIO, open the example _ino_ file and then click the _Build_ icon etc., and it should build that example for you.

xfce, 2018-03-24:
@crankyoldgit Thanks, and it works ^-^

closed 2018-03-24 by xfce