crankyoldgit/IRremoteESP8266 #440: Help setting up Air Conditioner controller with Web

SubmissiveMC, 2018-03-27:
Hello! I'm trying to setup my ESP8266 to take commands from my Google Home (Turn aircon on, Turn aircon off, Set aircon temperature to 20 degrees etc.) and obviously make the aircon take the appropriate action. I have an ESP8266 setup with my Dyson fan which uses the ESP8266-HTTP-IR-Blaster software with IFTTT to trigger webhooks when the google assistant receives a request. I would like to setup something similar for my air conditioner, or at least a solution that works with my google home, however, I have very limited experience. Could you point me in the direction of a guide or multiple guides that could assist me in getting something like this setup? I've collected some remote codes from IRrecvDumpV2 and all the signals were recognised for my Fujitsu air conditioner. 

Thanks so much!
Kind Regards.

crankyoldgit, 2018-03-27:
I do something similar but with [Home Assistant](https://www.home-assistant.io/) using the IRMQTTServer example code in this library, and the [Google Assistant component](https://www.home-assistant.io/components/google_assistant/)

My Home Assistant config is available [here](https://github.com/crankyoldgit/Home-AssistantConfig).

crankyoldgit, 2018-03-28:
Another (less secure) alternative is to make the web interface of the [IRMQTTServer.ino](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino) program publicly available on the internet, and have Google Home do a IFTTT web request to the correct url (see[ examples](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L46) in the IRMQTTServer code), and you can make it send what you want.

SubmissiveMC, 2018-03-30:
Hello! Thanks so much for your reply. With a lot of effort, I have setup home assistant and I'm using its MqttBroker, with my ESP8266 successfully connected to it, according to it's webpage. On the ESP8266's webpage, I am able to send fujitsu codes to the airconditioner and I can successfully turn it on and off, however, It does not work through HomeAssistant. 
Home assistant Config:
`switch:
  - platform: mqtt
    name: "Fujitsu Aircon"
    command_topic: "ir_server/send"
    payload_on: "33,0x1463001010FC0830210101000000AD"
    payload_off: "33,0x146300101002"
    optimistic: true
    qos: 0
    retain: false`
(I believe the fujitsu code is 33, I think the issue has something to do with this in the ESP8266 code, however, I have no idea what it means.)
"Ensure you zero-pad to the correct number of digits for the bit/byte size you want to send as some A/C units have units have different sized messages. e.g. Fujitsu A/C units."

The off code is 48 bits, and the on code is 120 bits.

Thankyou! :)

crankyoldgit, 2018-03-30:
For your _on_ code, it's padded to the full 120 bits, ditto for the _off_ code (48 bits) so that shouldn't be an issue.

Hmm ... it should convert that "payload_on" eventually to a `irsend.sendFujitsuAC(state, 15);` call.

If you monitor the `ir_server/sent` MQTT topic. It should respond when it saw the request in the `ir_server/send` topic and other potentially useful info.
IIRC it should also send stuff to the serial port too.

Can you try monitoring that topic on your MQTT server to see if gets an appropriate message and/or monitor the serial port? (e.g. `mosquito_sub -h ip_address_of_the_MQTT_broker -t irserver/sent`. Note: sent, not send)
This will confirm if the ESP8266 is talking to the MQTT server correctly and is hearing the MQTT requests, and if it is acting accordingly. That'll help me track down where the problem might be.

I'll be upfront, I haven't tested the Fujitsu AC part of the IRMQTTServer code. It's a bit more special than most and I probably should do that. I'll try to set something up to test it soon.

Oh, something else to try (long shot), ditch the leading "0x" for the hex code. It should work the same with or without it.

crankyoldgit, 2018-03-30:
FYI, I just tested the IRMQTTServer code by sending your suggested "on" message.
e.g.
`mosquito_pub -h my_mqtt_broker_address -t 'ir_server/send' -m '33,0x1463001010FC0830210101000000AD'

I got an answer back on the 'ir_server/sent' topic (i.e. "33,0x1463001010FC0830210101000000AD" using the previous comment's command I suggested) **and** I had a NodeMCU board running IRrecvDumpV2, and it detected the IR message and decoded it to the expected protocol & number etc.

So, I'm fairly confident the IRMQTTServer code is working, thus I think you've got something odd in your MQTT settings etc.

Note: You can always listen to the `sent` topic as well. If you see the message there, then you know home assistant is delivering the message and the problem must be on the IRMQTTServer settings.

crankyoldgit, 2018-03-30:
Oh, to answer your "33" point. That's just the protocol number used internally in IRremoteESP8266. i.e. `FUJITSU_AC` equals 33 in the library source code. It's got nothing to do with anything else. Basically, Fujitsu was about the 33rd protocol added to the library.

SubmissiveMC, 2018-04-04:
Sorry for the late response, this has been a very long and very interesting learning experience haha! The problem seemed to be using Homeassistant’s embedded MQTT server, when I switched to Mosquitto, everything started working! Since then, I moved on from my Windows server and purchased a raspberry Pi 3 model B, running hassbian, and just finalised adding google assistant component. It works everytime turning off the aircon, but struggles turning it on, requiring me to repeat myself 3/4 times to get it to work. I feel like this might have to do with the aircon needing the see the Infrared 3 times before it turns on, would this be right? And how would I go about doing this with homeassistant? 

Thanks so much for your help so far, so close now haha :)

crankyoldgit, 2018-04-04:
Here is how I do sequences on a single switch, which Google Assistant works with:

https://github.com/crankyoldgit/Home-AssistantConfig/blob/master/configuration.yaml#L138

https://github.com/crankyoldgit/Home-AssistantConfig/blob/master/scripts.yaml#L25

crankyoldgit, 2018-05-07:
Friendly ping.

crankyoldgit, 2018-05-12:
Marking closed due to no response in over a month. Feel free to re-open if you need.

closed 2018-05-12 by crankyoldgit