crankyoldgit/IRremoteESP8266 #441: Turning ON/OFF a Mitsubishi projector

hamen, 2018-04-01:
Hi all,

I'm trying to replicate the ON/OFF feature from my remote to my NodeMCU.

### Version/revison of the library used
v2.3.3

### Expected behavior
Being able to transmit the data payload previously received to toggle ON/OFF my projector.
I used `IRrecvDumpV2` to receive the ON signal. Here are a couple of dumps:

```
Timestamp : 000047.772
Encoding  : UNKNOWN
Code      : 96A1512F (38 bits)
Library   : v2.3.3

Raw Timing[75]:
   +  8402, -  4172,    +   554, -   490,    +   562, -   484,    +   560, -   514,
   +   532, -   512,    +   534, -  1566,    +   526, -  1572,    +   526, -  1542,
   +   560, -  1568,    +   532, -  4192,    +   534, -  1564,    +   532, -   484,
   +   560, -   512,    +   532, -   512,    +   532, -   514,    +   530, -   514,
   +   526, -  1570,    +   524, -   520,    +   526, - 28506,    +  8454, -  4170,
   +   560, -   514,    +   528, -   516,    +   526, -   520,    +   524, -   490,
   +   556, -  1572,    +   534, -  1534,    +   560, -  1568,    +   530, -  1538,
   +   558, -  4166,    +   560, -  1538,    +   558, -   490,    +   560, -   512,
   +   530, -   514,    +   532, -   484,    +   558, -   514,    +   532, -  1566,
   +   530, -   486,    +   554

uint16_t rawData[75] = {8402, 4172,  554, 490,  562, 484,  560, 514,  532, 512,  534, 1566,  526, 1572,  526, 1542,  560, 1568,  532, 4192,  534, 1564,  532, 484,  560, 512,  532, 512,  532, 514,  530, 514,  526, 1570,  524, 520,  526, 28506,  8454, 4170,  560, 514,  528, 516,  526, 520,  524, 490, 556, 1572,  534, 1534,  560, 1568,  530, 1538,  558, 4166,  560, 1538,  558, 490,  560, 512,  530, 514,  532, 484,  558, 514,  532, 1566,  530, 486, 554};  // UNKNOWN 96A1512F

Timestamp : 000097.106
Encoding  : UNKNOWN
Code      : 96A1512F (38 bits)
Library   : v2.3.3

Raw Timing[75]:
   +  8402, -  4202,    +   530, -   516,    +   524, -   492,    +   558, -   516,
   +   528, -   488,    +   562, -  1536,    +   560, -  1568,    +   524, -  1542,
   +   556, -  1542,    +   554, -  4170,    +   564, -  1536,    +   558, -   488,
   +   562, -   482,    +   560, -   514,    +   528, -   516,    +   532, -   482,
   +   562, -  1538,    +   556, -   518,    +   524, - 28528,    +  8432, -  4202,
   +   528, -   516,    +   532, -   484,    +   562, -   482,    +   556, -   488,
   +   560, -  1540,    +   562, -  1534,    +   556, -  1542,    +   560, -  1540,
   +   560, -  4192,    +   530, -  1570,    +   534, -   512,    +   532, -   484,
   +   556, -   488,    +   558, -   516,    +   528, -   488,    +   562, -  1536,
   +   562, -   512,    +   530

uint16_t rawData[75] = {8402, 4202,  530, 516,  524, 492,  558, 516,  528, 488,  562, 1536,  560, 1568,  524, 1542,  556, 1542,  554, 4170,  564, 1536,  558, 488,  562, 482,  560, 514,  528, 516,  532, 482,  562, 1538,  556, 518,  524, 28528,  8432, 4202,  528, 516,  532, 484,  562, 482,  556, 488, 560, 1540,  562, 1534,  556, 1542,  560, 1540,  560, 4192,  530, 1570,  534, 512,  532, 484,  556, 488,  558, 516,  528, 488,  562, 1536,  562, 512, 530};  // UNKNOWN 96A1512F
```

They all look a bit different, depending how long I press the button, I think, but the `Code` stays the same.

Then I tried to send the raw data with:

```
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_LED 4 

IRsend irsend(IR_LED);

uint16_t rawData[75] = {8402, 4202,  530, 516,  524, 492,  558, 516,  528, 488,  562, 1536,  560, 1568,  524, 1542,  556, 1542,  554, 4170,  564, 1536,  558, 488,  562, 482,  560, 514,  528, 516,  532, 482,  562, 1538,  556, 518,  524, 28528,  8432, 4202,  528, 516,  532, 484,  562, 482,  556, 488, 560, 1540,  562, 1534,  556, 1542,  560, 1540,  560, 4192,  530, 1570,  534, 512,  532, 484,  556, 488,  558, 516,  528, 488,  562, 1536,  562, 512, 530};  // UNKNOWN 96A1512F

void setup() {
  irsend.begin();
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {
  Serial.println("Sending a rawData capture from IRrecvDumpV2");
  irsend.sendRaw(rawData, 75, 38);
  delay(2000);
}
```

#### Circuit diagram and hardware used (if applicable)
[image]

The circuit looks OK, I checked the transmitter with my phone camera and it _blicks_ when the output data are transmitted.

Sending the data, I get the `Sending a rawData capture from IRrecvDumpV2` message on the console and than the LED blinks (on camera), so I believe that the board is actually working as expected. Unfortunately the projector does not turn ON 😞 

I tried different frequency than 38, i.e. , 36, 40 and other possible frequencies I read about in other closed issues.

I have also tried to use `AutoAnalyseRasData.sh`, but seems I can't find a way to feed it the raw data. I keep getting `sed: 1: "s/^.*uint.*{//i": bad flag in substitute command: 'i'` error 🤔.

Any help? I know this is not a A/C, but it would be cool to have some automation around my entertainment system 😄 

Thank you 😄

crankyoldgit, 2018-04-01:
First things first, can you please detail what model number the projector and the remote are?

Second, thanks for the detail report. It helps a lot!

That certainly is an unknown protocol alright. It's interesting, a normal "header" then 8 bits, then a header gap, then another 8 bits, then a large gap (28500 usecs) then a repeat of the earlier bits.

Not sure what is wrong with your version of `sed` on your system, but it works for me (re: AutoAnalyse etc)
Here is the output for the first sample in your issue:
```
Potential Mark Candidates (using a range of 200):
8454
562

Potential Space Candidates (using a range of 200):
28506
4192
1572
520

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK = 8454
BIT_MARK = 562
SPACE_GAP1 = 28506
HDR_SPACE = 4192
ONE_SPACE = 1572
ZERO_SPACE = 520

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+00001111
  Bits: 8
  Hex:  0xF (MSB first)
        0xF0 (LSB first)
  Dec:  15 (MSB first)
        240 (LSB first)
  Bin:  00001111 (MSB first)
        11110000 (LSB first)
UNEXPECTED->HDR_SPACE+10000010 GAP(28506)
  Bits: 8
  Hex:  0x82 (MSB first)
        0x41 (LSB first)
  Dec:  130 (MSB first)
        65 (LSB first)
  Bin:  10000010 (MSB first)
        01000001 (LSB first)
HDR_MARK+HDR_SPACE+00001111
  Bits: 8
  Hex:  0xF (MSB first)
        0xF0 (LSB first)
  Dec:  15 (MSB first)
        240 (LSB first)
  Bin:  00001111 (MSB first)
        11110000 (LSB first)
UNEXPECTED->HDR_SPACE+10000010
  Bits: 8
  Hex:  0x82 (MSB first)
        0x41 (LSB first)
  Dec:  130 (MSB first)
        65 (LSB first)
  Bin:  10000010 (MSB first)
        01000001 (LSB first)
  Note: Value is the same as the last one. Could be a repeated message.
```

I should be able to whip up a sender and a decoder for this protocol. It looks fairly simple.

But, on to your issue with using sendRaw(). As far as I can tell you seem to be using it correctly. So, in theory it should work.

It's probably a odd choice of words, but you said the LED _blinks_ (via a camera) when it sends the message. I hope you mean _flashes_ instead. The default state of the LED should be off. It should only be lit very briefly when it's sending the message. If it's on most of the time, then that's wrong. You probably want to either invert the LED via a circuit, or use the invert option in the irsend() object creation. (see source code).

Have you be able to get your "send" setup to work with other IR devices? e.g. TV, stereo, etc?
i.e. Are we sure it's in a _working_ state for other devices.

crankyoldgit, 2018-04-02:
Oh FYI, try a frequency of '33' or '33000'. The normal Mitsubishi protocol uses 33kHz looking at the source code.

hamen, 2018-04-02:
Hi David and thank you for the quick reply.

The projector is a Mitsubishi HC3000 and I'm happy to say that I was able to turn it ON and OFF thanks to your help!!! 🎉 

```
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_LED 4  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(IR_LED);

uint16_t ON[75] = {8402, 4202,  530, 516,  524, 492,  558, 516,  528, 488,  562, 1536,  560, 1568,  524, 1542,  556, 1542,  554, 4170,  564, 1536,  558, 488,  562, 482,  560, 514,  528, 516,  532, 482,  562, 1538,  556, 518,  524, 28528,  8432, 4202,  528, 516,  532, 484,  562, 482,  556, 488, 560, 1540,  562, 1534,  556, 1542,  560, 1540,  560, 4192,  530, 1570,  534, 512,  532, 484,  556, 488,  558, 516,  528, 488,  562, 1536,  562, 512, 530};  // UNKNOWN 96A1512F
uint16_t OFF[75] = {8448, 4204,  534, 482,  556, 516,  528, 518,  530, 516,  528, 1540,  560, 1568,  530, 1568,  534, 1536,  562, 4192,  526, 518,  534, 1564,  526, 520,  526, 518,  528, 518,  534, 510,  528, 1572,  532, 482,  560, 28524,  8432, 4202,  530, 514,  524, 520,  530, 486,  560, 484,  560, 1538,  560, 1540,  562, 1566,  530, 1540,  560, 4164,  560, 514,  532, 1566,  532, 482,  556, 488,  558, 516,  530, 486,  556, 1540,  560, 486,  558};  // UNKNOWN 14EA3E2B

void setup() {
  irsend.begin();
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {
  Serial.println("Sending a rawData capture from IRrecvDumpV2");
  irsend.sendRaw(OFF, 75, 33000);
  delay(2000);
}
````

The crucial part was the frequency: `33000` is the working one!
One HW note, the board has to be closer than the remote to work. Probably it's the circuit that needs more power. I really can't say.
One SW note, to properly turn the projector OFF, we must send the OFF command twice, same as the need to click twice the button with the remote!

I can finally move on with the rest of the project. I'm planning to write a blog post, so expect to be mentioned on Medium 😉

PS
I tried to use `AutoAnalyseRasData.sh` again. I installed a different `sed` using `brew install gnu-sed` and replacing the `sed` in the script with the newly installed `gsed`, but I still get errors. How should I use it?

Like this?
```
./AutoAnalyseRasData.sh 8402, 4202,  530, 516,  524, 492,  558, 516,  528, 488,  562, 1536,  560, 1568,  524, 1542,  556, 1542,  554, 4170,  564, 1536,  558, 488,  562, 482,  560, 514,  528, 516,  532, 482,  562, 1538,  556, 518,  524, 28528,  8432, 4202,  528, 516,  532, 484,  562, 482,  556, 488, 560, 1540,  562, 1534,  556, 1542,  560, 1540,  560, 4192,  530, 1570,  534, 512,  532, 484,  556, 488,  558, 516,  528, 488,  562, 1536,  562, 512, 530
```

This?
```
./AutoAnalyseRasData.sh uint16_t OFF[75] = {8448, 4204,  534, 482,  556, 516,  528, 518,  530, 516,  528, 1540,  560, 1568,  530, 1568,  534, 1536,  562, 4192,  526, 518,  534, 1564,  526, 520,  526, 518,  528, 518,  534, 510,  528, 1572,  532, 482,  560, 28524,  8432, 4202,  530, 514,  524, 520,  530, 486,  560, 484,  560, 1538,  560, 1540,  562, 1566,  530, 1540,  560, 4164,  560, 514,  532, 1566,  532, 482,  556, 488,  558, 516,  530, 486,  556, 1540,  560, 486,  558};
```

Just `./AutoAnalyseRasData` and then pasting in `uint16_t OFF[75] = {8448, 4204,  534, 482,  556, 516,  528, 518,  530, 516,  528, 1540,  560, 1568,  530, 1568,  534, 1536,  562, 4192,  526, 518,  534, 1564,  526, 520,  526, 518,  528, 518,  534, 510,  528, 1572,  532, 482,  560, 28524,  8432, 4202,  530, 514,  524, 520,  530, 486,  560, 484,  560, 1538,  560, 1540,  562, 1566,  530, 1540,  560, 4164,  560, 514,  532, 1566,  532, 482,  556, 488,  558, 516,  530, 486,  556, 1540,  560, 486,  558};`?

I would love to make it work on OSX and add some usage notes to the README 😄

crankyoldgit, 2018-04-02:
Glad it's working for you. Thanks for the update.

Re HW note:
Yes, a lot of that comes down to the hardware involved. Not all IR LEDs & circuits are created equal unfortunately. YMMV.

If you get a chance, please try out/test the new [mitsubishi_projector](https://github.com/markszabo/IRremoteESP8266/tree/mitsubishi_projector) branch. It (IRrecvDumpV2) should now support decoding this protocol, and you can use `sendMitsubishi2(0xF82);` instead of sendRaw() etc.

From the [usage](https://github.com/markszabo/IRremoteESP8266/blob/master/tools/AutoAnalyseRawData.sh#L104) section of the `AutoAnalyseRawData.sh` shell script:
```
  Reads an IRremoteESP8266 rawData declaration from STDIN and tries to
  analyse it.
```

Thus just running `./AutoAnalyseRawData.sh` and paste the `uint16_t rawData....};` line in, then hit control-d (to end the STDIN) should do it.

or:

Put the entire rawData line into a file with your favourite text editor. e.g. `vi /tmp/rawdata`
Then `./AutoAnalyseRawData.sh < /tmp/rawdata` should do the trick.

or:

```
echo "uint16_t rawData[75] = {8402, 4172,  554, 490,  562, 484,  560, 514,  532, 512,  534, 1566,  526, 1572,  526, 1542,  560, 1568,  532, 4192,  534, 1564,  532, 484,  560, 512,  532, 512,  532, 514,  530, 514,  526, 1570,  524, 520,  526, 28506,  8454, 4170,  560, 514,  528, 516,  526, 520,  524, 490, 556, 1572,  534, 1534,  560, 1568,  530, 1538,  558, 4166,  560, 1538,  558, 490,  560, 512,  530, 514,  532, 484,  558, 514,  532, 1566,  530, 486, 554};" | ./AutoAnalyseRawData.sh
```

or:

```
echo "uint16_t rawData[75] = {8402, 4172,  554, 490,  562, 484,  560, 514,  532, 512,  534, 1566,  526, 1572,  526, 1542,  560, 1568,  532, 4192,  534, 1564,  532, 484,  560, 512,  532, 512,  532, 514,  530, 514,  526, 1570,  524, 520,  526, 28506,  8454, 4170,  560, 514,  528, 516,  526, 520,  524, 490, 556, 1572,  534, 1534,  560, 1568,  530, 1538,  558, 4166,  560, 1538,  558, 490,  560, 512,  530, 514,  532, 484,  558, 514,  532, 1566,  530, 486, 554};" > /tmp/rawdata
./AutoAnalyseRawData.sh < /tmp/rawdata
```

Sorry, I don't have an OS-X machine anymore to test that script on.

crankyoldgit, 2018-04-02:
Oh, is there a model number for the remote control too please?

hamen, 2018-04-04:
Unfortunately I can't find the remote controller model. Sorry.

I'm gonna test the new API the next weekend ☺

crankyoldgit, 2018-04-05:
No worries about the remote model number. You tried. ;-)

I'll hold off making a new release of the library till after the weekend in case we find a bug etc.

crankyoldgit, 2018-04-05:
Oh, I should add, it's now in the `master` branch. So clone/download/pull from there please when you test it.

crankyoldgit, 2018-04-09:
This is now in the new  [v2.4.0](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.4.0) release.

hamen, 2018-04-30:
@crankyoldgit the `sendMitsubishi2()` works on the first run, but I'm trying to figure out why after a while it stops working.
I'm not sure where in the _pipeline_ is the error. I'm gonna try to log more during the next dev cycle, i.e. next weekend :joy:

crankyoldgit, 2018-05-01:
Glad to hear it worked. Not glad to hear you had problems on subsequent "runs".
Can you elaborate on the situation? i.e. is it the libraries example code or something you've put together?

FYI, if it works once, it should work with subsequent invocations. The library is fairly indifferent to repeated calls.

crankyoldgit, 2018-05-12:
Friendly ping @hamen 
Can you elaborate on the situation? i.e. is it the libraries example code or something you've put together?

crankyoldgit, 2018-05-18:
A final ping to @hamen before I close the issue due to inactivity.

crankyoldgit, 2018-05-22:
Closing due to inactivity.

closed 2018-05-22 by crankyoldgit