crankyoldgit/IRremoteESP8266 #443: using IRsend to send unrecognized codes

udi741, 2018-04-02:
### Version/revison of the library used
Version 0.11 August, 2009

Hello,
Is there any way to send code that is not belong to any "known" encoding?
for example:
when I used the IRrecvDumpV2 code i got this info-

`Encoding  : UNKNOWN
Code      : 80339F72 (43 bits)
Timing[86]: 
     + 950, -1100     + 950, -1100     + 950, -1050     + 950, -1050
     + 950, -1100     + 950, -1050     + 950, -1100     + 950, -1100
     + 950, -1050     + 950, -1100     +1900, -2100     + 950, -3000
     +3000, -1100     + 950, -1050     + 950, -1100     +1950, -2100
     + 950, -1100     + 950, -1100     + 950, -1050     +1950, -2100
     + 950, -1100     +1950, -1050     + 950, -1050     + 950, -1100
     + 950, -2100     + 950, -1100     + 950, -1100     + 950, -1100
     + 950, -1050     + 950, -1100     + 950, -1100     + 950, -1100
     + 950, -1100     + 950, -1050     + 950, -1100     + 950, -1100
     + 950, -1100     + 950, -1050     + 950, -1050     + 950, -1100
     + 950, -1100     +1900, -2100     + 950, -4050
uint16_t  rawData[86] = {950,1100, 950,1100, 950,1050, 950,1050, 950,1100, 950,1050, 950,1100, 950,1100, 950,1050, 950,1100, 1900,2100, 950,3000, 3000,1100, 950,1050, 950,1100, 1950,2100, 950,1100, 950,1100, 950,1050, 1950,2100, 950,1100, 1950,1050, 950,1050, 950,1100, 950,2100, 950,1100, 950,1100, 950,1100, 950,1050, 950,1100, 950,1100, 950,1100, 950,1100, 950,1050, 950,1100, 950,1100, 950,1100, 950,1050, 950,1050, 950,1100, 950,1100, 1900,2100, 950,4050 };  // UNKNOWN 80339F72

`

and as you can see there is no encoding,
i would like to just use the code i got and send it somehow.

Thanks for helping!

crankyoldgit, 2018-04-03:
Please read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code) on this very topic.

crankyoldgit, 2018-04-03:
P.S. You are using an old version of the library. I suggest you update.

closed 2018-04-03 by crankyoldgit