crankyoldgit/IRremoteESP8266 #448: problem  with samsung protocol -NOT working on TV, ( repeat problem?)

gnkarn, 2018-04-20:
found something strange on the samsung generated protocol

this is the code generated by the original remote
Encoding  : SAMSUNG
Code      : E0E040BF (32 bits)
Library   : v2.4.0

Raw Timing[135]:
   +  4582, -  4408,    +   656, -  1600,    +   630, -  1624,    +   656, -  1598, 
   +   654, -   472,    +   628, -   496,    +   630, -   496,    +   628, -   496, 
   +   630, -   496,    +   630, -  1626,    +   628, -  1626,    +   628, -  1626, 
   +   630, -   496,    +   628, -   496,    +   630, -   496,    +   628, -   496, 
   +   630, -   496,    +   628, -   498,    +   628, -  1626,    +   628, -   498, 
   +   628, -   496,    +   628, -   496,    +   630, -   496,    +   630, -   496, 
   +   628, -   498,    +   628, -  1626,    +   628, -   496,    +   628, -  1626, 
   +   628, -  1626,    +   630, -  1626,    +   628, -  1626,    +   630, -  1624, 
   +   630, -  1626,    +   628, - 46612,    +  4554, -  4434,    +   628, -  1626, 
   +   630, -  1624,    +   630, -  1626,    +   628, -   496,    +   630, -   496, 
   +   630, -   496,    +   630, -   496,    +   630, -   496,    +   630, -  1626, 
   +   630, -  1624,    +   630, -  1624,    +   630, -   494,    +   630, -   496, 
   +   628, -   496,    +   630, -   496,    +   630, -   496,    +   630, -   494, 
   +   630, -  1624,    +   654, -   470,    +   656, -   470,    +   656, -   470, 
   +   654, -   470,    +   656, -   470,    +   656, -   470,    +   656, -  1598, 
   +   656, -   470,    +   630, -  1624,    +   656, -  1598,    +   656, -  1600, 
   +   656, -  1598,    +   656, -  1598,    +   656, -  1600,    +   630

uint16_t rawData[135] = {4582, 4408,  656, 1600,  630, 1624,  656, 1598,  654, 472,  628, 496,  630, 496,  628, 496,  630, 496,  630, 1626,  628, 1626,  628, 1626,  630, 496,  628, 496,  630, 496,  628, 496,  630, 496,  628, 498,  628, 1626,  628, 498,  628, 496,  628, 496,  630, 496,  630, 496,  628, 498,  628, 1626,  628, 496,  628, 1626,  628, 1626,  630, 1626,  628, 1626,  630, 1624,  630, 1626,  628, 46612,  4554, 4434,  628, 1626,  630, 1624,  630, 1626,  628, 496,  630, 496,  630, 496,  630, 496,  630, 496,  630, 1626,  630, 1624,  630, 1624,  630, 494,  630, 496,  628, 496,  630, 496,  630, 496,  630, 494,  630, 1624,  654, 470,  656, 470,  656, 470,  654, 470,  656, 470,  656, 470,  656, 1598,  656, 470,  630, 1624,  656, 1598,  656, 1600,  656, 1598,  656, 1598,  656, 1600,  630};  // SAMSUNG E0E040BF
uint32_t address = 0x7;
uint32_t command = 0x2;
uint64_t data = 0xE0E040BF;

and this is the code generated by the library

encoding  : SAMSUNG
Code      : E0E040BF (32 bits)
Library   : v2.4.0

Raw Timing[67]:
   +  4500, -  4448,    +   578, -  1678,    +   580, -  1664,    +   556, -  1692, 
   +   554, -   574,    +   556, -   570,    +   610, -   546,    +   580, -   548, 
   +   556, -   546,    +   582, -  1668,    +   580, -  1668,    +   606, -  1642, 
   +   582, -   546,    +   580, -   546,    +   556, -   570,    +   584, -   572, 
   +   556, -   572,    +   608, -   498,    +   604, -  1642,    +   606, -   524, 
   +   578, -   546,    +   580, -   548,    +   554, -   598,    +   556, -   574, 
   +   582, -   548,    +   580, -  1644,    +   604, -   522,    +   604, -  1668, 
   +   582, -  1664,    +   582, -  1666,    +   582, -  1642,    +   634, -  1642, 
   +   580, -  1648,    +   590

uint16_t rawData[67] = {4500, 4448,  578, 1678,  580, 1664,  556, 1692,  554, 574,  556, 570,  610, 546,  580, 548,  556, 546,  582, 1668,  580, 1668,  606, 1642,  582, 546,  580, 546,  556, 570,  584, 572,  556, 572,  608, 498,  604, 1642,  606, 524,  578, 546,  580, 548,  554, 598,  556, 574,  582, 548,  580, 1644,  604, 522,  604, 1668,  582, 1664,  582, 1666,  582, 1642,  634, 1642,  580, 1648,  590};  // SAMSUNG E0E040BF
uint32_t address = 0x7;
uint32_t command = 0x2;
uint64_t data = 0xE0E040BF;

data looks the same , but not the repeat , and this generated code is not recognized by the TV .

gnkarn, 2018-04-20:
this are the differences found on pulses duration

4582 | 4408 | 656 | 1600 | 630 | 1624 | 656 | 1598
-- | -- | -- | -- | -- | -- | -- | --
654 | 472 | 628 | 496 | 630 | 496 | 628 | 496
630 | 496 | 630 | 1626 | 628 | 1626 | 628 | 1626
630 | 496 | 628 | 496 | 630 | 496 | 628 | 496
630 | 496 | 628 | 498 | 628 | 1626 | 628 | 498
628 | 496 | 628 | 496 | 630 | 496 | 630 | 496
628 | 498 | 628 | 1626 | 628 | 496 | 628 | 1626
628 | 1626 | 630 | 1626 | 628 | 1626 | 630 | 1624
630 | 1626 | 628 | 46612 | 4554 | 4434 | 628 | 1626
630 | 1624 | 630 | 1626 | 628 | 496 | 630 | 496
630 | 496 | 630 | 496 | 630 | 496 | 630 | 1626
630 | 1624 | 630 | 1624 | 630 | 494 | 630 | 496
628 | 496 | 630 | 496 | 630 | 496 | 630 | 494
630 | 1624 | 654 | 470 | 656 | 470 | 656 | 470
654 | 470 | 656 | 470 | 656 | 470 | 656 | 1598
656 | 470 | 630 | 1624 | 656 | 1598 | 656 | 1600
656 | 1598 | 656 | 1598 | 656 | 1600 | 630 | 0
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
4500 | 4448 | 578 | 1678 | 580 | 1664 | 556 | 1692
554 | 574 | 556 | 570 | 610 | 546 | 580 | 548
556 | 546 | 582 | 1668 | 580 | 1668 | 606 | 1642
582 | 546 | 580 | 546 | 556 | 570 | 584 | 572
556 | 572 | 608 | 498 | 604 | 1642 | 606 | 524
578 | 546 | 580 | 548 | 554 | 598 | 556 | 574
582 | 548 | 580 | 1644 | 604 | 522 | 604 | 1668
582 | 1664 | 582 | 1666 | 582 | 1642 | 634 | 1642
580 | 1648 | 590 | 0 | 0 | 0 | 0 | 0
  |   |   |   |   |   |   |  
  |   |   |   |   |   |   |  
82 | -40 | 78 | -78 | 50 | -40 | 100 | -94
100 | -102 | 72 | -74 | 20 | -50 | 48 | -52
74 | -50 | 48 | -42 | 48 | -42 | 22 | -16
48 | -50 | 48 | -50 | 74 | -74 | 44 | -76
74 | -76 | 20 | 0 | 24 | -16 | 22 | -26
50 | -50 | 48 | -52 | 76 | -102 | 74 | -78
46 | -50 | 48 | -18 | 24 | -26 | 24 | -42
46 | -38 | 48 | -40 | 46 | -16 | -4 | -18
50 | -22 | 38 |   |   |   |   |  

as an average , the short pulses are 50 us shorter , than the control and the long pulses are 39 us longer , than the pulses emited by the control.

crankyoldgit, 2018-04-21:
Feel free to adjust for your devices.
I have a samsung TV and it seems to work fine with it. It's also a common make & protocol, so I'd expect people to be reporting issues. So _it_ could be just your device/setup.

FYI, the first raw capture (from the orig device) has a repeat in it, try sending with a repeat perhaps?
i.e. `sendSAMSUNG(0xE0E040BF, 32, 1);` should reproduce the repeat in the original.

I'll try to recapture my remote to compare timings with what you've captured and see if we should adjust the library as a whole for Samsung etc.

FYI, most of the devices have a +/- 50-100ish usec tolerance to signal timing. So I'm not to worried/surprised by the timing differences.

gnkarn, 2018-04-21:
Yes, i see , i tested with other Samsung TV , and it works fine .
So for some reason , with this TV will have ti dig a little bit more and see what is making the difference .

crankyoldgit, 2018-05-07:
How did you go?

gnkarn, 2018-05-09:
Hi ,
It works fine the TV I'm using the application now , hopefully will end
installation this week and will report back with some pictures .

After this I may go with other Samsung  TV , same protocol but it was more
picky when I tested .

Thank you