crankyoldgit/IRremoteESP8266 #451: IR transmitting signal, but device (led sign) is not picking the signals up consistently

JBrowne031, 2018-05-04:
I have made a program using an Arduino Uno to receive IR signals from a remote that uses NEC. If I press the "1" button on my remote, the program uses an IF statement to print the button value "1" on an LCD. Once the value is displayed, the program sends the NEC value to a IR transmitter/emitter using irsend.sendNEC(0xXXXXXXX, 32) followed by a delay and the irrecv enableIRin(). My question is, how does a signal randomly get missed when it is sent to the device after it has been displayed? Is the IR emitter too close or too far away? Does the delay value have to be something larger than 1 seconds? Do I need a stronger IR emitting sensor?

I am using the IRremote library along with the liquidcrystal library. Not really any issues other than signalling not being received by the device consistently.

I expect the device to receive the same amount of signals every time, but it randomly misses signals that are being sent. This causes the LCD to read "1.23 24 10000" and the sign to read "1.2 241000". In this example the "3", " ", and "0" were received and printed to the LCD, but the sign missed the signals and caused a different printout. The whole project I am doing is entering and displaying in 1 room and sending it into another room (which uses a led sign).

I can upload code if general questions are not enough to give detailed reason for why IR signal transmitted is being randomly missed by my device.

crankyoldgit, 2018-05-04:
Because you mentioned you are using an **Arduino Uno**, please note that the [Arduino-IRremote](https://github.com/z3t0/Arduino-IRremote) library is **not** the same as this one. You probably need to ask your questions on their [issues page](https://github.com/z3t0/Arduino-IRremote/issues). This one is for ESP8266 chips only.

The libraries are similar and mostly compatible when porting code from the _Arduino-IRremote_ library to this (IRremoteESP8266) library, not the other direction sadly. We have a lot of features they can't easily do due to the memory limitations.

Without seeing your code and knowing exactly your setup, there are multiple places and reasons you might be missing the subsequent code. I can't tell from here but, for starters, yes, physical proximity can make a huge difference.
Next, typically you want to (re)enable IR capture (`enableIRin()`) as soon as you can. So probably before your post-send delay.

We reduced that problem in _our_ library by having two buffers so that the library can _immediately_ start capturing messages again after it decodes one. Thus effectively it almost always be listening for new messages. Also, the ESP8266 is a lot faster than the standard Arduino boards. ;-)

crankyoldgit, 2018-05-04:
P.S. Yes, make your code available for people to see somewhere would be very helpful to whom ever wants to help you.

Per our [Troubleshooting page](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide):

> Post your reduced code somewhere for us to look at. We are not mind readers. We work best when we can see what's going wrong. "In Code We Trust.", everything else is hearsay. Posting the code in the issue is fine. Git gist, pastebin, or git repo etc are all fine too. Bonus points if the formatting is correct.

JBrowne031, 2018-05-07:
Sorry for not including my code on the initial post. I was not able to access at the time. Please review my code and let me know what you think. It works but not 100% of the time.

#include <IRremote.h>
#include <LiquidCrystal.h>

int receiverpin = 2;
IRrecv irrecv(receiverpin);
decode_results results;
IRsend irsend;
LiquidCrystal lcd(12, 11, 5, 4, 8, 7);
int value;
int cursorRow = 0;
int cursorCol = 0;

void setup() {
  
  lcd.begin(16, 2);
  pinMode(receiverpin, INPUT);
  Serial.begin(9600);
  irrecv.enableIRIn();
  lcd.cursor();
  lcd.blink();
  lcd.clear();
  
}

void loop() {

  if(irrecv.decode(&results)){
    Serial.println(results.value, HEX);
    delay(1000);
    irrecv.enableIRIn();
    
    if (results.value == 0x3FCE817) //program button
    {
      Serial.println("program");
      lcd.setCursor(0,1); //lcd displays P on the bottom left hand corner and disappears after 1 second..
      lcd.print("P");
      delay(1000);
      lcd.clear();
      delay(1000);
      irsend.sendNEC(0x3FCE817, 32);  //3 program button sends to test IR emitter functionality..
      delay(455);
      irsend.sendNEC(0x3FCE817, 32);  //usually has 2 beaps and misses 1 send command..
      delay(455);
      irsend.sendNEC(0x3FCE817, 32);  //tried 5000 delay with serial print and had same issue..  
      delay(455);
      irrecv.enableIRIn();
      irrecv.resume();
    }
    if (results.value == 0x3FC48B7) //run button
    {
      for (int positionCounter = 0; positionCounter < 20; positionCounter++) 
      {
        // scroll position left:
        lcd.scrollDisplayRight();
        delay(150);
      }
      for (int positionCounter = 0; positionCounter < 36; positionCounter++) 
      {
        // scroll position right:
        lcd.scrollDisplayLeft();
        delay(150);
      }
      for (int positionCounter = 0; positionCounter < 16; positionCounter++)
      {
        // scroll position left:
        lcd.scrollDisplayRight();
        delay(150);
      }
[... 150 lines cut ...]
        irrecv.resume();
        cursorCol++;
      }
      if (results.value == 0x3FC10EF) //9 button
      {
        lcd.print("9");
        delay(1000);
        irsend.sendNEC(0x3FC10EF, 32);
        delay(455);
        irrecv.enableIRIn();
        irrecv.resume();
        cursorCol++;
      }
      if (results.value == 0x3FC38C7) //0 button
      {
        lcd.print("0");
        delay(1000);
        irsend.sendNEC(0x3FC38C7, 32);
        delay(455);
        irrecv.enableIRIn();
        irrecv.resume();
        cursorCol++;
      }
    }
  }

crankyoldgit, 2018-05-08:
Okay, I notice several things that are an issue/could be improved.

0) This really is using a different library to this one (IRremoteESP8266), you really need to seek help with the IRremote library folk. I can only give rough guidance. Report your issue with them at: https://github.com/z3t0/Arduino-IRremote/issues

1) You're making unnecessary redundant calls to `enableIRIn()`, you should only need to call it once, in your setup() function.

2) Move the `irrecv.resume()` as close as you can to the end of the successful decode() call. See a tip of how to do that next. This is so you reduce the window of no-capture to as little as practical.

3) From experience, printing to the serial port, especially at a low baud rate of 9600 takes ages.
e.g. `Serial.println(results.value, HEX);` will print on average about 9-10 chars. At 9600 baud, that will take about 1/100th of a second, or 10 milliseconds, or 10,000 microseconds. That's longer than the header of an NEC message and almost half as long as the official gap between NEC messages. In short, that's wasted time. Copy the result and `resume` the capture/decode, THEN start any debug messages and actions.
 
e.g.
```
  ...
  unsigned long saved_value = 0;
  if(irrecv.decode(&results)){
    saved_value = results.value;
    irrecv.resume();
    Serial.println(results.value, HEX);
    delay(1000);
    if (saved_value == 0x3FCE817) //program button
    {
    ...
```

3b) Ditto for the delay() calls. The IR receiver will be sitting idle (not listening) until you call the `resume()` method. Shorten that period where possible. In those gaps/delays, multiple NEC messages may have occurred.

4) You shouldn't need the majority of those `delay(1000)` calls after each `if` statement. i.e. you should be able to send straight away. You can probably also reduce the 445 calls to 100ms instead too. That is, 100ms after the `sendNEC()` should be an appropriate gap.

5) I'd recommend shielding the transmitting IR LED from the IR receiver module with something that will block the IR signal. e.g. opaque plastic or cardboard etc. That would saving you having to wait till after you have transmitted the received message. If you don't, then yes, you probably need to call `resume()` after you've done the sendNEC calls

6) You have a lot of duplicate code in your code. You should aim to reduce that. That's just a style nitpick. It won't affect your result.

7) Beware you are not checking the type of the decode/result value of the received message. That is, if it *is* a NEC message or perhaps a SONY one, or UNKNOWN etc.

8) I have no experience with the LCD module/library, again, it could be taking a while to process. Consider moving that to after you've re-transmitted the message and resume'd capture etc.

9) You should note that the NEC protocol has a special "repeat" message, so if when you are sending it multiple times, that's what you're trying to emulate (like holding the remote button down) then just sending the message 2+ times isn't exactly the same, but it will probably work.

Anyway, that's about all I can help you with here. As I originally said, their library and ours are different. So is the hardware. I've helped you as much as we easily can. You need to ask for more help from them, not us.

JBrowne031, 2018-05-08:
Thanks for your time and assistance! I know I have alot of work to do on the coding haha

crankyoldgit, 2018-05-12:
Good luck. Marking this issue closed, as technically, it has nothing to do with our library, and we've helped as much as we reasonably can.

closed 2018-05-12 by crankyoldgit