crankyoldgit/IRremoteESP8266 #452: WIFI AP not initialising correctly

DiggiD, 2018-05-07:
Hi, 
Trying to implement the MQTTServer example sketch (as is) on an ESP8266. 
im using library version 2.4 and the latest arduino IDE to compile and flash.
but i cant seem to get it connected...

the serial monitor spits out the following on a 10-15s repeat:
> `
ets Jan  8 2013,rst cause:2, boot mode:(3,4)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff2c20 end: 3fff3080 offset: 01a0

>>>stack>>>
3fff2dc0:  00000000 3fff48fc 4024c6c0 401004d8  
3fff2dd0:  40226bca 00000000 40240000 40106bd8  
3fff2de0:  402397dd 00000017 60000200 40239802  
3fff2df0:  054d15ec 40221e0b 3fff48fc 3fff449c  
3fff2e00:  40226ea2 3fff48fc 3fff449c 00000000  
3fff2e10:  0000000a 3fff02c0 3fff023c 3fff04ac  
3fff2e20:  00000000 40239e9b 3fff023c 3fff449c  
3fff2e30:  00000018 3fff1cfc 3fff3fe4 3fff1cfc  
3fff2e40:  4024c708 00000064 3fff2e78 402224f5  
3fff2e50:  3fff023c 3fff449c 3fff1cfc 00000000  
3fff2e60:  00000064 402224dc 3fff449c 3fff1cfc  
3fff2e70:  0fe83a2e 0000ce2a 00000000 00000000  
3fff2e80:  00000000 00000000 00000000 00000000  
3fff2e90:  00000000 00000000 00000000 00000000  
3fff2ea0:  00000000 00000000 00000000 40223853  
3fff2eb0:  3fff449c 3fff3fe4 3fff3fe4 402238a0  
3fff2ec0:  00000000 3fff02c0 3fff10b8 00000000  
3fff2ed0:  4023fbc5 00000003 00000003 3fff3fe4  
3fff2ee0:  4023fd23 00000003 00000003 4023fcd7  
3fff2ef0:  3fff1f8c 00000001 00000001 3fff3ffc  
3fff2f00:  4023fde6 00000003 3fff3fe4 00000001  
3fff2f10:  402043d5 00000000 00000000 4020f0b3  
3fff2f20:  3ffe8b6b 00000000 3fff1b14 40209ed7  
3fff2f30:  3fff1f64 00000142 00000142 4010020c  
3fff2f40:  00000001 3fff3fe4 3fff2f7c 40100690  
3fff2f50:  3ffe8b6b 4024b994 3fff1f8c 00000000  
3fff2f60:  00000001 3fff3fe4 3fff1b14 4020a05d  
3fff2f70:  00000000 00000000 00000000 00000000  
3fff2f80:  00000000 00000000 3fff2fb0 40100690  
3fff2f90:  3fffdad0 00000000 3fff1b14 3fff204c  
3fff2fa0:  3fffdad0 00000000 3fff1b14 4020a0b4  
3fff2fb0:  00000000 00000000 00000000 00000000  
3fff2fc0:  00000000 00000000 3fff3fe4 0000000f  
3fff2fd0:  00000009 0000000a 0000000a 4020136f  
3fff2fe0:  feefeffe 00000000 3fff1b14 40202f68  
3fff2ff0:  3fffdad0 00000001 3fff1f8c 4020e494  
3fff3000:  0001c200 0000001c 00000000 4020bac8  
3fff3010:  feefeffe 00000000 3fff2044 40203031  
3fff3020:  feefeffe feefeffe feefeffe feefeffe  
3fff3030:  feefeffe feefeffe feefeffe feefeffe  
3fff3040:  feefeffe feefeffe feefeffe feefeffe  
3fff3050:  feefeffe feefeffe feefeffe 3fff204c  
3fff3060:  3fffdad0 00000000 3fff2044 4020fbd0  
3fff3070:  feefeffe feefeffe 3fff2060 40100710  
<<<stack<<< 
`

thanks for your help

darshkpatel, 2018-05-07:
Did you put any delays in the sketch?
Also , is the power supply beefy enough for the esp?

crankyoldgit, 2018-05-07:
> *WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:

This seems to indicate it worked at least once to configure the WiFi settings. If you didn't, then it might be reading from previous values. If that's the case, you may need to [completely erase the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) of the ESP8266 to clear those values.

Also, what version of the library are you using? Exactly what version of the Arduno IDE? and what platform? (win10/x86 linux/etc?) What chip/module are you using? (nodemcuV1/Wemos Mini D1/etc)
We will need that info to try to reproduce this.

crankyoldgit, 2018-05-07:
Also, what version of the ESP8266 board library are you using? i.e. Look in the Board Manager.

DiggiD, 2018-05-07:
@crankyoldgit 
yes that seems likely.. the ESP-01 im using, was previously a tasmota relay.
ill try erasing the flash and trying again. in the meanwhile ill try using a new board.

im using version 2.4 of the IRremoteESP8266 / Arduino IDE v1.8 / MacOS / ESP8266-01

DiggiD, 2018-05-07:
ESP8266 Board v2.4 as well

crankyoldgit, 2018-05-07:
I haven't tested it on an ESP-01 but in theory it should work, if you disable all the serial output etc. But the sending GPIO probably won't work without on a ESP-01 board, you'll need to change it and you'll need to invert the signal if you want to drive an IR-LED off it. I really recommend a Nodemcu or Wemos board instead.

https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#can-i-use-this-library-with-an-esp-01-module

See also: https://github.com/markszabo/IRremoteESP8266/issues/7#issuecomment-360837185

DiggiD, 2018-05-07:
hmmm... i did read that before i began this project.
im using GPIO3 on the ESP-01 and ive managed to send tv codes via irsend example sketches
so fingers crossed:)

DiggiD, 2018-05-07:
i tried using a new board and that worked.. 
it showed me the captive portal- i could connect to the WLAN network..
but in the serial monitor it keeps throwing similar exceptions as before

> ets Jan  8 2013,rst cause:2, boot mode:(3,4)

>load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1

>Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

>ctx: cont 
sp: 3fff2c20 end: 3fff3080 offset: 01a0

>>>stack>>>
3fff2dc0:  00000000 3fff48fc 4024c6c0 401004d8  
3fff2dd0:  40226bca 00000000 40240000 40106bd8  
3fff2de0:  402397dd 00000017 60000200 40239802  
3fff2df0:  054d15ec 40221e0b 3fff48fc 3fff449c  
3fff2e00:  40226ea2 3fff48fc 3fff449c 00000000  
3fff2e10:  0000000a 3fff02c0 3fff023c 3fff04ac  
3fff2e20:  00000000 40239e9b 3fff023c 3fff449c  
3fff2e30:  00000018 3fff1cfc 3fff3fe4 3fff1cfc  
3fff2e40:  4024c708 00000064 3fff2e78 402224f5  
3fff2e50:  3fff023c 3fff449c 3fff1cfc 00000000  
3fff2e60:  00000064 402224dc 3fff449c 3fff1cfc  
3fff2e70:  0fe83a2e 0000ce2a 00000000 00000000  
3fff2e80:  00000000 00000000 00000000 00000000  
3fff2e90:  00000000 00000000 00000000 00000000  
3fff2ea0:  00000000 00000000 00000000 40223853  
3fff2eb0:  3fff449c 3fff3fe4 3fff3fe4 402238a0  
3fff2ec0:  00000000 3fff02c0 3fff10b8 00000000  
3fff2ed0:  4023fbc5 00000003 00000003 3fff3fe4  
3fff2ee0:  4023fd23 00000003 00000003 4023fcd7  
3fff2ef0:  3fff1f8c 00000001 00000001 3fff3ffc  
3fff2f00:  4023fde6 00000003 3fff3fe4 00000001  
3fff2f10:  402043d5 00000000 00000000 4020f0b3  
3fff2f20:  3ffe8b6b 00000000 3fff1b14 40209ed7  
3fff2f30:  3fff1f64 00000142 00000142 4010020c  
3fff2f40:  00000001 3fff3fe4 3fff2f7c 40100690  
3fff2f50:  3ffe8b6b 4024b994 3fff1f8c 00000000  
3fff2f60:  00000001 3fff3fe4 3fff1b14 4020a05d  
3fff2f70:  00000000 00000000 00000000 00000000  
3fff2f80:  00000000 00000000 3fff2fb0 40100690  
3fff2f90:  3fffdad0 00000000 3fff1b14 3fff204c  
3fff2fa0:  3fffdad0 00000000 3fff1b14 4020a0b4  
3fff2fb0:  00000000 00000000 00000000 00000000  
3fff2fc0:  00000000 00000000 3fff3fe4 0000000f  
3fff2fd0:  00000009 0000000a 0000000a 4020136f  
3fff2fe0:  feefeffe 00000000 3fff1b14 40202f68  
3fff2ff0:  3fffdad0 00000001 3fff1f8c 4020e494  
3fff3000:  0001c200 0000001c 00000000 4020bac8  
3fff3010:  feefeffe 00000000 3fff2044 40203031  
3fff3020:  feefeffe feefeffe feefeffe feefeffe  
3fff3030:  feefeffe feefeffe feefeffe feefeffe  
3fff3040:  feefeffe feefeffe feefeffe feefeffe  
3fff3050:  feefeffe feefeffe feefeffe 3fff204c  
3fff3060:  3fffdad0 00000000 3fff2044 4020fbd0  
3fff3070:  feefeffe feefeffe 3fff2060 40100710  
<<<stack<<<

crankyoldgit, 2018-05-07:
Another ESP-01 board, or a nodemcu/wemos board?

Just a long shot, have you double checked the mqtt settings? (Or tried it without the MQTT enabled? e.g. Uncomment: `#define MQTT_ENABLE`)

DiggiD, 2018-05-07:
another ESP-01  im committed to this board unfortunately- ive got waaay too many- planning on giving them away to friends as working IR blasters :)

ill try with MQTT disabled-- thanks!

DiggiD, 2018-05-08:
disabling MQTT didnt change anything-- ive got exactly the same errors.

> debug(serial.Print);
give me errors

so does
<debug(Serial.print);
debug(Serial)

also, is there an easier way of cleaning the flash thru the sketch itself? 
i keep having to go thru new ESPs everytime i get the error because it uses last known config to autoconnect to the WLAN, :p

crankyoldgit, 2018-05-08:
Yes there is. However as the code is set up, you need to call it from the web interface, which you are not getting too.

Hence why I linked this earlier: https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region
i.e.
```
esptool.py erase_flash
```

As for debug, there is a routine builtin. See: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L184

e.g.
```
debug("This message will be sent to the serial port via a Serial.printf().");
```

crankyoldgit, 2018-05-08:
Just checking, what version of the WifiManager library have you got installed?

Also, just did a search on "WifiManager ESP-01" and got this:
https://github.com/tzapu/WiFiManager/issues/361

It _may_ not work on ESP-01's as is.

crankyoldgit, 2018-05-08:
Yeah. That seems to explain it a bit. You *should* see a line on the serial console like "WiFi connected. IP address: 192.168.XX.YY" from this line
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L807 if the wifi had set up correctly.

Your logs indicate you didn't, so I think the problem is in/with WifiManager..

Try changing `  if (!wifiManager.autoConnect()) {` to `if (!wifiManager.startConfigPortal("OnDemandAP")) {`.

I got the idea from that issue & https://github.com/tzapu/WiFiManager/blob/master/examples/OnDemandConfigPortal/OnDemandConfigPortal.ino

crankyoldgit, 2018-05-08:
Also https://github.com/tzapu/WiFiManager/issues/537 indicates that the issue could be with WifiManager AND mDNS.

Things to try: remove the mdns stuff, and/or upgrade the esp8266 arduino core board library to 2.4.1 (as per comments in that issue)

crankyoldgit, 2018-05-08:
And more digging:
> *WM: Connection result:
*WM: 1

That `1` is the result of a `WiFi.waitForConnectResult()` which means `WL_NO_SSID_AVAIL`.
https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFi/src/include/wl_definitions.h#L53

I'm guessing the SSID and/or password selected/provided doesn't work or there is some other "wifi" problem.

DiggiD, 2018-05-08:
ok..
done all of the above.
erased flash
updated wifimanager and ESP8266 board to latest versions
changed to this `(!wifiManager.startConfigPortal("OnDemandAP"))`
commented out mDNS and MQTT from the sketch to focus on the wifi part of things.
still getting an error->

>*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Exception (28):
epc1=0x4020f33c epc2=0x00000000 epc3=0x00000000 excvaddr=0x05ff4dd5 depc=0x00000000

>ctx: cont 
sp: 3fff2160 end: 3fff23c0 offset: 01a0

>>>stack>>>
3fff2300:  40203eff 3fff233c 00000000 3fff13a0  
3fff2310:  05ff4dd5 00ffffff 0100a8c0 3fff1390  
3fff2320:  3fffdad0 00000000 3fff0ed4 40202e76  
3fff2330:  00000000 00000000 00000000 401071d8  
3fff2340:  c600a8c0 0000001c 00000000 feefeffe  
3fff2350:  feefeffe 00000000 3fff1388 402035c3  
3fff2360:  feefeffe feefeffe feefeffe feefeffe  
3fff2370:  feefeffe feefeffe feefeffe feefeffe  
3fff2380:  feefeffe feefeffe feefeffe feefeffe  
3fff2390:  feefeffe feefeffe feefeffe 3fff1390  
3fff23a0:  3fffdad0 00000000 3fff1388 4020d414  
3fff23b0:  feefeffe feefeffe 3fff13a0 40100710  
<<<stack<<<

but atleast we know its definitely a wifi thing now.

 >*WM: Connection result: 
*WM: 3

the `3` result does mean its connected according to WL_definitions.h

DiggiD, 2018-05-08:
yeah really wierd...

once its "connected" with the above WL result = 3, it spits out all those exceptions.

BUT, im able to connect to it on my local network-- unfortunately it shows me the WifiConfig portal only.

crankyoldgit, 2018-05-08:
Weird.
So, debug/serial-wise. Is it getting past the `setup_wifi()` and/or mdns stuff?
e.g. are you seeing the output of `  debug("WiFi connected. IP address: " + WiFi.localIP());`?

Failing all that, it might just pay to setup wifi manually the old fashioned way instead of using WifiManager.

E.g. Similar to how it is done in https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRServer/IRServer.ino

DiggiD, 2018-05-08:
tried switching back to your original statement 

`if (!wifiManager.autoConnect())`

gives me the following

>*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.198

>Exception (28):
epc1=0x4020f434 epc2=0x00000000 epc3=0x00000000 excvaddr=0x05ff4dc8 depc=0x00000000

crankyoldgit, 2018-05-08:
The fact you are not seeing that debug line I'm listing indicates to me that it is still dying in the WifiManager library/code. Perhaps trying reverting the board manager component back to 2.3.x for the esp8266 core.

Or .. try configuring the wifi manually. :-/

DiggiD, 2018-05-08:
yeah i think ill approach the manual method first..
will update

cheers @crankyoldgit :]

crankyoldgit, 2018-05-10:
That's totally weird that it works on wemos etc, and not on the ESP-01. Could be a difference in compilers between platformio and the arduino IDE.

Can you please try the following on your ESP01 and see if it still causes it to berk?:

`debug("WiFi connected. IP address: " + WiFi.localIP().toString());`

Also, did you try the firmware I uploaded?

DiggiD, 2018-05-10:
`debug("WiFi connected. IP address: " + WiFi.localIP().toString());`
works perfectly! 
now WIFI manager works great too. no errors. using startconfigPortal() instead of autoConnect() is more stable though.

`Also, did you try the firmware I uploaded?`
im not sure how to upload a .bin firmware using the arduino IDE actually, so i havent yet

crankyoldgit, 2018-05-11:
Yay!
I'll add a fix to make that permanent.

As for uploading the firmware, as you've now got the IRMQTTServer loaded on the device, you can just upload the firmware I supplied via the web/http interface on the ESP8266. i.e. Very bottom of the main page on the website running on the device. Again, I'm really interested in the result as it may be a difference from my test/build environment of platformio vs. the typical user's Arduino IDE one.

P.S. I needed to upload a *lot* when developing it, and I didn't want to continually physically have to unplug things as I use the device in my house constantly. ;-)

closed 2018-05-11 by crankyoldgit

DiggiD, 2018-05-13:
sorry for the delay @crankyoldgit 

i uploaded your firmware and it worked flawlessly. 
did you use the `.toString()` call in the platformio firmware??

crankyoldgit, 2018-05-13:
No. That was the unmodified code. i.e. before adding the `.toString()` but it was compiled under the PlatformIO IDE instead the standard Arduino IDE.
That pretty much confirms my theory that the "compile" environment is/was the root of the issue.

Anyway, we've modified the code to work around the Arduino IDE issue, and as most folk will use that, it's probably for the best.

So, thanks VERY much for testing it and allowing me to confirm at least part of the root cause.

Go forth, and make your ESP-01 modules useful!

P.S. We don't have an "official" circuit for sending (or receiving) on the ESP-01, so if you care to document what you end up with, please consider letting us know/pointing us to it so we can add it to our docs. We still get a fair few requests on how to use the ol' ESP-01 boards.

DiggiD, 2018-05-13:
thank you, for your prompt and awesome support!

my circuit is quite simple, similar to many IR blasters online.

VCC=3.3v 
receiver:
Output PIN of TSOP IR receiver is connected to GPIO-0 or 2  (without pullup resistor as TSOP pin pulls up anyways).

Sender:
To avoid problems with GPIO-0/2 stability I’m using GPIO-3 (RX port) for IR output directly connected to the base of a 2n2222 transistor. IR LED anode is connected to VCC with 10-100 Ohm current limiting resistor and cathode is connected to the collector of 2n2222 and  Emitter to GND

[image]

crankyoldgit, 2018-05-13:
Many thanks. https://github.com/markszabo/IRremoteESP8266/wiki/ESP01-Send-&-Receive-Circuit

DiggiD, 2018-05-14:
thank you!

we may have to re-open this.

noticed abnormal startup issue..

using your platformIO firmware/ or standard IRMQTTserver/arduinoIDE compile-
on power down and power up of ESP8266-01
-last set IP is retained.
-browser to IP shows WIFIManager config panel..
-wifi credentials are re entered
-esp reboots correctly and shows IRMQTTserver webpage known IP address.

should i start a new issue?

crankyoldgit, 2018-05-14:
I'm not sure I understand what you are saying. i.e. what the problem is. That sounds like how I would expect it to be. If it doesn't have a previous setup (that the firmware recognizes/is in the same spot etc) then it does the whole wifi setup process. Then reboots, which then uses the "new" config and no-longer asks the config etc.

Or am I missing something?

DiggiD, 2018-05-14:
oops never mind.. my mistake!

 i kept this statement in 
`(!wifiManager.startConfigPortal("OnDemandAP"))`

it was causing the config portal to be shown on every restart.
all good now..

DiggiD, 2018-05-17:
@crankyoldgit ummm ive noticed another small bug... 
i think it could be circuit specific to me using the pin 3 on the ESP-01.
after extensive testing with my working IRSERVERMQTT code and the updated library version, i believe ive figured out the issue.

using  `irsend.begin()` before  `Serial.begin()` causes the IR circuit to fail i.e. ir LED remains dark.

tested and retested this for a whole day with different firmwares.. including the platformio one you sent me.

crankyoldgit, 2018-05-17:
The firmware I provided/uploaded will be using pin 4 (See https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L133) so it won't transmit with an IR led on Pin 3.
I assume you've changed that on your code.

If I understand it correctly, Pin 3 on the ESP-01 is the Serial TX on the ESP8266. Thus, sending serial output is going to cause issues. Either disable it completely (e.g. turn off [DEBUG](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L157)) then you should be able to use Pin 3 fine. As we describe in the FAQs, the ESP-01 is tricky. Your options are limited.

I recommend you use the ESP8266 RX pin (which I think is Pin 2 on the ESP-01 a.k.a. GPIO1). It should be fine to use as long as nothing serial is SENT to the ESP-01. You'll need to change this line `Serial.begin(115200);` to `Serial.begin(115200,SERIAL_8N1,SERIAL_TX_ONLY);` per https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending

crankyoldgit, 2018-05-18:
@DiggiD Any chance you could provide a Fritzing diagram with the IR led part of the circuit connected to Pin 2 (GPIO1) ... assuming the above works.

Thanks in advance.

DiggiD, 2018-05-21:
i believe,
GPIO 3 is the Rx pin
GPIO 1 is the Tx pin 
?

crankyoldgit, 2018-05-21:
Yes, ignore my complete lack-of ESP-01 expertise. ;-) Basically, I think you should be using the RX pin on the ESP side, when `SERIAL_TX_ONLY` is set. Which, as far as I understand it should free up the RX pin for use.

If I'm reading it right, RX == GPIO3 == Pin 7 (according to https://circuits4you.com/wp-content/uploads/2016/12/ESP-01-Pin-Out.png)
i.e. `IR_LED = 7;`

crankyoldgit, 2018-05-21:
G'ah. Not my day.
RX == GPIO3 == D3 == Pin 7 (according to https://circuits4you.com/wp-content/uploads/2016/12/ESP-01-Pin-Out.png)
i.e. `IR_LED = 3;`  // Not 7 as above

DiggiD, 2018-05-21:
hehe all good.
so the only thing i should change is to set `SERIAL_TX_ONLY`
got it!

crankyoldgit, 2018-05-21:
Yes, and if that doesn't work, set `DEBUG` to `false`

DiggiD, 2018-05-21:
checked!
 both SERIAL_TX_ONLY and  setting  `DEBUG` to `false` work

PR #455 (2018-05-11): Ensure WiFi.localIP() is converted to a string when needed.
Due to a potential compiler difference (e.g. Platformio vs. Arduino IDE) it
seems that `WiFi.localIP()` wasn't being automatically associated with the
string varient when used in a string context. This manifest as a crash on a ESP-01 board.
Forcing the conversion should fix it in all cases.

Fixes #452

Verified by @DiggiD
commit 3c11f651 (2018-05-11): Ensure WiFi.localIP() is converted to a string when needed. (#455)

Due to a potential compiler issue (e.g. Platformio vs. Arduino IDE) it
seems that WiFi.localIP() wasn't being automatically associated with the
string varient when used in a string context.
Forcing the conversion should fix it in all cases.

Fixes #452

Checked by @DiggiD