crankyoldgit/IRremoteESP8266 #459: IRMQTTSERVER-  /sent vs /send - Payload mismatch in serial monitor and MQTT

DiggiD, 2018-05-14:
sending HITACHI AC codes using mqtt with IRMQTTSERVER compiled to ESP8266-01
works great, AC responds to commands.

this is the result im getting in the serial monitor when i send mqtt payload to /send topic: "40,0xB2AE4D51F061840000000030B8"

>0002273.587: MQTT Payload (raw): 40,0xB2AE4D51F061840000000030B8
0002273.816: Sent the IR message.
0002273.816: Type: 40
0002273.817: Code: 0x61840000000030B8
0002273.817: Bits: 0
0002273.817: Repeats: 0

using MQTT Lens subscribed to my send/sent topics i see

>/send
40,0xB2AE4D51F061840000000030B8
/sent
40,61840000000030B8,0,0

crankyoldgit, 2018-05-14:
Should be fixed in this branch: https://github.com/markszabo/IRremoteESP8266/tree/issue-459
@DiggiD Can you please test this to see if it gives the correct results in the serial output and via MQTT?

DiggiD, 2018-05-14:
checked!

>0000443.204: MQTT Payload (raw): 40,0xB2AE4D51F061840000000030B8
0000443.433: Sent the IR message.
0000443.433: Type: 40
0000443.434: Code: 
0000443.434: 0xB2AE4D51F061840000000030B8
0000443.434: Repeats: 0

>/send
40,0xB2AE4D51F061840000000030B8
/sent
40,0xB2AE4D51F061840000000030B8

:) thanks!

crankyoldgit, 2018-05-14:
Thanks for confirming. Just waiting on the code review then I'll merge it into the master branch.

commit 560372e3 (2018-05-14): BUG: Incorrect acknowledgements for some A/C protocols.
Forgot to update this section of code for other A/C protocols, so using
a standard function to keep it up-to-date in future.

Fixes #459
PR #460 (2018-05-14): BUG: IRMQTTServer: Incorrect acknowledgements for some A/C protocols.
Forgot to update this section of code for other A/C protocols, so using
a standard function to keep it up-to-date in future.

Fixes #459
commit 7c06012e (2018-05-14): BUG: Incorrect acknowledgements for some A/C protocols. (#460)

Forgot to update this section of code for other A/C protocols, so using
a standard function to keep it up-to-date in future.

Fixes #459
closed 2018-05-14 by crankyoldgit