crankyoldgit/IRremoteESP8266 #462: Feature Request (Static IP)

NewUser9, 2018-05-17:
Please add Static IP configuration. (I know we can configure this in the router but when I do so ESP does WDT Reset and continues to loop)

Regards.

crankyoldgit, 2018-05-17:
FWIW, I use static IP assignment on my router AND use the IRMQTTServer example code. It works without WDT'ing.

You may also want to download the latest "master" branch and compile/use the updated example code.
e.g. https://github.com/markszabo/IRremoteESP8266/commit/3c11f651af53e23c92160622d2b652deba85d548

crankyoldgit, 2018-05-18:
In which sketch? (I'm going to assume the IRMQTTServer one)

Most calls to the library to do a repeat use a suitable gap between sends, per protocol spec. or typical-case.

If you want a gap of a second or so, then you'll have to add that yourself or control what ever is triggering the http/mqtt requests. If you are trying to do complex automation, you either need to code it yourself or have something like Home Assistant/NodeRed etc do the higher-level stuff.

NewUser9, 2018-05-18:
Thanks for your help.

darshkpatel, 2018-05-18:
Marking it closed then :)

closed 2018-05-18 by darshkpatel