crankyoldgit/IRremoteESP8266 #467: Best way to work wth too many raw data arrays?

syalujjwal, 2018-05-23:
I have been working on a prototype for a wifi based universal remote controller and i have been using the IR-Remote library for the same. But i have a couple of questions which google couldn't help with.

So, the problem is that there are some remotes like hitachi or some other brands that aren't recognised by the library and the coding type shows "UNKNOWN". The raw dump shows an array of 211 integers like such:

Timestamp : 012012.059
Encoding  : UNKNOWN
Code      : 794B7E67 (106 bits)
Library   : v2.4.0

Raw Timing[211]:
   +  3424, -  3352,    +   378, -  1336,    +   376, -   454,    +   458, -  1254, 
   +   378, -  1334,    +   378, -   454,    +   460, -   352,    +   396, -  1336, 
   +   460, -   356,    +   394, -  1334,    +   458, -   364,    +   466, -  1256, 
   +   458, -   356,    +   392, -  1336,    +   458, -  1258,    +   456, -  1254, 
   +   378, -   452,    +   460, -   360,    +   470, -  1256,    +   458, -   298, 
   +   532, -   366,    +   466, -  1256,    +   458, -  1256,    +   458, -   358, 
   +   472, -  1256,    +   458, -  1254,    +   458, -   358,    +   472, -   304, 
   +   526, -  1258,    +   456, -   298,    +   532, -   362,    +   466, -   368, 
   +   464, -  1256,    +   456, -  1258,    +   456, -  1258,    +   456, -  1256, 
   +   378, -  1336,    +   458, -   360,    +   470, -   304,    +   524, -   364, 
   +   466, -   364,    +   464, -   310,    +   522, -  1232,    +   482, -  1256, 
   +   456, -   304,    +   526, -   310,    +   520, -   306,    +   524, -   370, 
   +   462, -  1256,    +   456, -  1260,    +   454, -  1232,    +   480, -   306, 
   +   524, -   366,    +   464, -   312,    +   516, -  1236,    +   478, -   368, 
   +   462, -   368,    +   462, -   368,    +   386, -   454,    +   454, -   372, 
   +   456, -   374,    +   458, -   372,    +   460, -   306,    +   524, -   372, 
   +   458, -   370,    +   460, -   308,    +   520, -   312,    +   444, -   452, 
   +   376, -   454,    +   452, -   308,    +   524, -   368,    +   462, -   372, 
   +   458, -   306,    +   522, -   374,    +   458, -   368,    +   460, -   306, 
   +   524, -   374,    +   458, -   308,    +   522, -   372,    +   460, -   370, 
   +   458, -   374,    +   460, -   364,    +   464, -   370,    +   460, -   304, 
   +   450, -   454,    +   452, -   308,    +   522, -   370,    +   460, -   368, 
   +   386, -   452,    +   452, -   374,    +   456, -   312,    +   520, -  1260, 
   +   452, -  1262,    +   452, -   376,    +   456, -   372,    +   458, -   376, 
   +   454, -   378,    +   380, -  1334,    +   452, -  1260,    +   452, -  1264, 
   +   378, -  1336,    +   378, -  1310,    +   476, -   320,    +   438, -   452, 
   +   378, -   428,    +   402

uint16_t rawData[211] = {3424, 3352,  378, 1336,  376, 454,  458, 1254,  378, 1334,  378, 454,  460, 352,  396, 1336,  460, 356,  394, 1334,  458, 364,  466, 1256,  458, 356,  392, 1336,  458, 1258,  456, 1254,  378, 452,  460, 360,  470, 1256,  458, 298,  532, 366,  466, 1256,  458, 1256,  458, 358,  472, 1256,  458, 1254,  458, 358,  472, 304,  526, 1258,  456, 298,  532, 362,  466, 368,  464, 1256,  456, 1258,  456, 1258,  456, 1256,  378, 1336,  458, 360,  470, 304,  524, 364,  466, 364,  464, 310,  522, 1232,  482, 1256,  456, 304,  526, 310,  520, 306,  524, 370,  462, 1256,  456, 1260,  454, 1232,  480, 306,  524, 366,  464, 312,  516, 1236,  478, 368,  462, 368,  462, 368,  386, 454,  454, 372,  456, 374,  458, 372,  460, 306,  524, 372,  458, 370,  460, 308,  520, 312,  444, 452,  376, 454,  452, 308,  524, 368,  462, 372,  458, 306,  522, 374,  458, 368,  460, 306,  524, 374,  458, 308,  522, 372,  460, 370,  458, 374,  460, 364,  464, 370,  460, 304,  450, 454,  452, 308,  522, 370,  460, 368,  386, 452,  452, 374,  456, 312,  520, 1260,  452, 1262,  452, 376,  456, 372,  458, 376,  454, 378,  380, 1334,  452, 1260,  452, 1264,  378, 1336,  378, 1310,  476, 320,  438, 452,  378, 428,  402};  // UNKNOWN 794B7E67
Now for such AC remotes which are unknown, which are more common than i thought would be, it is a matter of best implementation. I know I can save these arrays in variables and send them using .sendRaw(array, sizeof(array, freq) but then I will have to store about 15 such arrays for multiple brands. Even if I save ir codes for five different protocols, i'll be making a table of 5 * 15 = 75 huge arrays. i don't think that is a good practice nor the ESP8266 will be able to store all that.

What can be the best way to do this? What am I missing? Is there a better way to do this?

crankyoldgit, 2018-05-23:
For starters, upgrade to **v2.4.1** of the library
Per the release notes at https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.4.1: _"Support of Hitachi A/C 13 & 53 byte messages. (#461)"_ Your example message is a 13 byte hitachi. ;-) It should hopefully be detected.

The "best way" is to write a decoder and sender for each protocol. That cuts it down to a much smaller memory footprint on the ESP8266. (eg. 13 bytes, instead of ~422 bytes per message)

If you try the [tools/auto_analyse_raw_data.py](https://github.com/markszabo/IRremoteESP8266/blob/master/tools/auto_analyse_raw_data.py) utility, you can feed it the rawData line and you can ask it to try to work it out for you and give you a rough template for some code to generate the message.

The **real** _best way_ is to not store the rawData on the ESP8266 at all. e.g. Send the raw data to the ESP8266 each time it needs to send something. That is, store the collections of rawdata on a PC etc.
e.g. Look at the raw data sending capabilities of [IRMQTTServer.ino](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L59).
And do something like:
`http://<your_esp8266's_ip_address>/ir?type=30&code=38000,3424,3352,378,1336,376,454,458,1254,378,...`
_P.S. IRMQTTServer (and a few other external projects) are pretty much your desired "wifi based universal remote controller" already._

You should also realise that the ESP8266 modules come with between 0.5Meg and 4Meg of storage. The 4Meg variety being the most common now-a-days. That's a _heck_ of a lot of raw data arrays. Assuming 512 bytes per array (which should store the one you listed) you could store approx 7000+ such arrays and still have room for the code. That's without resorting to any clever compression or encoding.

syalujjwal, 2018-05-24:
Thanks @crankyoldgit , this was really helpful. I'll risk bothering you and the forum with a silly question maybe.

So i plan to make a process where in configuration i turn on the ir receiver and ask the user to press the on-off button pointing to it. From that I'll make out the coding type. Now it could be any of the over types of coding from the list. Lets just say we are working only on the AirConditioner ones here. Then after the config process I want to send an on or off signal using the recognised protocol. So for this, do I have to run a if-else/switch case for all protocols based on the recognised protocol and then send the command? Or am I missing something fairly simpler than this?

crankyoldgit, 2018-05-24:
I'm not really sure I understand what your trying to do, maybe explain it with some pseudo-code?
If you are wanting to determine what an incoming message is, then I can't think of what you can do other than `if` statements or a `switch` statement to decide what it is/what to do.

I do want to point out the FAQ item https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me as I feel we are starting to stray into that territory a bit.

crankyoldgit, 2018-06-01:
This seems done/answered and no further update so marking closed.

closed 2018-06-01 by crankyoldgit