crankyoldgit/IRremoteESP8266 #472: LG AC control

baegopooh, 2018-06-05:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revison of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._
2.4.1

### Expected behavior
_What steps did you do and what should it have done?_
Trying to contol LG AC using this library(with espeasy and esp12s, if it's relevant)
### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

LG AC not working as intended

This is new issue for old(closed) issue about LG AC timing.
https://github.com/markszabo/IRremoteESP8266/issues/285

LG HDR MARK is 9ms or 4.5ms. 

1) It's known among korean electricians that LGE used one of 2 chips for ir remote. It's UPD6121(aka "NEC") or TC9012. And UPD6121 use 9ms for HDR MARK, and TC9012 uses 4.5ms for HDR MARK.
 
2) LGE official instruction manual(althogh this is for TV, LGE shares simillar remote control chips) shows remote control protocol, clearly states HDR MARK as 9ms(on page 37).

http://www.lge.co.kr/common/download.jsp?path=/upload/PDF/GLZ/27729/&fileName=1402250953_MFL67684113_06.pdf&originFileName=1402250953_MFL67684113_06.pdf

This instruction manual also shows BIT MARKS and BIT SPACES, similar to what i posted before.

So, even if old codes work(apperantly it worked for many people), it's just becuase old code signals was in error margin, not because old code signals had correct timing.

crankyoldgit, 2018-06-05:
As stated in #285 "I'll need the full raw captures from the as-is IRrecvDumpV2.ino example code"

After a skimming over page 37 of that document, and verifying with known-good Pronto codes for "modern" LG devices, that protocol looks and appears identical to the NEC protocol. As you described:
> It's UPD6121(aka "NEC")

NEC uses a 32 bit protocol, has 9ms HDR mark pulse followed by a 4.5ms space. The manual indicates the same bit encoding, the same 32-bit code construction (e.g. the last half of the message is 8 bits of data followed by the inverted data) and it shares the same "repeat" code as far as I can tell. i.e. It' is the NEC protocol, not a LG one.

Passing "known good" pronto codes confirms this as the timings match the ones you describe, and the NEC timings. and the library correctly detects the inbound signal as a valid NEC code.

I think what may be going on here (in the bigger picture) is you are trying to send a "code" to an LG device with the `sendLG()` function because you assume(d) quite logically it would be the LG protocol. I think if you use the `sendNEC()` function(*) instead, it will just work for you, because, it is actually the NEC protocol. If you capture it with the library, I think you'll find your signal is actually an NEC message (which lots and lots of manufacturers use).

e.g.

Here is an existing pronto code for a discrete power-on for a LG TV. (I know, not the same as an LG AC but I can't find a good pronto or otherwise example out there).

```
0000 006d 0026 0000 0155 00aa 0016 0015 0016 0015 0016 0040 0016 0015 0016 0015 0016 0015 0016 0014 0016 0015 0016 0040 0016 0040 0016 0015 0016 0040 0016 0040 0016 0040 0016 0040 0016 0040 0016 0015 0016 0015 0016 0040 0016 0015 0016 0015 0016 0015 0016 0040 0016 0040 0016 0040 0016 0040 0016 0015 0016 0040 0016 0040 0016 0040 0016 0014 0016 0015 0016 060b 0155 0055 0016 0e58 0155 0055 0016 00aa
```
(Taken from http://www.remotecentral.com/cgi-bin/mboard/rc-discrete/thread.cgi?7244)

Simulating/converting the above pronto code yields:
```
$ cd tools
tools$ make all
<lots of output removed>
tools$ ./gc_decode -prontohex -raw "0000 006d 0026 0000 0155 00aa 0016 0015 0016 0015 0016 0040 0016 0015 0016 0015 0016 0015 0016 0014 0016 0015 0016 0040 0016 0040 0016 0015 0016 0040 0016 0040 0016 0040 0016 0040 0016 0040 0016 0015 0016 0015 0016 0040 0016 0015 0016 0015 0016 0015 0016 0040 0016 0040 0016 0040 0016 0040 0016 0015 0016 0040 0016 0040 0016 0040 0016 0014 0016 0015 0016 060b 0155 0055 0016 0e58 0155 0055 0016 00aa"
Code length 80
Code type      3 (NEC)
Code bits      32
Code value     0x20df23dc
Code address   0x4
Code command   0xc4
uint16_t rawbuf[76] = {
    8866, 4420, 572, 546, 572, 546, 572, 1664, 
    572, 546, 572, 546, 572, 546, 572, 520, 
    572, 546, 572, 1664, 572, 1664, 572, 546, 
    572, 1664, 572, 1664, 572, 1664, 572, 1664, 
    572, 1664, 572, 546, 572, 546, 572, 1664, 
    572, 546, 572, 546, 572, 546, 572, 1664, 
    572, 1664, 572, 1664, 572, 1664, 572, 546, 
    572, 1664, 572, 1664, 572, 1664, 572, 520, 
    572, 546, 572, 40222, 8866, 2210, 572, 95472, 
    8866, 2210, 572, 4420};
```
i.e. It's detected (as expected) as an NEC code, and has the timings you described and expected from the manual. **NOT** as code type of LG.

The 8000ms hdr mark signal is an OLDER LG protocol.

Restating the original request. Please try capturing the physical remote using the IRrecvDumpV2 code. The output (especially the rawbuf line) will be invaluable to help determine what is going on.

To send exactly what that above pronto code requested you'd use:
`sendNEC(0x20df23dc, 1);  // Send code 0x20df23dc with a single repeat code/sequence.`

(* Note: Like `sendSherwood()` looking at the pronto code(s) it looks like it always sends a "repeat" code. e.g. the final: `8866, 2210, 572, 4420};` section is a NEC repeat.)

crankyoldgit, 2018-06-11:
Ping @baegopooh for more information. i.e. raw captures.

crankyoldgit, 2018-06-19:
Another friendly ping for more information.

crankyoldgit, 2018-07-04:
@baegopooh Closing this issue due to no response. Feel free to update it and we can re-open it.

baegopooh, 2018-07-15:
It's been a while, was busy at work. and as i had no spare esp8266 to take a dump, i ordered aruino uno and used ardunino uno irremote to take dump.

So, i took a dump by 2 lg aircon remote finally(both power off signal)
https://imgur.com/a/BYRP4SO

First one(left one in my picture) is what i currently have in my room, manufactured in 2007.

Encoding  : LG
Code      : 88C0051 (28 bits)
Timing[59]: 
     +8350, -4200     + 550, -1600     + 550, - 550     + 550, - 550
     + 500, - 550     + 550, -1600     + 550, - 550     + 550, - 550
     + 500, - 550     + 550, -1650     + 500, -1650     + 550, - 500
     + 550, - 600     + 500, - 550     + 500, - 550     + 550, - 550
     + 550, - 550     + 500, - 600     + 500, - 550     + 500, - 600
     + 500, - 550     + 550, - 550     + 550, -1600     + 550, - 550
     + 500, -1650     + 550, - 550     + 500, - 550     + 550, - 550
     + 550, -1600     + 500

 Second one(right one in my picture) is what i have in my hometown, manufactured 2014
ncoding  : LG
Code      : 88C0051 (28 bits)
Timing[59]: 
     +8750, -4150     + 500, -1600     + 500, - 500     + 500, - 550
     + 500, - 500     + 550, -1500     + 550, - 500     + 500, - 550
     + 500, - 500     + 500, -1550     + 550, -1500     + 500, - 550
     + 500, - 500     + 500, - 500     + 550, - 500     + 500, - 500
     + 550, - 500     + 500, - 550     + 500, - 500     + 500, - 500
     + 550, - 500     + 550, - 500     + 500, -1550     + 500, - 500
     + 550, -1550     + 500, - 500     + 550, - 500     + 500, - 550
     + 500, -1550     + 500

Both of these remote works on LG AC(manufactured in 2007). and my simulated ir signal with 9ms HDR mark (dump as below) also works on this AC.

Encoding  : LG
Code      : 88C0051 (28 bits)
Timing[59]: 
     +8900, -4450     + 550, -1700     + 500, - 600     + 500, - 600
     + 550, - 600     + 500, -1700     + 500, - 600     + 550, - 600
     + 500, - 600     + 550, -1650     + 550, -1700     + 550, - 550
     + 500, - 600     + 550, - 600     + 550, - 550     + 500, - 600
     + 500, - 600     + 550, - 550     + 550, - 600     + 500, - 600
     + 550, - 550     + 550, - 600     + 500, -1700     + 500, - 600
     + 550, -1700     + 550, - 550     + 550, - 550     + 550, - 550
     + 550, -1700     + 500

crankyoldgit, 2018-07-16:
Thanks for the info. That makes a good case for increasing the value of `LG_HDR_MARK & `LG_HDR_SPACE`.

I am reluctant to bump it all the way to 9000 usecs, given it's working for a lot of people at 8000.
I'm thinking lets aim for half way and we _might_ get coverage for old and newer devices.

Can you try it with:
```
#define LG_HDR_MARK_TICKS            170U  // Previously was 160. Should make HDR mark of 8500.
#define LG_HDR_SPACE_TICKS            85U  // Prev. was 80. Should make HDR space of 4250
```
in https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_LG.cpp#L24

And see if it works with your old and newer AC devices?

crankyoldgit, 2018-07-16:
Do those values I proposed work for your LG 2014 A/C unit too?

(See also #488 where the LG header mark is also around 8400usec)

P.S. If you haven't already, upgrade to 2.4.2 of the library for some improved timings on sending.

baegopooh, 2018-07-17:
2014 LG A/C is at my hometown, so it will take some time before i can test it.
I thought i used 2.4.2, but will check again after work.

crankyoldgit, 2018-07-17:
You may have been using 2.4.2 now, but the original issue report said 2.4.1.

baegopooh, 2018-07-18:
I was using 2.4.1. changed to 2.4.2. and new timing, and it works also.(on 2007 LG AC). Will be back when i test my 2014 machine. Thank you everyone for timely update

crankyoldgit, 2018-07-27:
FYI, the changes are now live in the new v2.4.3 release of the library.