crankyoldgit/IRremoteESP8266 #476: encodeSony bug

xfdr0805, 2018-06-28:
uint32_t IRsend::encodeSony(uint16_t nbits, uint16_t command,
                            uint16_t address, uint16_t extended)
{
  uint32_t result = 0;
  switch (nbits)
  {
  case 12: // 5 address bits.
    result = address & 0x1F;
    break;
  case 15: // 8 address bits.
    result = address & 0xFF;
    break;
  case 20: // 5 address bits, 8 extended bits.
    result = address & 0x1F;
    **_result |= (extended & 0xFF) << 6;_**//should be 6 not 5,
    break;
  default:
    return 0; // This is not an expected Sony bit size/protocol.
  }
  result = (result << 7) | (command & 0x7F); // All sizes have 7 command bits.
  return reverseBits(result, nbits);         // sendSony uses reverse ordered bits.
}

sony dvd ir send ok

irsend.sendSony(irsend.encodeSony(20, cmd, address, 0x71), 20);

crankyoldgit, 2018-06-28:
The Sony 20-bit protocol is 7 command bits, 5 address bits, 8 extended bits.
I'm fairly sure it should be `5`.
A left shift of 6 would introduce a `0` (binary) between the `address` & `extended`
e.g.
0b1 << 5 is 0b100000 (i.e. a left shift of 5 bits, or 5 bits added.)
0b1 << 6 would be 0b1000000 (6 bits added)
We left shift by 5 so we don't over write the `5` address bits.

Or am I missing something?

crankyoldgit, 2018-07-02:
Do you concur?

xfdr0805, 2018-07-03:
Yes, I agree, but my actual test is not correct

e.x Remote Controller Next Button
used "IRrecvDumpV2" decode 
results are as follows:
Encoding  : SONY
Code      : 6AB47 (20 bits)
Library   : v2.4.2
...................................
uint32_t address = 0x1A;
uint32_t command = 0x7156;
uint64_t data = 0x6AB47;

When I send it out with the following statement, the decoded result is
irsend.sendSony(irsend.encodeSony(20, 0x56, 0x1A, 0x71), 20);

Encoding  : SONY
Code      : 6AB47 (20 bits)
Library   : v2.4.2
...................................
uint32_t address = 0x1A;
uint32_t command = 0x7156;
uint64_t data = 0x6AB47;
-----------------------------------
address:   0x1A  00011010
com:         0x56  01010110
extended: 0x71  01110001

1.result = address & 0x1F;
result = 11010
2.result |= (extended & 0xFF) << 6;
result |= 1110001**000000** //six zero
result = 1110001**0**11010
result |= 1110001**00000** //five zero
result = 111000111010 //error
3.result = (result << 7) | (command & 0x7F);
result = 1110001011010**0000000**|1010110
result = 11100010110101010110
result = 01101010101101000111
0x6AB47: 1101010101101000111

crankyoldgit, 2018-07-03:
Thanks for that.
I finally figured it out. I took a me a while and writing some test cases. The `encodeSony()` routine is correct. The issue is you (and me included) accidentally mistook `command = 0x7156;` as `command = 0x56` & `extended = 0x71`. As if it was a simple 8 bits + 8 bits.

But `command` is only is only 7 bits, not 8 bits. So while `command = 0x56` (the lower 7 bits), `extended` is the **not** the top 8 bits (0x71) of 0x7156 (16bits). It's the top 8 bits of a **15** bit value. It's 0x7156 >> 7 which is 0xE2. Obviously, an easy mistake to fall into. Made worse that the library can't report the three fields in results structure, so it has to combine command & extended into the same value.

Using your notes/style it should be:

address: 0x1A 11010 (5 bits)
com: 0x56 1010110 (7 bits)
extended: 0xE2 11100010 (8 bits)

crankyoldgit, 2018-07-03:
If you have a look at the PR/commit I just added, it explains & verifies it in code via some unit tests if that helps you.

closed 2018-07-03 by crankyoldgit

xfdr0805, 2018-07-04:
According to the agreement, it must have been shifted to the left by 5 bits. So I didn't understand what went wrong at the time. It was only from the decoding result that I found 1 bit less, so I changed it to the left by 6 bits, which is working normally. But this didn't solve the problem fundamentally, but now I understand.
I made this mistake too,I just updated and verified the code. It works well. 
Thank you again for your help!
Thank you again for your help!
Thank you again for your help!

crankyoldgit, 2018-07-04:
No worries. An easy mistake to make. Heck, I wrote the code and even I made the same mistake you did. ;-)

Out of curiosity, why are you using the `encodeSony()` routine at all?

xfdr0805, 2018-07-04:
Because I want to control DVD, TV, Amplifier, Music Server and other devices through infrared remote control, their protocols are different. There are probably NEC, RC6, and SONY. I will send the address and command to ESP8266 through WebSocket, and ESP8266 will receive it. The address and command are encoded and then sent out, so encodeNEC, encodeSony, and encodeRC6 are used. In this way, different devices are selected in the browser, and the function of the corresponding button changes, reaching the goal of discarding a bunch of remote controllers.
[pic](http://120.55.44.3/pic/pic1.jpg)

crankyoldgit, 2018-07-04:
You could just take the output "Code" from the dumpv2 program, and avoid using the various encodeBlah() routines.

Also, check out the IRMQTTServer example code. It allows you to send pretty much any IR code via MQTT or HTTP.

xfdr0805, 2018-07-04:
Yes, I will test IRMQTT when I have time. This may be a good idea, but I have to control other devices, so I use WebSocket + json to communicate, and the actual use is very good.