crankyoldgit/IRremoteESP8266 #477: IRrecvDumpV2 isnt working on a nodemcu

dindan100, 2018-06-28:
### Version/revison of the library used
v2.4.2

### Expected behavior
_What steps did you do and what should it have done?_
Tried to extract the raw data from my samung tv \ AC controler:

### Actual behavior
No response on the console

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_
Happens all the time

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc._
Identical to the one on the git page.

### I have followed the steps in the [Troubleshooting Guide] Yes

### Has this library/code previously worked as expected for you?
No

### Other useful information
_More information is always welcome. Be verbose._

- nodemcu v3 
- Latest library files

** IRrecvDemo is working just fine!
** IRrecvDump is generating an excpetion upon RST (which can only be seen when using 74880 as the baud rate) however when pressing the remote keys the data apears on the console!
** IRrecvDumpV2 is generating an excpetion upon RST (which can only be seen when using 74880 as the baud rate) and nothing on screen when  pressing the remote keys

crankyoldgit, 2018-06-28:
@dindan100 I'm afraid you probably got a dodgy nodemcu board. I suggest getting another one from a different seller. See this comment: https://github.com/markszabo/IRremoteESP8266/issues/415#issuecomment-366595184

Closed due to duplicate of Issue #415

closed 2018-06-28 by crankyoldgit

dindan100, 2018-06-28:
Dear David,

Thanks for taking the time in trying to assit! much appricated!
It seems that this board is working just fine with all other skecthes i tried (including web server) unlike the one in #415 however there is nothing like expreince and so i will offcourse trying to find another board from another seller, check it and update this case to let everyone in the future searching for such issues know what went wrong.

Thanks again!

dindan100, 2018-06-28:
BTW: will it work on ESP32? (i can try ofcourse but i am not sure it is supported if it fails)

crankyoldgit, 2018-06-28:
No idea if it will work on the ESP32, I haven't tried it. Worth a try though. I expect it won't work without some hackery.

Good luck with another board. Personally, I've been lucky with every nodemcu and wemos mini I've purchased, so I don't have first hand experience with this particular problem. At least these boards are cheap.

FYI, you should get startup text at **115200** Baud when running IRrecvDumpV2.ino. e.g. https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L170

If you are not, then something is VERY wrong.

dindan100, 2018-06-29:
Hi David,

I have checked 3 diffrnet esp models on the same computer with the same issue happening all of them which lead me to think that the problem is not a faulty nodemcu but something with the computer arduino software itself (altought the upload seems to be working just fine for this sketch or any sketch for that matter :)
Anyhow i have borrowed another laptop from my sister and walla! all skecthes including this one is working just fine! so in other words no code problem here :)
Still havent got it whats wrong with my computer as i reinstaled arduino software and i am using the same cables as for my laptop but at least we know now who's to blame.

Have a great weekdend!

crankyoldgit, 2018-06-29:
Very interesting. That explains the difference from the previous case/issues where WIFI didn't work either and this one. Might be the USB drivers perhaps?
If you find out what the cause is, please let me know and I'll add it to the FAQ.

dindan100, 2018-06-29:
Hi David,

Nope it's not the USB drivers as both computers use the same adapter with the same driver version and date.
It's something in the Arduino app which caused this which I cannot really explain what (I suspect its related to leftovers on the ESP which this code cannot replace\delete during the build\upload as the code worked only when uploading using tools->erase flash-> erase all content from the Arduino UI)
However those are the steps I have taken to solve this issue (which is now working just fine :)

1. Open your Arduino software
2. Choose File -> preference and press the link in the bottom to open the directory containing your preferences file.
3. Uninstall Arduino.
4. Remove the Arduino directory containing your preferences.txt file and its content.
5. Download and install the latest Arduino software.
6, Open Arduino app than choose File -> preference and set your preferred sketchbook location and set the following: http://arduino.esp8266.com/stable/package_esp8266com_index.json to your additional boards Manger url.
7. Install the esp8266 board and add the IRremoteESP8266 library.
8. Close and open Arduino and upload your IRrecvDumpV2 code to your ESP from the examples section (Make sure to use tools->erase flash-> erase all content from the Arduino UI option!!!)

crankyoldgit, 2018-06-30:
Thanks. I've updated the FAQ for this. https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#Some_of_the_examples_dont_produce_output_that_I_expect_and_I_have_to_use_a_baud_rate_of_74880_Whats_up

crankyoldgit, 2019-06-19:
@dindan100 As of [v2.6.2](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.6.2) this library now supports the ESP32.