crankyoldgit/IRremoteESP8266 #478: Minor Enhancements for  the MQTTserver Exmaple

dindan100, 2018-07-02:
Hi David,
I have just found and used your MQTTserver example and I am amazed with the quilty and willingness of you to help us all out so THANK you!
As such I thought I will share 2 minor issues I have changed to the example you might consider to add to the example.
1.	Add a note after the #define HOSTNAME "ir_server"  // Name of the device you want in mDNS.
//Changing this value will also change your MQTT path…
//E.g:
// ir_server/send or ir_server/sent 
//will be changed to [HOSTNAME]/send and [HOSTNAME]/sent

2.	Adding wifiManager.setSTAStaticIPConfig(IPAddress(x,x,x,xx), IPAddress(x,x,x,x), IPAddress(255,255,255,0));
after
wifiManager.setTimeout(300);  // Time out after 5 mins.

This will open 3 additional textboxes during the configuration allowing user to use fixed IP, router, mask (I Usually use fixed IP for my home servers)
 
** I also believe that some sort of username and password should be used before the web page is loaded as we don’t want anyone connected to our network to use this page without authentication even if from DOS point of view however I don’t know I do it, will try thought :)

crankyoldgit, 2018-07-02:
> 1. Add a note after the #define HOSTNAME

Yep. Sounds like a good idea. I'll add something like that soon.

> 2. Adding wifiManager.setSTAStaticIPConfig();

I'll look into it. I've been thinking of cleaning up and setting system parameters dynamically rather than compiled in.

Re: Password authentication.

Agreed. _Ideally_ yes, there should be something, but this isn't a real "project", it's just an example.
Adding basic http password auth is pretty weak. It complicates the URL based sending for starters, requiring the password in each request. As it isn't over SSL, there is also the whole password in the clear text thing. Sure, there is the whole XSS/XSRF potential issue but if you change the address, that's as good as a different password. If someone is exposing this port _directly_ to the Internet or their network isn't trusted, then they probably shouldn't be using this or they have accepted the risk.

Again, it's an example, not a complete polish project, and most of all, it meets my needs, which is why I wrote it. :-)

closed 2018-07-03 by crankyoldgit

crankyoldgit, 2018-07-27:
FYI, the changes are now live in the new v2.4.3 release of the library.

PR #480 (2018-07-03): Add static IP config option to IRMQTTServer
* Allow an easy method for users to use a static (non-dhcp) method for IP setup.
* Clarify some of the implications of changing HOSTNAME.

Fixes #478
commit 16465302 (2018-07-03): Add static IP config option to IRMQTTServer (#480)

* Allow an easy method for users to use a static (non-dhcp) method for IP setup.
* Clarify some of the implications of changing HOSTNAME.

Fixes #478