crankyoldgit/IRremoteESP8266 #479: General questions about sending Coolix

CWempe, 2018-07-02:
Sorry for opening an issue, but I feel like I am standing in front of a wall.

I want to use a *Wemos D1 mini* to controll my ceap AC.

With the `IRDumpV2` sketch I figured out my remote uses the COOLIX protocol.

Now I want to create a simple `TurnOnCoolixAC` sketch to test my hardware.
But I cannot figure out how to modify the existing examples to support COOLIX.

While the examples use something like this:

```
IRMitsubishiAC mitsubir(IR_LED); 
```

It does not work for Coolix.

```
TurnOnCoolixAC:35: error: 'IRCoolixAC' does not name a type

 IRCoolixAC coolixir(IR_LED);  // Set the GPIO to be used to sending the message

 ^
```

I also looked at the `IRMQTTServer`sketch, but I don't know where I would find a list of the available `protocol_num`.
But that is anoteher problem.

Once I know what to do I am would be happy to create pull requests to add `TurnOnCoolixAC` and improve the documentation if necessary. 😃

crankyoldgit, 2018-07-02:
> You are over thinking it a little. 😀
> Those TurnOnBlahAC examples are for when we have a proper encoder for an
> AC protocol. We don't have that for coolix unfortunately.
> Basically you want to take the hex code you got from the dump and put it
> into IRsendDemo using the send coolix(0x1234567); function.
>
> If you search for COOLIX in the mqtt example code you should find the
> protocol number. I think it is 15.

'''sendCoolix(0x1234567);''' ... Blasted mobile phone keyboard. :-/

crankyoldgit, 2018-07-03:
Yep. It's 15. See: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L321
The protocol numbers are technically and canonically defined here: https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L198
e.g. RC5 = 1, RC6 = 2, etc ...

And technically it's `sendCOOLIX(0x12345);` etc. See: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L1139

crankyoldgit, 2018-07-09:
@CWempe Just following up. Have we answered this enough for you?

CWempe, 2018-07-09:
Sorry, I had no time to try further.
But I guess your information will help me.
I will reopen this issue in case I need more help.

Thanks again.

closed 2018-07-09 by CWempe

CWempe, 2018-08-20:
It is working.
I am using the `IRMQTTServer` sketch.
Publishing `15,B21F28,24,3` to `ir_server/send` via mqtt starts the AC.
OFF: `15,B27BE0,24,3`

I collected the values via `IRrecvDump`.

Thanks