crankyoldgit/IRremoteESP8266 #482: Error in script auto_analyse_raw_data.py

tarzan115, 2018-07-06:
I have a timing raw value like this:
```
uint16_t rawbuf[272] = {3485, 3512, 864, 864, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 2620, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 3485, 3512, 864, 864, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 2620, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 864, 3485, 3512, 864, 13996, 3485, 3512, 864, 864, 864, 864, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 864, 864, 864, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 864, 864, 864, 864, 864, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 864, 864, 864, 3485, 3512, 864, 864, 864, 864, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 864, 864, 864, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 864, 864, 864, 864, 864, 864, 2620, 864, 2620, 864, 864, 864, 2620, 864, 2620, 864, 864, 864, 864, 3485, 3512, 864, 13996};
```

when I analyze with file `auto_analyse_raw_data.py` I got this with error
```
$python main.py --stdin
Found 272 timing entries.
Potential Mark Candidates:
[3485, 864]
Potential Space Candidates:
[13996, 3512, 2620, 864]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK   = 3485
HDR_SPACE  = 3512
BIT_MARK   = 864
ONE_SPACE  = 2620
ZERO_SPACE = 864
SPACE_GAP = 13996

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+01011111010111110100000001000000
  Bits: 32
  Hex:  0x5F5F4040 (MSB first)
        0x0202FAFA (LSB first)
  Dec:  1600077888 (MSB first)
        33749754 (LSB first)
  Bin:  0b01011111010111110100000001000000 (MSB first)
        0b00000010000000101111101011111010 (LSB first)
HMHDR_MARK+HDR_SPACE+01011111010111110100000001000000
  Bits: 32
  Hex:  0x5F5F4040 (MSB first)
        0x0202FAFA (LSB first)
  Dec:  1600077888 (MSB first)
        33749754 (LSB first)
  Bin:  0b01011111010111110100000001000000 (MSB first)
        0b00000010000000101111101011111010 (LSB first)
HMHDR_MARK+HDR_SPACE+ GAP(13996)Traceback (most recent call last):
  File "main.py", line 406, in <module>
    main()
  File "main.py", line 402, in main
    parse_and_report(data, arg_options.margin, arg_options.gen_code)
  File "main.py", line 227, in parse_and_report
    total_bits = decode_data(message, defines, function_code, output)
  File "main.py", line 294, in decode_data
    message.display_binary(binary_value)
  File "main.py", line 71, in display_binary
    num = int(binary_str, 2)
ValueError: invalid literal for int() with base 2: ''
```

but when I try with file `AutoAnalyseRawData.sh` I got this:
```
Potential Mark Candidates (using a range of 200):
3485
864

Potential Space Candidates (using a range of 200):
13996
3512
2620
864

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK = 3485
BIT_MARK = 864
SPACE_GAP1 = 13996
HDR_SPACE = 3512
ONE_SPACE = 2620
ZERO_SPACE = 864

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+01011111010111110100000001000000
  Bits: 32
  Hex:  0x5F5F4040 (MSB first)
        0x202FAFA (LSB first)
  Dec:  1600077888 (MSB first)
        33749754 (LSB first)
  Bin:  01011111010111110100000001000000 (MSB first)
        00000010000000101111101011111010 (LSB first)
HMHDR_MARK+HDR_SPACE+01011111010111110100000001000000
  Bits: 32
  Hex:  0x5F5F4040 (MSB first)
        0x202FAFA (LSB first)
  Dec:  1600077888 (MSB first)
        33749754 (LSB first)
  Bin:  01011111010111110100000001000000 (MSB first)
        00000010000000101111101011111010 (LSB first)
HMHDR_MARK+HDR_SPACE+ GAP(13996)
HDR_MARK+HDR_SPACE+00101111001011110110110001101100
  Bits: 32
  Hex:  0x2F2F6C6C (MSB first)
        0x3636F4F4 (LSB first)
  Dec:  791637100 (MSB first)
        909571316 (LSB first)
  Bin:  00101111001011110110110001101100 (MSB first)
        00110110001101101111010011110100 (LSB first)
HMHDR_MARK+HDR_SPACE+00101111001011110110110001101100
  Bits: 32
  Hex:  0x2F2F6C6C (MSB first)
        0x3636F4F4 (LSB first)
  Dec:  791637100 (MSB first)
        909571316 (LSB first)
  Bin:  00101111001011110110110001101100 (MSB first)
        00110110001101101111010011110100 (LSB first)
HMHDR_MARK+HDR_SPACE+ GAP(13996)

(standard_in) 2: syntax error
(standard_in) 2: syntax error
```

can you help me what is wrong?

tarzan115, 2018-07-06:
this is timing raw in Panasonic A/C

crankyoldgit, 2018-07-06:
I just added a commit to branch https://github.com/markszabo/IRremoteESP8266/tree/issue_482 that should fix it.

BTW, how did you capture that signal? It's far too ordered. All the different timings are exactly the same, to the usec.

The main bug was I've never seen a "gap" signal immediately after a header sequence, so didn't know I needed to handle that sort of a case. Unit test added for future etc.

Also ... your capture ends with a space (and a gap), which is odd in itself. Normally a capture is an odd number of elements, thus ends with a mark signal.

PR to follow shortly.

crankyoldgit, 2018-07-06:
With that PR/commit(s) it should produce output like:
```
$ ./auto_analyse_raw_data.py -g -f /tmp/issue_482
Found 272 timing entries.
Potential Mark Candidates:
[3485, 864]
Potential Space Candidates:
[13996, 3512, 2620, 864]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK   = 3485
HDR_SPACE  = 3512
BIT_MARK   = 864
ONE_SPACE  = 2620
ZERO_SPACE = 864
SPACE_GAP = 13996

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+01011111010111110100000001000000
  Bits: 32
  Hex:  0x5F5F4040 (MSB first)
        0x0202FAFA (LSB first)
  Dec:  1600077888 (MSB first)
        33749754 (LSB first)
  Bin:  0b01011111010111110100000001000000 (MSB first)
        0b00000010000000101111101011111010 (LSB first)
HDR_MARK+HDR_SPACE+01011111010111110100000001000000
  Bits: 32
  Hex:  0x5F5F4040 (MSB first)
        0x0202FAFA (LSB first)
  Dec:  1600077888 (MSB first)
        33749754 (LSB first)
  Bin:  0b01011111010111110100000001000000 (MSB first)
        0b00000010000000101111101011111010 (LSB first)
HDR_MARK+HDR_SPACE+GAP(13996)HDR_MARK+HDR_SPACE+00101111001011110110110001101100
  Bits: 32
  Hex:  0x2F2F6C6C (MSB first)
        0x3636F4F4 (LSB first)
  Dec:  791637100 (MSB first)
        909571316 (LSB first)
  Bin:  0b00101111001011110110110001101100 (MSB first)
        0b00110110001101101111010011110100 (LSB first)
HDR_MARK+HDR_SPACE+00101111001011110110110001101100
  Bits: 32
  Hex:  0x2F2F6C6C (MSB first)
        0x3636F4F4 (LSB first)
  Dec:  791637100 (MSB first)
        909571316 (LSB first)
  Bin:  0b00101111001011110110110001101100 (MSB first)
        0b00110110001101101111010011110100 (LSB first)
HDR_MARK+HDR_SPACE+GAP(13996)
Total Nr. of suspected bits: 128

Generating a VERY rough code outline:

// WARNING: This probably isn't directly usable. It's a guide only.
#define HDR_MARK 3485U
#define BIT_MARK 864U
#define HDR_SPACE 3512U
#define ONE_SPACE 2620U
#define ZERO_SPACE 864U
#define SPACE_GAP = 13996U
#define XYZ_BITS 128U
#define XYZ_STATE_LENGTH 16U
// DANGER: More than 64 bits detected. A uint64_t for 'data' won't work!
// Function should be safe up to 64 bits.
void IRsend::sendXYZ(const uint64_t data, const uint16_t nbits, const uint16_t repeat) {
  enableIROut(38);  // A guess. Most common frequency.
  for (uint16_t r = 0; r <= repeat; r++) {
    // Header
    mark(HDR_MARK);
    space(HDR_SPACE);
    // Data
    // e.g. data = 0x5F5F4040, nbits = 32
    sendData(BIT_MARK, ONE_SPACE, BIT_MARK, ZERO_SPACE, data, nbits, true);
    // Header
    mark(HDR_MARK);
    space(HDR_SPACE);
    // Data
    // e.g. data = 0x5F5F4040, nbits = 32
    sendData(BIT_MARK, ONE_SPACE, BIT_MARK, ZERO_SPACE, data, nbits, true);
    // Header
    mark(HDR_MARK);
    space(HDR_SPACE);
    // Gap
    mark(BIT_MARK);
    space(SPACE_GAP);
    // Header
    mark(HDR_MARK);
    space(HDR_SPACE);
    // Data
    // e.g. data = 0x2F2F6C6C, nbits = 32
    sendData(BIT_MARK, ONE_SPACE, BIT_MARK, ZERO_SPACE, data, nbits, true);
    // Header
    mark(HDR_MARK);
    space(HDR_SPACE);
    // Data
    // e.g. data = 0x2F2F6C6C, nbits = 32
    sendData(BIT_MARK, ONE_SPACE, BIT_MARK, ZERO_SPACE, data, nbits, true);
    // Header
    mark(HDR_MARK);
    space(HDR_SPACE);
    // Gap
    mark(BIT_MARK);
    space(SPACE_GAP);
    space(100000);  // A 100% made up guess of the gap between messages.
  }
}

// Alternative >64 bit Function
void IRsend::sendXYZ(uint8_t data[], uint16_t nbytes, uint16_t repeat) {
  // nbytes should typically be XYZ_STATE_LENGTH
  // data should typically be:
  //   uint8_t data[XYZ_STATE_LENGTH] = {0x5F, 0x5F, 0x40, 0x40, 0x5F, 0x5F, 0x40, 0x40, 0x2F, 0x2F, 0x6C, 0x6C, 0x2F, 0x2F, 0x6C, 0x6C};
  // data[] is assumed to be in MSB order for this code.
  for (uint16_t r = 0; r <= repeat; r++) {
    sendGeneric(HDR_MARK, HDR_SPACE,
                BIT_MARK, ONE_SPACE,
                BIT_MARK, ZERO_SPACE,
                BIT_MARK
                100000, // 100% made-up guess at the message gap.
                data, nbytes,
                38000, // Complete guess of the modulation frequency.
                true, 0, 50);
}
```

**FYI**: It seems to repeat itself a lot. If that is consistent across all buttons/modes, you can easily shrink it down to a total data size of 64 bits.

tarzan115, 2018-07-09:
I found this data by searching on the internet. I am trying to decode from pronto hex and analyze it to run in my project. 
thank you so much for this.
I really appreciate it.

closed 2018-07-09 by tarzan115

BABITA0005, 2019-01-10:
when I analyze with file auto_analyse_raw_data.py I got this with error in pycharm

usage: scratch.py [-h] [-g] [-f FILE] [-r MARGIN] [--stdin] [rawdata]
scratch.py: error: one of the arguments rawdata -f/--file --stdin is required

crankyoldgit, 2019-01-10:
Answered in #595

PR #483 (2018-07-09): [Bug] Handle Space Gaps better in auto analyse tool.
* Handle situations where we get a gap space immediately after a header.
* Remove errant 'HM' state being displayed.
* Several code lint cleanups.
* Unit tests coverage for the new code and bug.

Fixes #482
commit 5451e333 (2018-07-09): [Bug] Handle Space Gaps better in auto analyse tool. (#483)

* Handle situations where we get a gap space immediately after a header.
* Remove errant 'HM' state being displayed.
* Several code lint cleanups.
* Unit tests coverage for the new code and bug.

Fixes #482