crankyoldgit/IRremoteESP8266 #484: Coolix proto possible structure

Hamper, 2018-07-07:
I'm detect some signals from RG57K7(B)/BGEF remote control for Beko BINR 070/071 split-type air conditioner and try to parse possible binary structure. Can anyone check these signals with other devices with coolix proto?
```
0b1011 0010 0001 1111 0000 1000 - 17 C, auto
0b1011 0010 0001 1111 0000 0000 - 17 C, cool (fan auto)
0b1011 0010 1001 1111 0000 0000 - 17 C, cool (fan min)
0b1011 0010 0001 1111 0000 0100 - 17 C, dry
0b1011 0010 1011 1111 0000 1100 - 17 C, heat

0b1011 0010 1011 1111 1110 0100 - fan mode, fan auto
0b1011 0010 1001 1111 1110 0100 - fan mode, fan min
0b1011 0010 0101 1111 1110 0100 - fan mode, fan med
0b1011 0010 0011 1111 1110 0100 - fan mode, fan max

Fixed signals:
{
0b1011 0010 0111 1011 1110 0000 - off
0b1011 0010 0110 1011 1110 0000 - swing
0b1011 0010 1110 0000 0000 0011 - sleep mode

0b1011 0101 1111 0101 1010 0010 - turbo
0b1011 0101 1111 0101 1010 0101 - led
0b1011 0101 1111 0101 1010 1010 - self clean
}

0b1011 1010 0101 1010 xxxx 1010 - zone follow 26
0b1011 1010 1101 1010 xxxx 1010 - zone follow 26 current zone temp
0b1011 1010 0101 1001 xxxx 1010 - zone follow 25
0b1011 1010 1101 1001 xxxx 1010 - zone follow 25 current zone temp
0b1011 1010 1101 1000 xxxx 1010 - zone follow 24 current zone temp
0b1011 1010 0101 1011 xxxx 1010 - zone follow 27
                 1100           - 28
                 1101           - 29

Possible structure:
prefix | zf | op   | fan speed | x | current temp | temp  | mode | unknown
0b1011 | 0  | 010  | 000       | 1 | 1111         | 0000  | 10   | 00

mode:
00 - cool
01 - dry, fan mode with temp 1110 (can't use with zone follow)
10 - auto
11 - heat

zf (zone follow mode):
0 - off
1 - on

op:
010 - normal mode, off, swing, sleep
101 - turbo, led on/off, self clean, zone follow

fan speed:
001 - max
010 - medium (in zone follow mode: periodically send current temp, sound signal disabled)
100 - min
101 - auto
110 - with zone follow mode op: enable zone follow (with normal beep)
111 - fixed for modes: sleep, turbo, led, self clean

x:
0 - swing, sleep
1 - other modes

current temp (from remote control sensor for zone follow):
... (not checked)
1000 - 24 C
1001 - 25 C
1010 - 26 C
1011 - 27 C
1100 - 28 C
1101 - 29 C
...(not checked)
1111 - unknown, normal mode

temp:
0000 - 17 C
0001 - 18 C
0011 - 19 C
0010 - 20 C
0110 - 21 C
0111 - 22 C
0101 - 23 C
0100 - 24 C
1100 - 25 C
1101 - 26 C
1001 - 27 C
1000 - 28 C
1010 - 29 C
1011 - 30 C
1110 - fan mode

```

crankyoldgit, 2018-07-07:
Those `temp:` values at the end look very odd. How confident are you they are correct AND are temp values? i.e. Not a checksum.

Hamper, 2018-07-07:
I check full range of temperatures values in different modes, and only this 4 bits completely determined the temperature value and this values are same in different modes with same temperatures e. g. `0b1011 0010 0001 1111 [1101] 0100` is value for '26 C, dry mode'. I can't see dependence between real temperature and this bits but this sequence is really same for all other modes.

crankyoldgit, 2018-07-07:
As I said, it's odd. But hey, A/C protocols don't surprise me anymore. ;-)
What's with the `xxxx` bits in the zone section?

I hope you have good luck finding someone else with a coolix device. Tried looking up the Global Cache codes for it maybe?

Hamper, 2018-07-07:
Same sequence in other lib: https://github.com/ToniA/arduino-heatpumpir/blob/3d4decf06a5d548e03a121143e74c4c93574c076/R51MHeatpumpIR.cpp#L15

Hamper, 2018-07-07:
xxxx is just same temperature bits

crankyoldgit, 2018-09-10:
@Hamper & @Hollako,
I've coded up most of the functionality for the Coolix class based on @Hamper's work on decoding the Coolix protocol.

Care to try it out? (https://github.com/markszabo/IRremoteESP8266/tree/v2.5-coolix-improvements)

crankyoldgit, 2018-09-20:
Marking this issue closed due to staleness. Experimental support for deep decoding of Coolix protocol per this issue has been added to 2.5 which will be released soon.

closed 2018-09-20 by crankyoldgit

crankyoldgit, 2018-09-20:
v2.5.0 has been released.

Hollako, 2018-09-30:
Hello,

I tired the Coolix codes and i can confirm that is working perfectly, however i still only missing the syntax that i should use in Arduino sketch in order to send the Power Off command.

But again all the rest codes are working normally.

Regards,

crankyoldgit, 2018-09-30:
@Hollako Thanks for the confirmation. I'll update that it appears to work in the code comments.
In theory, all you need to do is the following to generate an "off" signal.

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Coolix.h>

IRCoolixAC coolix(GPIO_IR_PINOUT);

void setup() {
  coolix.begin();
  coolix.setPower(false);
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {
  Serial.println("Sending a Coolix OFF message");
  coolix.send();  // Sends what ever message we've already constructed etc.
  delay(5000);  // Wait 5 seconds.
}
```
or
```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Coolix.h>

IRsend irsend(GPIO_TO_USE_TO_SEND_IR);

void setup() {
  irsend.begin();
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {
  Serial.println("Sending a Coolix OFF message");
  irsend.sendCOOLIX(kCoolixOff);  // or  irsend.sendCOOLIX(0xB27BE0);
  delay(5000);  // Wait 5 seconds.
}
```

Note: Neither example has been tested or even compiled. So don't expect it to work out of the box. ;-)