crankyoldgit/IRremoteESP8266 #485: Support for Haier YR-W02

non7top, 2018-07-08:
This seems to be a bit different from what's already present for Haier AC

Timestamp : 000176.865
Encoding  : UNKNOWN
Code      : 6BB7252B (115 bits)
Library   : v2.4.2

Raw Timing[229]:
   +  2998, -  3086,    +  2998, -  4460,    +   568, -  1640,    +   596, -   492, 
   +   514, -  1690,    +   590, -   496,    +   566, -   532,    +   592, -  1596, 
   +   570, -  1618,    +   518, -   584,    +   590, -   538,    +   524, -   536, 
   +   568, -   532,    +   590, -  1596,    +   516, -   612,    +   568, -   538, 
   +   522, -  1638,    +   586, -   500,    +   512, -   614,    +   568, -   538, 
   +   520, -   538,    +   586, -   538,    +   566, -   540,    +   520, -   538, 
   +   586, -   538,    +   522, -   538,    +   588, -   538,    +   568, -   538, 
   +   520, -   538,    +   586, -   538,    +   566, -   538,    +   520, -   540, 
   +   588, -  1596,    +   590, -   536,    +   568, -   538,    +   520, -  1592, 
   +   640, -   538,    +   520, -   540,    +   588, -   538,    +   568, -   538, 
   +   516, -   562,    +   566, -   538,    +   518, -   542,    +   586, -   540, 
   +   566, -  1596,    +   590, -   538,    +   566, -   538,    +   516, -   544, 
   +   586, -   538,    +   516, -   542,    +   588, -   540,    +   564, -   540, 
   +   468, -   590,    +   588, -   538,    +   566, -   540,    +   466, -   590, 
   +   588, -   538,    +   514, -   544,    +   588, -   538,    +   566, -   538, 
   +   468, -  1692,    +   606, -   526,    +   466, -   592,    +   588, -   538, 
   +   568, -   490,    +   588, -   538,    +   566, -   540,    +   466, -   592, 
   +   588, -   538,    +   566, -   538,    +   466, -   592,    +   588, -   538, 
   +   568, -   492,    +   586, -   540,    +   566, -   540,    +   468, -   590, 
   +   588, -   538,    +   568, -   516,    +   488, -   590,    +   588, -   538, 
   +   568, -   492,    +   588, -   538,    +   566, -   518,    +   488, -   590, 
   +   588, -   540,    +   564, -   518,    +   490, -   590,    +   588, -   538, 
   +   562, -   496,    +   588, -   538,    +   566, -   518,    +   488, -   590, 
   +   588, -   538,    +   562, -   522,    +   488, -   588,    +   590, -   538, 
   +   560, -   498,    +   588, -   540,    +   564, -   522,    +   486, -   590, 
   +   590, -   538,    +   560, -   524,    +   488, -   588,    +   588, -  1598, 
   +   514, -   608,    +   564, -  1600,    +   548, -   536,    +   586, -   538, 
   +   568, -  1594,    +   590, -  1618,    +   578, -  1606,    +   606, -  1582, 
   +   590, -  1596,    +   590, -  1616,    +   580

uint16_t rawData[229] = {2998, 3086,  2998, 4460,  568, 1640,  596, 492,  514, 1690,  590, 496,  566, 532,  592, 1596,  570, 1618,  518, 584,  590, 538,  524, 536,  568, 532,  590, 1596,  516, 612,  568, 538,  522, 1638,  586, 500,  512, 614,  568, 538,  520, 538,  586, 538,  566, 540,  520, 538,  586, 538,  522, 538,  588, 538,  568, 538,  520, 538,  586, 538,  566, 538,  520, 540,  588, 1596,  590, 536,  568, 538,  520, 1592,  640, 538,  520, 540,  588, 538,  568, 538,  516, 562,  566, 538,  518, 542,  586, 540,  566, 1596,  590, 538,  566, 538,  516, 544,  586, 538,  516, 542,  588, 540,  564, 540,  468, 590,  588, 538,  566, 540,  466, 590,  588, 538,  514, 544,  588, 538,  566, 538,  468, 1692,  606, 526,  466, 592,  588, 538,  568, 490,  588, 538,  566, 540,  466, 592,  588, 538,  566, 538,  466, 592,  588, 538,  568, 492,  586, 540,  566, 540,  468, 590,  588, 538,  568, 516,  488, 590,  588, 538,  568, 492,  588, 538,  566, 518,  488, 590,  588, 540,  564, 518,  490, 590,  588, 538,  562, 496,  588, 538,  566, 518,  488, 590,  588, 538,  562, 522,  488, 588,  590, 538,  560, 498,  588, 540,  564, 522,  486, 590,  590, 538,  560, 524,  488, 588,  588, 1598,  514, 608,  564, 1600,  548, 536,  586, 538,  568, 1594,  590, 1618,  578, 1606,  606, 1582,  590, 1596,  590, 1616,  580};  // UNKNOWN 6BB7252B

This is what I've come up with so far
https://www.dropbox.com/sh/w0bt7egp0fjger5/AADRFV6Wg4wZskJVdFvzb8Z0a?dl=0

Can't figure out about the last three bits, apparently there should be a checksum there. Some help would be appreciated.

crankyoldgit, 2018-07-09:
Looks like you have a longer variant. The one we have existing code for is 9 data bytes (72 bits) and yours is 14 data bytes (112 bits)

I can probably knock something up in a day or so, that does basic sending/receiving but no decoding.

crankyoldgit, 2018-07-09:
Can you let us know what model the A/C unit is **_AND_** what model the remote is please!

non7top, 2018-07-09:
Remote is YR-W02, the unit is white, not sure how to find the exact model name.

crankyoldgit, 2018-07-09:
The remote is the most important bit, Thanks! Any idea roughly how old the unit is?

Looking at your spreadsheet data ... I notice that **every** second bit seems to always be zero (0) and you have 28 bytes .. not 14. I think somehow you are doubling something somewhere with your calculations.

vberezhnoy11, 2018-07-09:
Unit name is HSU-09HMC203

non7top, 2018-07-09:
Example dump from the new branch
```
Timestamp : 000037.101
Encoding  : HAIER_AC_YRW02
Code      : 40002040020092A6 (112 bits)
Library   : v2.4.2

Raw Timing[229]:
   +  3106, -  2976,    +  3108, -  4370,    +   592, -  1596,    +   590, -   536, 
   +   568, -  1594,    +   590, -   538,    +   568, -   538,    +   468, -  1690, 
   +   606, -  1578,    +   606, -   524,    +   468, -  1692,    +   606, -   524, 
   +   466, -   592,    +   586, -  1620,    +   574, -   510,    +   588, -   538, 
   +   566, -  1592,    +   608, -   524,    +   468, -   590,    +   588, -   538, 
   +   566, -   540,    +   466, -   592,    +   588, -   540,    +   566, -   538, 
   +   468, -   612,    +   566, -   538,    +   468, -   590,    +   588, -   538, 
   +   566, -   540,    +   468, -   590,    +   588, -   538,    +   566, -   518, 
   +   488, -  1692,    +   550, -   534,    +   588, -   538,    +   566, -  1594, 
   +   592, -   538,    +   564, -   520,    +   488, -   590,    +   590, -   538, 
   +   560, -   498,    +   588, -   538,    +   564, -   522,    +   488, -   590, 
   +   590, -  1624,    +   488, -   590,    +   588, -   538,    +   556, -   528, 
   +   488, -   610,    +   562, -   524,    +   488, -   590,    +   590, -   538, 
   +   558, -   528,    +   486, -   590,    +   588, -   538,    +   556, -   506, 
   +   512, -   610,    +   558, -   528,    +   486, -   590,    +   590, -  1624, 
   +   490, -   586,    +   590, -   538,    +   552, -   508,    +   566, -   534, 
   +   590, -   494,    +   514, -   588,    +   590, -   538,    +   552, -   508, 
   +   566, -   534,    +   590, -   540,    +   522, -   536,    +   566, -   534, 
   +   590, -   494,    +   566, -   534,    +   590, -   540,    +   524, -   536, 
   +   568, -   532,    +   592, -   538,    +   522, -   538,    +   560, -   564, 
   +   566, -   492,    +   566, -   534,    +   590, -   538,    +   524, -   536, 
   +   512, -   614,    +   566, -   538,    +   522, -   538,    +   512, -   612, 
   +   568, -   494,    +   560, -   564,    +   566, -   540,    +   520, -   538, 
   +   512, -   612,    +   568, -   538,    +   520, -   540,    +   586, -   538, 
   +   568, -   492,    +   512, -   614,    +   568, -   538,    +   522, -   538, 
   +   588, -   538,    +   566, -   538,    +   520, -   540,    +   588, -  1618, 
   +   580, -   524,    +   568, -   540,    +   518, -  1640,    +   606, -  1582, 
   +   590, -   538,    +   568, -  1596,    +   590, -  1618,    +   578, -  1608, 
   +   606, -  1578,    +   606, -   502,    +   590

uint16_t rawData[229] = {3106, 2976,  3108, 4370,  592, 1596,  590, 536,  568, 1594,  590, 538,  568, 538,  468, 1690,  606, 1578,  606, 524,  468, 1692,  606, 524,  466, 592,  586, 1620,  574, 510,  588, 538,  566, 1592,  608, 524,  468, 590,  588, 538,  566, 540,  466, 592,  588, 540,  566, 538,  468, 612,  566, 538,  468, 590,  588, 538,  566, 540,  468, 590,  588, 538,  566, 518,  488, 1692,  550, 534,  588, 538,  566, 1594,  592, 538,  564, 520,  488, 590,  590, 538,  560, 498,  588, 538,  564, 522,  488, 590,  590, 1624,  488, 590,  588, 538,  556, 528,  488, 610,  562, 524,  488, 590,  590, 538,  558, 528,  486, 590,  588, 538,  556, 506,  512, 610,  558, 528,  486, 590,  590, 1624,  490, 586,  590, 538,  552, 508,  566, 534,  590, 494,  514, 588,  590, 538,  552, 508,  566, 534,  590, 540,  522, 536,  566, 534,  590, 494,  566, 534,  590, 540,  524, 536,  568, 532,  592, 538,  522, 538,  560, 564,  566, 492,  566, 534,  590, 538,  524, 536,  512, 614,  566, 538,  522, 538,  512, 612,  568, 494,  560, 564,  566, 540,  520, 538,  512, 612,  568, 538,  520, 540,  586, 538,  568, 492,  512, 614,  568, 538,  522, 538,  588, 538,  566, 538,  520, 540,  588, 1618,  580, 524,  568, 540,  518, 1640,  606, 1582,  590, 538,  568, 1596,  590, 1618,  578, 1608,  606, 1578,  606, 502,  590};  // HAIER_AC_YRW02 40002040020092A6
uint32_t address = 0x0;
uint32_t command = 0xDE04;
uint64_t data = 0x40002040020092A6;
```

non7top, 2018-07-09:
Unit name is HSU-09HMC203

crankyoldgit, 2018-07-09:
@non7top Thanks for the copy of the dump. It highlighted there was subtle error in it's display of the state code. i.e. it shouldn't display `address`, `command`, & `data` but should show the state instead.

I've added two more commits to that branch/PR. Care to test it again please?

crankyoldgit, 2018-07-09:
Excellent. Thanks. Great news that you've worked out the checksum. That is typically the most challenging bit.
If you want to construct another spreadsheet based on that data and highlight which bits are which we can try to add a full encoder/decoder for all the modes etc.

In the meantime, I'll merge the PR into the master branch.

non7top, 2018-07-10:
I added spreadsheet to dropbox and here are the details:

Bit 1-2 - header
A6

Bit 3 - temperature
0-16 .. E-30

Bit 4 - swing
0 - power off
1 - 2 top
2 - 1 top
3 - 2 bot (only in heat)
A - 1 bot
C - Auto (airflow)

Bit 8 - health
0 - off
2 - on

Bit 9 - power
4 - on
0 - off

Bit 11 - fan
6 - 1
4 - 2
2 - 3
A - Auto

Bit 13 - turbo mode
0 - off
4 - high
8 - low

Bit 15 - mode
0 - Auto
2 - Cold
4 - dry
8 - heat
C - ventilation

Bit 17 - sleep mode
0 - off
8 - on

Bit 26, 27 28 (+probably 25) - Checksum
Sum of previous bits

crankyoldgit, 2018-07-10:
> Bit 1-2 - header
...

I think you mean _Byte_ or _Nibble_ (https://en.wikipedia.org/wiki/Nibble)

e.g. There is no Byte 26, 27, & 28. There are only 14 bytes in the state/message.

I concur that it looks like the last byte (i.e. state[13]) (Your 27-28) is the sum of the previous bytes. (well, all I checked was it behaves like it is, not verified)
e.g. for entry "16 cold fan3, swing top, health" it is 0x2B.
However, this doesn't explain what the last-1 byte is. (ie. state[12]).
e.g. It's 0x01 in "16 cold fan3, swing top, health" and 0x00 in "17 cold fan3, swing top, health" and others. Those two states only differ in state[1] (your 3-4). 0x02 vs 0x12 respectively. The checksum is still the sum of bytes 0-12.

You need to work out what those bytes just before the checksum are, otherwise we can't calculate a new desired state.

If the remote has a clock on it, it maybe be the time encoded into the signal. Check for that. It could also be timers or sleep modes etc etc. Basically there is something, maybe a mode, or some other data or state of the remote you need to workout before we can reproduce synthetically.

non7top, 2018-07-10:
Yup, you are right. state[12] is last button pressed
0 - temp up
1 - temp down
4 - fan
5 - power
6 - mode
2 - swing and airflow
7 - health
8 - turbo mode
B - sleep

And the checksum is two last octets of sum of all bytes. 

uint8_t state[14] = {0xA6, 0xEA, 0x00, 0x00, 0x40, 0x20, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x05, 0x75};
The sum is 0x275, but only 75 is there.

crankyoldgit, 2018-07-10:
Wow. Excellent work. I would not have been able to work that out.

Re: Checksum.
Yes. It's quite common for the checksum to be bit-masked to a set number of bits. Typically 4 or 8.
e.g. `sum = sum & 0xFF;`

Do you have a link to the manual for the device?

non7top, 2018-07-10:
Was only able to find it in russian https://mcgrp.ru/files/viewer/84556/16
The device doesn't have clock, so I don't find the simple timer functions of any use and didn't bother to record them.

crankyoldgit, 2018-07-10:
Can you please re-verify your spreadsheet entry for `30 cold fan3, swing top, health, power high` (last entry)?
It has a mode of 0 (auto) but your description of the mode is `cold`.

non7top, 2018-07-10:
Yup, apparently I mistyped that entry, correct one is (turbo button pressed) which seems to be correct
uint8_t state[14] = {0xA6, 0xE2, 0x00, 0x02, 0x40, 0x20, 0x40, 0x20, 0x00, 0x00, 0x00, 0x00, 0x08, 0x52};

crankyoldgit, 2018-07-11:
Can you also please re-check that _Fan 1_ really is 6? I have a sneaking suspicion that it's 8.

crankyoldgit, 2018-07-11:
In the meantime, can you try the updated haier_yrw02 branch? It should allow construction of the state for the remote, and full mode decoding in IRrecvDumpV2.

non7top, 2018-07-11:
Ok, here are some findings.

Fan is reproducible to be first nibble in bit 6. Current decoding works in all situations, except when Sleep mode is on. If Sleep mode is on, then first nibble still can decode fan state the same way as I provided, but second nibble seems to be random, at least I couldn't find any correlations. Considering this I'd rather skip setting Sleep mode at all (especially I don't find it usable).

```
uint8_t state[14] = {0xA6, 0x9C, 0x00, 0x02, 0x40, 0xA0, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x0B, 0x4F}; cool 25, health, fan auto, swing auto, sleep off
Mesg Desc.: Power: On, Button: 11 (Sleep), Mode: 2 (COOL), Temp: 25C, Fan: 160 (AUTO), Swing: 12 (Auto), Sleep: Off, Health: On

uint8_t state[14] = {0xA6, 0x9C, 0x00, 0x02, 0x40, 0xA8, 0x00, 0x20, 0x80, 0x00, 0x00, 0x00, 0x0B, 0xD7}; cool 25, health, fan auto, swing auto,  sleep on
Mesg Desc.: Power: On, Button: 11 (Sleep), Mode: 2 (COOL), Temp: 25C, Fan: 168 (UNKNOWN), Swing: 12 (Auto), Sleep: On, Health: On

uint8_t state[14] = {0xA6, 0x9C, 0x00, 0x02, 0x40, 0x67, 0x3A, 0x20, 0x80, 0x00, 0x00, 0x00, 0x04, 0xC9}; cool 25, health, fan 1, swing auto,  sleep on
Mesg Desc.: Power: On, Button: 4 (Fan), Mode: 2 (COOL), Temp: 25C, Fan: 103 (UNKNOWN), Swing: 12 (Auto), Sleep: On, Health: On

uint8_t state[14] = {0xA6, 0x9C, 0x00, 0x02, 0x40, 0x47, 0x38, 0x20, 0x80, 0x00, 0x00, 0x00, 0x04, 0xA7}; cool 25, health, fan 2, swing auto,  sleep on
Mesg Desc.: Power: On, Button: 4 (Fan), Mode: 2 (COOL), Temp: 25C, Fan: 71 (UNKNOWN), Swing: 12 (Auto), Sleep: On, Health: On

uint8_t state[14] = {0xA6, 0x9C, 0x00, 0x02, 0x40, 0x27, 0x36, 0x20, 0x80, 0x00, 0x00, 0x00, 0x04, 0x85}; cool 25, health, fan 3, swing auto,  sleep on
Mesg Desc.: Power: On, Button: 4 (Fan), Mode: 2 (COOL), Temp: 25C, Fan: 39 (UNKNOWN), Swing: 12 (Auto), Sleep: On, Health: On

uint8_t state[14] = {0xA6, 0x9C, 0x00, 0x02, 0x40, 0x20, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x0B, 0xCF}; cool 25, health, fan 3, swing auto,  sleep off
Mesg Desc.: Power: On, Button: 11 (Sleep), Mode: 2 (COOL), Temp: 25C, Fan: 32 (HIGH), Swing: 12 (Auto), Sleep: Off, Health: On
```
More details about swing. One mode is only possible in heat, another one is only possible in other modes. This is how it can be set on remote, didn't try yet to set that manually.

Following is possible in Heat mode:
1 - 2 top (air flow is forward top)
A - 1 bot (air flow is forward bottom)
3 - 2 bot (air flow is vertical down)
C - Auto (airflow)

Following is possible in all other modes:
1 - 2 top (air flow is forward top)
2 - 1 top (air flow is forward middle)
3 - 2 bot (air flow is vertical down)
C - Auto (airflow)

Also turbo mode is not decoded, it sets fan to more high or more low speeds then standard values, like two additional fan modes.

crankyoldgit, 2018-07-12:
I've got a pending commit that fixes most of it, but I _think_ your definition of what bits control Turbo is incorrect.
e.g.
> uint8_t state[14] = {0xA6, 0x9C, 0x00, 0x02, 0x40, 0x27, 0x36, 0x20, 0x80, 0x00, 0x00, 0x00, 0x04, 0x85}; cool 25, health, fan 3, swing auto,  sleep on

This has a `3` in the 13 nibble.
> Bit 13 - turbo mode
0 - off
4 - high
8 - low

I _think_ it's controlled only by the top 2 bits of nibble 13. (aka. top 2 bits of state[6]).

The other odd-data detected when in _sleep_ mode are possibly timer data. That's my guess. We are ignoring it for now, but we should probably work out what it is at some point.

crankyoldgit, 2018-07-12:
While I think of it, does setting Turbo mode affect the value of Fan at all?
e.g. If Fan was on the lowest setting, and then you set Turbo to High, does the Fan value in the message also change?

crankyoldgit, 2018-07-27:
FYI, the changes are now live in the new v2.4.3 release of the library.

non7top, 2018-07-31:
I was finally able able to get back to this project. Big thanks for making it happen and all your help with it. 
I took one of the examples and adopted it a bit and it worked as expected, different temperatures also worked.
I'm going to do some more extended sketch this and do more testing.

This worked:
```
  ac.setPower(true);
  ac.setFan(HAIER_AC_YRW02_FAN_LOW);
  ac.setSwing(HAIER_AC_YRW02_SWING_AUTO);
  ac.setTemp(30);
  ac.setMode(HAIER_AC_YRW02_COOL);
  ac.setTurbo(HAIER_AC_YRW02_TURBO_HIGH);

#if SEND_HAIER_AC_YRW02
  // Now send the IR signal.
  ac.send();
```

crankyoldgit, 2018-07-31:
Excellent. I'll mark this issue closed then. As it's now "live" in the latest version, if you find a problem please create a new issue for it. Hopefully you won't find any problems (haha .. yeah right! ;-)

closed 2018-07-31 by crankyoldgit