crankyoldgit/IRremoteESP8266 #488: How to send a data not divisor by 8 bit

tarzan115, 2018-07-11:
### Version/revison of the library used
latest git https://github.com/markszabo/IRremoteESP8266/commit/9506c39cfee50fa497650d6f7d8f73ca7c7aa10c

### Expected behavior
Hi @crankyoldgit 
I detect a code from LG A/C like this:
```
Timestamp : 001720.790
Encoding  : LG
Code      : 88C0051 (28 bits)
Library   : v2.4.2

Raw Timing[59]:
   +  8400, -  4148,    +   574, -  1504,    +   572, -   466,    +   576, -   462, 
   +   570, -   470,    +   574, -  1502,    +   572, -   466,    +   576, -   462, 
   +   570, -   470,    +   572, -  1504,    +   572, -  1504,    +   572, -   466, 
   +   576, -   462,    +   570, -   468,    +   574, -   464,    +   568, -   474, 
   +   568, -   468,    +   576, -   462,    +   570, -   468,    +   574, -   464, 
   +   568, -   470,    +   572, -   464,    +   568, -  1510,    +   576, -   462, 
   +   570, -  1506,    +   570, -   470,    +   572, -   466,    +   576, -   460, 
   +   572, -  1508,    +   568

uint16_t rawData[59] = {8400, 4148,  574, 1504,  572, 466,  576, 462,  570, 470,  574, 1502,  572, 466,  576, 462,  570, 470,  572, 1504,  572, 1504,  572, 466,  576, 462,  570, 468,  574, 464,  568, 474,  568, 468,  576, 462,  570, 468,  574, 464,  568, 470,  572, 464,  568, 1510,  576, 462,  570, 1506,  570, 470,  572, 466,  576, 460,  572, 1508,  568};  // LG 88C0051
uint32_t address = 0x88;
uint32_t command = 0xC005;
uint64_t data = 0x88C0051;
```

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
I used `auto_analyse_raw_data.py` file to analyze that code to store and I got
```
Found 59 timing entries.
Potential Mark Candidates:
[8368, 550]
Potential Space Candidates:
[4182, 1538, 500]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK   = 8368
HDR_SPACE  = 4182
BIT_MARK   = 550
ONE_SPACE  = 1538
ZERO_SPACE = 500

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+1000100011000000000001010001
  Bits: 28
  Hex:  0x88C0051 (MSB first)
        0x8A00311 (LSB first)
  Dec:  143392849 (MSB first)
        144704273 (LSB first)
  Bin:  0b1000100011000000000001010001 (MSB first)
        0b1000101000000000001100010001 (LSB first)

Total Nr. of suspected bits: 28
```

now I want to use method `sendGeneric()` with data type is `uint8_t *` because I have some other code with data larger 64 bit. but I don't know how to send that data (`0x88C0051`).
I tried split it by 3 case:
```c
uint8_t *data = {0x88, 0xC0, 0x05, 0x10} // padding in back of last byte
```
```c
uint8_t *data = {0x88, 0xC0, 0x05, 0x01} // padding in front of last byte
```
```c
uint8_t *data = {0x08, 0x8C, 0x00, 0x51} // padding in front of first byte
```
#### Example code used
and my small code is
```c
#include <IRremoteESP8266.h>
#include <IRSend.h>

IRsend irsend(0);
void setup()
{
    irsend.begin();
}
void loop()
{
  uint16_t HDR_MARK   = 8368;
  uint16_t HDR_SPACE  = 4182;
  uint16_t BIT_MARK   = 550;
  uint16_t ONE_SPACE  = 1538;
  uint16_t ZERO_SPACE = 500;
  // uint8_t *data = {0x88, 0xC0, 0x05, 0x10};
  // uint8_t *data = {0x88, 0xC0, 0x05, 0x01};
  uint8_t *data = {0x08, 0x8C, 0x00, 0x51};

  irsend.sendGeneric(HDR_MARK, HDR_SPACE, BIT_MARK, ONE_SPACE, BIT_MARK, ZERO_SPACE, BIT_MARK, 0, data, 4, 38000, true, 0, 50);
  delay(5000);
}
```
I send and I capture again, it always is 32 bit, not 28 bit anymore.
my question is how can I send exactly 28 bit with correct data. I know the library have `uint64_t` but I want to know because I have other code more than 64 bit.

Thank you in advance!

crankyoldgit, 2018-07-11:
The very short answer is, you can't send non 8-bit aligned data with `sendGeneric()` using `uint8_t*` data type method. Please note: There are *three* sendGeneric methods. One for `uint8_t*`, the others for `uint64_t` which allows arbitrary bit sizes.

The long answer:

Based on your OP, for the LG message, you really just want to do this:
```
irsend.sendLG(0x88C0051);  // data from the code/data lines of the dump. This will do everything for you.
```

You really shouldn't use `sendGeneric()` to reproduce that LG message. If you do, you need to use the non-array/non-pointer method(s).

If you insist on still using the `sendGeneric()` methods. There are three variants:
* The _first_ one (https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L333) is when `data` is an integer (`uint64_t` but of course you can pass something smaller to it) and can have between 1 & 64 bits (`nbits`) transmitted. This works for most _simple_ space-encoded protocols to simple devices. e.g. TVs, stereos, some A/C units.
* The _second_ one (https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L362) is the same as the first, except it has a `mesgtime` parameter for some protocols that have a _fixed total time_ for a message. This variant is only used when we *really* understand the protocol. e.g. There are published specifications etc.
* The _third_ one, which you are using above (https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L436) is similar to the first, except it uses a *pointer* to an array of 8-bit chars (`*dataptr`), and the nr. of bytes (`nbytes`) to be sent. This is useful for large A/C-like protocols which are based around 8-bit aligned data and are larger than a `uint64_t` in size.

There is no _one-size-fits-all_ version of `sendGeneric()`. We could potentially add a 4th variant that acts like the _third_ one, but in addition has the concept of a nr. of bits to be sent but that leads to issues, like the padding issue you explored, plus some other issues.

The only _one-size-fits-all_ method is `sendRaw()`, which has the worst storage case, but has the highest reproduction fidelity because of it. This can reproduce all protocols, even non-space encoded ones. See the RC4/RC5/RC6 protocols for example.

tarzan115, 2018-07-12:
thank you for the useful information, I will use `sendRaw()` method instead

closed 2018-07-12 by tarzan115