crankyoldgit/IRremoteESP8266 #489: general question.

DasVoelkel, 2018-07-12:
I am searching for errors in my code, and it seems i have some "Exception (4)" so, some kind of interrupt problem. 
I take it your library uses the ICACHE_RAM_ATTR attribute correctly ? 
It's unlikely that infra is the problem, but i just want to make sure

Kudos!

crankyoldgit, 2018-07-12:
Yes, as far as I know it does. We are used in numerous projects and have had no reports of interrupt issues. We have also gone to extreme lengths to reduce our IRAM footprint. Interrupts & IRAM is only consumed by this library if you use the receiving (`IRrecv.h`) portion of the library. The sending (`IRsend,h`) portion does not consume any IRAM or use any interrupts.

All the interrupt (and hence IRAM) code is here: https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L35 in the two methods: `read_timeout()` and `gpio_intr()`

closed 2018-07-12 by crankyoldgit

DasVoelkel, 2018-07-13:
thanks a lot :)