crankyoldgit/IRremoteESP8266 #493: Midea decode not working

jimmys01, 2018-07-21:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revison of the library used
2.4.2

### Expected behavior
Decode the Midea Protocol

### Actual behavior
Coolix shows up instead

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
From IRRECDUMPv2 i get this:
Encoding  : COOLIX
Code      : B27BE0 (24 bits)
 Raw Timing  [199]:  +4458,-4328,+606,-1574,+606,-512,+580,-1602,+582,-1576,+604,-512,+582,-512,+580,-1598,+582,-514,+582,-514,+580,-1578,+606,-512,+580,-512,+580,-1574,+604,-1598,+582,-512,+582,-1604,+580,-516,+580,-1598,+582,-1598,+580,-1574,+606,-1574,+604,-512,+582,-1600,+582,-1600,+582,-1602,+582,-512,+580,-512,+580,-512,+582,-512,+582,-1596,+582,-488,+606,-516,+580,-1604,+582,-1574,+606,-1598,+580,-512,+582,-512,+580,-488,+604,-512,+580,-516,+582,-514,+580,-512,+580,-512,+580,-1578,+604,-1598,+582,-1576,+604,-1598,+582,-1574,+556,-5228,+4460,-4320,+608,-1576,+604,-512,+582,-1576,+606,-1596,+582,-512,+580,-512,+582,-1574,+606,-514,+580,-514,+580,-1602,+582,-512,+580,-514,+580,-1576,+604,-1574,+606,-512,+582,-1578,+608,-490,+606,-1598,+582,-1598,+582,-1574,+606,-1598,+580,-512,+582,-1602,+582,-1574,+606,-1602,+582,-512,+584,-508,+582,-512,+582,-512,+580,-1598,+582,-510,+584,-512,+582,-1604,+582,-1596,+582,-1598,+582,-512,+580,-512,+582,-512,+582,-512,+582,-512,+582,-488,+606,-512,+580,-490,+604,-1602,+582,-1574,+606,-1574,+606,-1574,+606,-1600,+532

uint16_t rawData[199] = {4458, 4328,  606, 1574,  606, 512,  580, 1602,  582, 1576,  604, 512,  582, 512,  580, 1598,  582, 514,  582, 514,  580, 1578,  606, 512,  580, 512,  580, 1574,  604, 1598,  582, 512,  582, 1604,  580, 516,  580, 1598,  582, 1598,  580, 1574,  606, 1574,  604, 512,  582, 1600,  582, 1600,  582, 1602,  582, 512,  580, 512,  580, 512,  582, 512,  582, 1596,  582, 488,  606, 516,  580, 1604,  582, 1574,  606, 1598,  580, 512,  582, 512,  580, 488,  604, 512,  580, 516,  582, 514,  580, 512,  580, 512,  580, 1578,  604, 1598,  582, 1576,  604, 1598,  582, 1574,  556, 5228,  4460, 4320,  608, 1576,  604, 512,  582, 1576,  606, 1596,  582, 512,  580, 512,  582, 1574,  606, 514,  580, 514,  580, 1602,  582, 512,  580, 514,  580, 1576,  604, 1574,  606, 512,  582, 1578,  608, 490,  606, 1598,  582, 1598,  582, 1574,  606, 1598,  580, 512,  582, 1602,  582, 1574,  606, 1602,  582, 512,  584, 508,  582, 512,  582, 512,  580, 1598,  582, 510,  584, 512,  582, 1604,  582, 1596,  582, 1598,  582, 512,  580, 512,  582, 512,  582, 512,  582, 512,  582, 488,  606, 512,  580, 490,  604, 1602,  582, 1574,  606, 1574,  606, 1574,  606, 1600,  532};  // COOLIX B27BE0
uint64_t data = 0xB27BE0;

### Steps to reproduce the behavior
You need an Midea AC controler

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Other useful information
If you need any sample captures please let me know

crankyoldgit, 2018-07-21:
FWIW, I've taken a look at the raw data you've provided, and it really looks like a COOLIX message to me.

The physical device might be a Midea branded device, but the message you have included is indeed a COOLIX protocol message. The coolix protocol requires each 8 bits sent to be inverted of the previous 8 bits.
e.g.
The raw data decodes to `0xB24D7B84E01F`. 0xB2 == ~0x4D, 0x7B == ~0x84, & 0xE0 == ~0x1F

The Midea protocol requires a 48 bit message, that is repeated a second time, but that entire 48 bit message is inverted. That's not the case here.
 
It is not uncommon for a Brand X device to use another manufacturer's message protocol.

The message you have there should be able to be reproduced with:
```
irsend.sendCOOLIX(0xB27BE0, 24, 1);  // The raw capture you provided has a repeat in it.
```

crankyoldgit, 2018-07-21:
Just as a FYI:
```
IRremoteESP8266/src$ ../tools/auto_analyse_raw_data.py -f /tmp/issue_493 
Found 199 timing entries.
Potential Mark Candidates:
[4460, 608]
Potential Space Candidates:
[5228, 4328, 1604, 516]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK   = 4460
HDR_SPACE  = 4328
BIT_MARK   = 608
ONE_SPACE  = 1604
ZERO_SPACE = 516
SPACE_GAP = 5228

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+101100100100110101111011100001001110000000011111GAP(5228)
  Bits: 48
  Hex:  0xB24D7B84E01F (MSB first)
        0xF80721DEB24D (LSB first)
  Dec:  196045854531615 (MSB first)
        272709516702285 (LSB first)
  Bin:  0b101100100100110101111011100001001110000000011111 (MSB first)
        0b111110000000011100100001110111101011001001001101 (LSB first)
HDR_MARK+HDR_SPACE+101100100100110101111011100001001110000000011111
  Bits: 48
  Hex:  0xB24D7B84E01F (MSB first)
        0xF80721DEB24D (LSB first)
  Dec:  196045854531615 (MSB first)
        272709516702285 (LSB first)
  Bin:  0b101100100100110101111011100001001110000000011111 (MSB first)
        0b111110000000011100100001110111101011001001001101 (LSB first)

Total Nr. of suspected bits: 96
```

crankyoldgit, 2018-07-22:
FYI Here is another example of another A/C manufacturer using the Coolix protocol. See #484

closed 2018-07-22 by crankyoldgit

jimmys01, 2019-03-14:
This resolves the confusion i had when i was replaying those messages, mistakenly thinking that the midea and coolix protocols where mixing up.

CRCinAU, 2019-03-14:
I also thought my A/C was Midea - see: #624 

The unit would 'sort of' understand if I sent it Midea, but not quite function.... Until we figured out it was using Coolix protocol...