crankyoldgit/IRremoteESP8266 #498: Decoding Blue Star AC Remote

just24nar, 2018-08-14:
auto_analyse_raw_data.py (v2.4.3)

following is the output I'm getting for the auto_analyse_raw_data.py script for my BlueStar AC remote.

data Found 199 timing entries.
process.js:57
Potential Mark Candidates:
[4308, 600]
Potential Space Candidates:
[4998, 4172, 1550, 492]
Guessing encoding type:
Looks like it uses space encoding. Yay!
Guessing key value:
HDR_MARK   = 4308
HDR_SPACE  = 4172
BIT_MARK   = 600
ONE_SPACE  = 1550
ZERO_SPACE = 492
SPACE_GAP = 4998
Decoding protocol based on analysis so far:
HDR_MARK+HDR_SPACE+101100100100110101111011100001001110000000011111GAP(4998)
  Bits: 48
  Hex:  0xB24D7B84E01F (MSB first)
        0xF80721DEB24D (LSB first)
  Dec:  196045854531615 (MSB first)
        272709516702285 (LSB first)
  Bin:  0b101100100100110101111011100001001110000000011111 (MSB first)
        0b111110000000011100100001110111101011001001001101 (LSB first)
HDR_MARK+HDR_SPACE+101100100100110101111011100001001110000000011111
  Bits: 48
  Hex:  0xB24D7B84E01F (MSB first)
        0xF80721DEB24D (LSB first)
  Dec:  196045854531615 (MSB first)
        272709516702285 (LSB first)
  Bin:  0b101100100100110101111011100001001110000000011111 (MSB first)
        0b111110000000011100100001110111101011001001001101 (LSB first)
Total Nr. of suspected bits: 96

For my current project i need to be able to manipulate the data, calculate my own checksum for the data and send it to the AC, can you please tell me how do I calculate CHECKSUM for the above Data??

I've been able to calculate checksum for several remotes using the method given in this website - https://electronics.stackexchange.com/questions/183785/understanding-ir-protocol-checksum-generation

However the i'm unable to do so for this remote, also can you tell me what is the significance of "GAP(4998)" mentioned in the results ?

Thanks

crankyoldgit, 2018-08-14:
`GAP(4998)` means there is a gap (space) of ~5000usecs between two sections of the message.
Looking at the results you've list above, the message/data repeats with identical data. So that gap is _probably_ the space between two repeating messages.

Are you sure there is even a checksum for the 48 bits?
If you want someone to help you work out the checksum (if there is one), you will need to supply a lot more samples of messages .. and what the "other" bits mean.

If there is a checksum, it is likely to be the last transmitted 4 to 8 bits of the message.

just24nar, 2018-08-27:
> Just following up. Did you get further?

I didn't get enough time to test this out but I think you provided me just
enough information to solve this issue. Thanks !

crankyoldgit, 2018-09-09:
Another follow up. How's it going? Got any more data to share?

crankyoldgit, 2018-09-28:
Marking this issue stale, and closing. If you update it, I'll reopen.

closed 2018-09-28 by crankyoldgit