crankyoldgit/IRremoteESP8266 #499: decode a led light remote

alinelena, 2018-08-16:
Dear All,

I try to decode a generic ir rgb light.
this one https://www.gearbest.com/smart-lighting/pp_1020748.html?wid=1433363&currency=GBP&vip=4264385&gclid=CjwKCAjwwdTbBRAIEiwAYQf_E618hO2kVp6ONgH-AFFdmBUcJabhNd-32dNJZFPAbrw1KwVKC7B6sRoCwdEQAvD_BwE

i have got IrevvDump2 and collected few samples around 15 in two cases buffer size 128 and 256.

```
Timestamp : 000536.773
Encoding  : UNKNOWN
Code      : C39056E3 (36 bits)
Library   : v2.4.3
```

when I try auto_analyse_raw_data.py on the 128

```
Found 81 timing entries.
Potential Mark Candidates:
[9316, 5326, 898, 668, 370, 150]
Potential Space Candidates:
[40118, 9064, 2102, 1882, 828, 526, 322, 14]
DANGER: Unexpected and unused mark timings!
Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK   = 9316
HDR_SPACE  = 526
BIT_MARK   = 150
ONE_SPACE  = 322
ZERO_SPACE = 14
SPACE_GAP1 = 40118
SPACE_GAP2 = 9064
SPACE_GAP3 = 2102
SPACE_GAP4 = 1882
SPACE_GAP5 = 828

```

when on the 256 something like this

```
Found 56 timing entries.
Potential Mark Candidates:
[40416, 20956, 5510, 3134, 2262, 2034, 756, 546, 16]
Potential Space Candidates:
[8956, 672, 442, 148]

Guessing encoding type:
Sorry, it looks like it is Mark encoded. I can't do that yet. Exiting.
```

obviously this is not good....so any way on putting me in the right direction will be useful. 

as best practice. 
1. how many samples areconsidered ok?
2. shall one sample different keys or the same key?

as a side effect i discovered that one of my old dvb-t tunner remote control can change colours and modes in the bulb. unfortunately after opening it i could not see anything on chip that helps. 

samples
[codes-256.txt](https://github.com/markszabo/IRremoteESP8266/files/2293329/codes-256.txt)
[codes-128.txt](https://github.com/markszabo/IRremoteESP8266/files/2293326/codes-128.txt)

crankyoldgit, 2018-08-16:
Looking at the results in `codes-128.txt` you'll notice a lot of short pulse signals. I think you are probably experiencing a frequency mismatch. See the FAQ on this ([I'm getting some random odd values <50usecs in my capture of an IR message...](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#Im_getting_some_random_odd_values__50_usecs_in_my_capture_of_an_IR_message_What_is_up_with_that))

In short, as is, you can't trust the raw captures you are getting at present. There is far too much corruption in the results to be of much use. Basically, you need to clean up the signal as presented to the ESP8266 from the IR receiver. That needs to be addressed before you can even try to decode this signal.

It's most likely a frequency mismatch, but it could be a number of other things like a bad circuit/module/cpu etc.

crankyoldgit, 2018-08-16:
BTW, what is the part number of the IR receiver module you are using, and how are you using it? (i.e. circuit diagram)

crankyoldgit, 2018-08-16:
Also, try not holding the remote so close to the receiver. That's another potential cause.
FWIW, those cheap remotes often are 38kHz ones and often use the NEC protocol, in my experience.

alinelena, 2018-08-16:
problem solved... the receiver was the problem. Got a better one and I have managed to catch the signal. Indeed the protocol it is NEC, using only last 4 bytes and totally ignoring the first 4 bytes. I was able to send back message too.

Thank you for your help.

crankyoldgit, 2018-08-17:
Glad to hear you found success. Marking this issue closed.

closed 2018-08-17 by crankyoldgit