crankyoldgit/IRremoteESP8266 #500: How to store raw data?

ayushsharma82, 2018-08-20:
### Version/revison of the library used
v2.4.3

How can i store the Raw data coming from IR receiver into a unit16 array?

crankyoldgit, 2018-08-21:
Can you please elaborate more on what you are trying to do?

The `IRrecvDumpV2` example code will tell you what the raw array is, assuming you want to compile in static values.

If you want to do it dynamically, you'll need to copy the contents of the array, like you would any other array-copy operation in C++. e.g. A quick google search gives this: http://www.cplusplus.com/forum/general/199358/

ayushsharma82, 2018-08-21:
I am trying to save my remote codes into a JSON File.
for that i need an array of raw IR Codes. Example the uint16_t array which is returned to Serial by resultToSourceCode(); function.

My Question is how can i get the uint16 array from decode_results.

crankyoldgit, 2018-08-21:
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h#L95

Assuming `capture` is an instance of the `decode_results` class.
e.g.
```
decode_results capture;  // An instance of the decode_results class.
```

Then `capture.rawbuf` is how you reference the "array". i.e. It's a pointer to the first element of the array. `capture.rawbuf[0]` is the first 16 bit element. Note: `capture.rawbuf[1]` is the first *real* element of the raw capture, ignore element 0. Remember that `capture.rawlen` is how long the raw capture was. Thus, `capture.rawbuf[capture.rawlen - 1]` is the *last* element of the raw capture.

crankyoldgit, 2018-08-22:
Oh, while I think of it. Please note that the contents of that array (`decode_results.rawbuf`) is in ticks, NOT usecs. You'll need to convert it. e.g. Multiply each value by [RAWTICK](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h#L34)

Per previous example:
```
first_mark_signal_in_usecs = capture.rawbuf[1] * RAWTICK;  // Rawbuf is the number of ticks, not Nr. of usecs.
```

ayushsharma82, 2018-08-27:
This is how i did it: 

```
uint16_t rawdata[] = {};

for(int i=0; i < sizeof(capture.rawbuf); i++){
   rawdata[i] = capture.rawbuf[i] * RAWTICK;
}
``` 

Next time when you want to get data. you can get size of array rawdata by calling `sizeof(rawdata)`

Thanks for all the support.

crankyoldgit, 2018-08-28:
FYI, a few caveats with that code:
1) You may be copying more data than you need to. That code copies the entire array, even parts that were not used. If you copy using the length stored (capture.rawlen) it will always copy less, thus being faster/more efficient.
2) You can start with `i=1`, the first entry (i.e. 0) is never used.
3) You're going to get corrupt data if you encounter an entry in capture.rawbuf[] that is >= 32768, as 2 * 32768 = 65536 which DOES NOT fit into a `uint16_t`, you should use a `uint32_t` instead. Values that large are uncommon, but are possible.
4) Be cautious when using `sizeof()`. It is often only calculated at compile-time depending on the compiler used. `capture.rawbuf` is/can be dynamically allocated. Thus you may get an incorrect length. i.e. Your code may not be very portable.

closed 2018-08-28 by crankyoldgit

matbks, 2019-07-03:
I only want to send this part of the result to firebase: 
uint16_t rawData[41] = {2694, 864,  498, 840,  502, 368...}

how can i save only this part in esp32?

crankyoldgit, 2019-07-03:
@matbks See https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me