crankyoldgit/IRremoteESP8266 #505: Decoding Samsung AC IR Remote

Hollako, 2018-08-26:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revison of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._

### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar branded BBQ turns on and cooks me some ribs.

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2018-08-27:
See https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#The_libraryexample_code_doesnt_understand_my_Air_Conditioner_remote_

It's missing because it requires a) access to the device (which you probably have) and b) a lot of time/effort.

For an example of the process we go about for adding support, see #485 
If you want the support, you are going to have to put some effort in to capture and pull apart the message codes to work out which bits control the various functions. I and others have found creating a spreadsheet helps to try to decode by hand what is in the message. Once we have that worked out, then adding the code is then possible.

I'm happy to help with the code aspects, but doing the initial message decoding is best done by someone who has the remote/device to hand.

Hollako, 2018-08-27:
Sure no problem, I am ready to help with small hints from your side i will start the decoding process. as a first step i should get the Raw Data then convert them to bits (1 or 0), please advise if am wrong ?!?

crankyoldgit, 2018-08-27:
What does the output of `IRrecvDumpV2` look like?
If it is only `UNKNOWN`, then yes, the Raw Data is what you are going to have to use.
To help convert that to binary, try using the [auto_analyse_raw_data.py](https://github.com/markszabo/IRremoteESP8266/blob/master/tools/auto_analyse_raw_data.py) program.

If `IRrecvDumpV2` recognises it as a named protocol (e.g. SAMSUNG), then you likely can just use/convert the hex value.

Hollako, 2018-08-28:
Hello,

true, It gives me Unknown.

On the other hand I have another AC remote where the output of the IRecvDmpV2 is Coolix. However i couldn't the commands where i can send the Coolix codes separately (Mode, Temp, Fans, etc..) any advise please ?

crankyoldgit, 2018-08-29:
The Coolix protocol hasn't been fully reverse engineered yet. There is some work to decode the Coolix protocol in Issue #484 - Feel free to contribute to there to help decode the bit patterns/values.

Hollako, 2018-09-02:
Hello,
Actually for the moment am focusing on the Samsung codes, once done will contribute on coolix for sure.

However, am facing some difficulties on the decoding, i used the python file to convert the raw to bits. first i found several outputs after conversion. am not sure the Bin sequence that am using is the correct one. then i tried to compare between the temperatures with same Mode Cool, Fan Auto, Swing Auto. but things still not clear enough despite that i found some bits are changing where they probably are the temperature bits but not enough.

For example i detected the same code twice with same temperature then i copied the Raw sequence extracted from IRrecvDumpV2 sketch then i converted these Raw signal in order to get the same Bin sequence but unfortunately I got 2 different Bin sequence with typically same IR  code.

Please any hint to go Further with the correct reversing engineering of an AC IR codes?

crankyoldgit, 2018-09-03:
https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue
Especially see points 1, 8, 11, & 12. ;-)

i.e. Can you give us some data/examples? It's hard(er) to understand what is going on without actual data.

crankyoldgit, 2018-09-03:
Can you include here the text for the lines you are giving as input to the auto analyse program?
e.g. like the last line of your screen shot. `uint16_t rawData[etc] = {......};`

Especially for the two messages that you say are the same, but decode differently.

FYI, text is far better than a graphical screenshot, as it allows people to use the data you provide to try to duplicate or analyse the experience you are having.

Typically, if the state of the remote is the same when the same button is pressed, we would expect exactly the same message both times. Note that the "clock" on the remote may have changed and thus the message may change too if the time is included.

So, as indicated earlier. More data please. Don't be afraid to go overboard in that respect.

From what I can see so far, these samsung messages certainly are an unusual protocol so far.
What model is the A/C Unit and the Remote?

Hollako, 2018-09-03:
Hey again, a good news, i tried the heat mode with temp from 16 to 30 and i could recognize the temperature bits clearly.

A very hint you regarding the time of the remote that may vary the code as well.

And referring the raw data are all included in the excel sheet i sent but i was not sure if i am on the right track by decoding the AC IR. Hopefully now am in a step forward. And i need to make sure that the code is 56 bits, is it correct ?

[Samsung Decoding.xlsx](https://github.com/markszabo/IRremoteESP8266/files/2344454/Samsung.Decoding.xlsx)

Anyway i moving forward to know the rest of the code (mode, fan, swing).

Hollako, 2018-09-03:
True 112 bits but the first sequence is exactly the same between all the codes. So am working on the second sequence only. then we can add the first one as it is.

crankyoldgit, 2018-09-03:
@Hollako If you try the branch https://github.com/markszabo/IRremoteESP8266/tree/v2.5-dev-samsung-ac it should hopefully help you with your decoding efforts. It's based on the data I as able to get from your spreadsheet.

Hollako, 2018-09-04:
Hello,

I am facing an issue with the new "auto_analyse_raw_data.py" file am not able able to convert the raw to bin sequence anymore. and here is below the error in python:

> Traceback (most recent call last):
  File "C:\Python27\Lib\site-packages\pythonwin\pywin\framework\scriptutils.py", line 326, in RunScript
    exec codeObject in __main__.__dict__
  File "C:\Users\Ehanna\Desktop\Samsung Decoding\auto_analyse_raw_data.py", line 7, in <module>
    import argparse
  File "C:\Python27\lib\argparse.py", line 86, in <module>
    import copy as _copy
  File "C:\Python27\lib\copy.py", line 52, in <module>
    import weakref
  File "C:\Python27\lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

Help please !!

crankyoldgit, 2018-09-04:
Which `new "auto_analyse_raw_data.py" file` are you using? The one in the `v2.5-dev-samsung-ac` branch?

And what command line arguments (and files/data) are you giving it?

I just tested the one in `v2.5-dev-samsung-ac` and it's working fine for me on a linux system.

Looking at the Traceback provided, it seems the problem is in the `argparse` library, not with the analyse script. i.e. Check your python 2.7 installation etc.

crankyoldgit, 2018-09-04:
P.S. In theory, if you use `IRrecvDumpV2` in that branch, you shouldn't need they analyse program anymore. It will give you the answer in HEX. etc.

Hollako, 2018-09-04:
The python version installed is 2.7.14, wired.

Anyway i followed your steps and replace the `IRremoteESP8266` with `v2.5-dev-samsung-ac` branch and now i am getting the Encoding SAMSUNG_AC and HEX number as you mentioned above.

So the next step now is to convert the HEX numbers into binary? and continue the decoding.

Many thanks for your support, maybe i am asking some silly questions but this is my first time in decoding IR codes. hopefully later on i will be more efficient in this community to enrich this library with more and more of A/C codes.

crankyoldgit, 2018-09-04:
Re: Converting. It depends. Some times some people find it easier to work in Hexidecimal, some people prefer binary. There are plenty of hex <=> bin converters/calculators out there.

If you look at #509 you can see some other tips/approaches/gotchas for working out decoding the state into the functions of the A/C unit.

I often recommend working out the bit order (the code you are using assumes MSB at the moment, it could be wrong) by trying to workout the Temperature feature. Often, but not always, the temperature bits/byte/nibble increments by 1 for each extra degree. That sequencing helps determine which order the bits are in. Usually.

Hollako, 2018-09-04:
Hello,

I think am done with the converting into binary for all the main function, Power, Temperature, Mode, Fan Speed and Swing. I adopted the `IRremoteESP8266` with `v2.5-dev-samsung-ac` branch to finalize it cause i don't know hwy the Python is not working anymore with m. But the good thing is that i could continue with the `IRrecvDumpV2` of the Samsung AC branch. and yes i am using the MSB.

Please find attached the excel file "Summary" sheet.

So now what;s next ?
[Samsung AC Decoding.xlsx](https://github.com/markszabo/IRremoteESP8266/files/2349466/Samsung.AC.Decoding.xlsx)

crankyoldgit, 2018-09-05:
Thanks. Looking at the fan data. It's clear the fan data is in LSB First order. So I've pushed a change to that branch to reverse the bit order.
That will mean all the hex/binary data will change. You'll need to update/re-capture.

I noticed you've marked a 4-bit section of data as Unknown. In order for us to succeed we will **need** to know why those bits are changing before we can make new messages.
There are several possibilities for them:
1) They are a checksum. But it's very odd as it's in the middle of the message. Typically a checksum is at the end.
2) The time on the remote.
3) Some other function.

If it is the checksum, you need to calculate how it's derived before we can construct a message from components. This is often the most frustrating and challenging part.
What you want to do is to see if you can change a single bit elsewhere in the message, to see if those checksum bits change. Then work backwards to see how you could calculate those changing values.
Here are some examples of checksum algorithms I've seen in other protocols.
* The sum of all the previous/other bytes but modded (i.e. %) to fit into a the size of the checksum.
* The sum of all the lower half of all the bytes and modded to fit into a the size of the checksum.
* The XOR of the previous bytes ...
* The 2's compliment of the bytes ...
Basically, have a look through some of the other protocols to see how they handle checksums.

FYI, Due to the nature of the checksum calc, you can't obviously include the bits of the checksum in the calculation of itself. i.e. You can safely exclude them.

Hollako, 2018-09-05:
Hello,

First let me understand why it's changed to LSB, for my when i was working on the python file i was copying the MSB binary sequence. and when i continue with the new branch including the Samsung AC codes the sequence still same and the same bits still at same place. Maybe the enumeration of the bits are wrong cause i started from 112 on left to 1 on right. maybe i should swap the enumeration.

Regarding the unknown bits, is it obligatory to know them? cause there is more bits like Comfort, Beep, Quiet etc... i clarified which bits but i don't know if it's necessary to use them or consider them in IRRemote library.

However, the checksum is a new challenge for me but what if they are Time bits? should we consider them as well ?

Thank you,

crankyoldgit, 2018-09-05:
To see if it is a time code, try to produce the same "message" (i.e. all the same hex values) with the remote in the same state .. but only with a different time on the remote.
e.g.
Send a "on cool 21C fan auto etc" at 12:01 ... and then at 12:02 ... etc. If it is a time code, then it will change, if it isn't the messages will be the same,

crankyoldgit, 2018-09-05:
In short. If any bits change when when we press a button that we want to emulate, we need to understand and explain why they've changed.

Hollako, 2018-09-05:
Well noted, will go further to know the unknown bits.

Hollako, 2018-09-10:
Hello,

Sorry for this delay, have been busy lately.

Anyway I did clarified the "Unknown" bits and i can confirmed that they are a checksum bits. But couldn't know how they are calculated specially that these bits are in the middle of the message and not at the end or beginning.

I need a small tip to know how to start please !

Thank you,

crankyoldgit, 2018-09-10:
Some google searches for "samsung air conditioner ir codes" highlights a number of Arduino implementations. Have a look at what some of them have for the checksum calc.

It looks like it _might_ be the number of '1's in the binary before the checksum value, or something related to that. That final number may be a inverted or a 2's compliment but it's a start for some hints.

crankyoldgit, 2018-09-10:
E.g. https://stackoverflow.com/questions/11863920/samsung-ir-codes-checksum

crankyoldgit, 2018-09-20:
Marking this issue resolved due to low-feedback and that the experimental code has now been merged into 2.5, which will be released soon.

Hollako, 2018-09-20:
Hello,

I could know the relation between the Checksum (bits 41 to 44) and the bits 1 to 40 but without knowing exactly how it's calculated.

But am sure that the number of bit 1 is giving the same checksum.

For example when the of bit 1 (between bit N°1 to bit N°40) is 18, the checksum is 0111.
another example if the bit 1 number is 20 the checksum is 0011.

I have attached the excel file in the "Mode" sheet many sequence insure this relation but till now i couldn't know the algorithm of this calculation.

Help please !!

[Samsung AC Decoding.xlsx](https://github.com/markszabo/IRremoteESP8266/files/2400000/Samsung.AC.Decoding.xlsx)

crankyoldgit, 2018-09-20:
v2.5.0 has been released.

Hollako, 2018-09-23:
Hello,

First of all Many thanks for your support and effort.

I tried the V2.5.0 version and i found small issues regarding the Samsung IR codes, listed below:

1- The Power Off is not resolved n the `IRrecvDumpV2` and giving unknown code.
2- the Fan speed values are 0 for Auto, 2 for Low, 4 for Mid and 5 for High.
3- the Swing Mode is changing between On and Off (values 0 and 1) but the transmitted code is the same, and i receive the same code in `IRrecvDumpV2`

The Fan Speed is not as a high priority but Power Off and Swing Mode are not working. I am ready to help if something still missing from my side.

Thank you,

crankyoldgit, 2018-09-23:
Can you please provide the entire output of IRrecvDumpV2 for each signal that has an issue? I need that information to determine what might be wrong.

1) See above. I'll need that to fix it.
2) The numbers were based on your spreadsheet data. If you disagree and think I've mis-read the data, please let me know.
3) Thanks. I'll look into it, but the tests I've done seem to indicate it is changing the state and the message. Again, please provide the the Dump output for the signals, and the code you are using to set & change the Swing options.

Hollako, 2018-09-24:
Hello,

For the Fan Speed, noted. I didn't know that you follow the binary sequence and convert it to decimal.

As for the swing mode, Please find attached the excel file, in the swing sheet you will find the binary sequence where i clarified the swing bits 34,45,46. When move the bits are 010 and when stop the bits are 111. Even i tried 2 (decimal of 010) and 7 (decimal of 111), same results.

And for the Power also please look at the fan sheet where i clarified the power bits which are bits number 99, 59 and 60. Power Off ROW.

[Samsung AC Decoding.xlsx](https://github.com/markszabo/IRremoteESP8266/files/2409578/Samsung.AC.Decoding.xlsx)

crankyoldgit, 2018-09-24:
Can you please just cut and paste the dump output here?

crankyoldgit, 2018-09-24:
As for swing, The spreadsheet says to me it's bits 34,35, & 36. Where is the stuff saying it is 34, 45, 46?
Are you attaching the correct spreadsheet? I think a better solution is to convert it into a Google Sheets document and make it read-only/public and put the link here.

Hollako, 2018-09-24:
Hello,

Regarding the swing is my typo mistake 34, 35 and 36. And i will try the boolean codes.

Regarding the rest codes i will attache them once ready.  

Thank you.

crankyoldgit, 2018-09-24:
@Hollako thanks for that data. It's most helpful.

crankyoldgit, 2018-09-25:
I've found the bug for why setting and reading the Swing status wasn't working. I changed the wrong byte (8 instead of 9).
However, your data showed there is another bit changing that you haven't explained/documented.
By "your" naming, it would be bit 40. It changes between the two settings/samples.

e.g.:
Swing On:  `02920F000000F00102AF718011F0`
Swing Off: `02920F000000F001E2FE718011F0`

The difference is bytes 8 & 9 (first byte reading from the left is Byte 0.):
On:  `02AF`
Off: `E2FE`

Byte 8 contains the checksum, so you can ignore that change.
But byte 9 changes from 0xAF (On) to 0xFE (Off). The first nibble (top 4 bits) of each byte (`A`/on and `F`/off respectively) is explained by your spreadsheet/comments about how swing works. However the lowest bit of that byte (Your bit 40) changes. i.e. `F` to `E`)
@Hollako You **need** to explain that.

I'll upload a commit/branch where I'm working on the corrections in a sec.

crankyoldgit, 2018-09-25:
FYI, I'm working on the fixes in the [samsung-ac-fixes](https://github.com/markszabo/IRremoteESP8266/tree/samsung-ac-fixes) branch.

crankyoldgit, 2018-09-25:
Regarding the power on/off. Yes, the new data you've provided indicates there are a total of three 56-bit sections in those messages, rather than the typical two. (think 21 bytes, instead of the normal 14)
From the two data samples you've provided, the middle one is the "new" one, and it seems fixed at least.
Decoding/reading it shouldn't be to hard a problem, but creating/sending/knowing when to send it is the **harder** part.

That's a significant change, and I'll have to think about how best to support that.

crankyoldgit, 2018-09-25:
First pass added for supporting the unusual length Samsung A/C message has been added to that branch. @Hollako Can you please try it out?

Hollako, 2018-09-26:
Dear David,

Regarding the Swing, it's done.

But for the power mainly also is done but i have noticed a difference between the code sent from the Original Samsung Remote and the code sent from the ESP8266 IRTX device based on the new IRremote8266 library.

For now am doing my tests on on the `IRrecvDumV2` and not a real Samsung AC Split. Please find below the code received from the Original Samsung AC remote and the IR code send by the ESP8266 IRTX.

Original Samsung Remote:
Power On
```
Timestamp : 001639.550
Encoding  : SAMSUNG_AC
Code      : 02920F000000F001D20F0000000001E2FE718011F0 (168 bits)
Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Beep: Off, Clean: Off, Quiet: Off
Library   : v2.5.0

Raw Timing[349]:
   +   658, - 17896,    +  3026, -  8942,    +   562, -   462,    +   538, -  1458, 
   +   540, -   462,    +   536, -   462,    +   536, -   464,    +   536, -   488, 
   +   512, -   486,    +   516, -   484,    +   560, -   440,    +   536, -  1434, 
   +   562, -   462,    +   536, -   462,    +   538, -  1434,    +   564, -   460, 
   +   538, -   462,    +   538, -  1456,    +   540, -  1436,    +   562, -  1438, 
   +   560, -  1456,    +   540, -  1462,    +   536, -   488,    +   534, -   464, 
   +   538, -   462,    +   538, -   462,    +   536, -   462,    +   536, -   462, 
   +   538, -   460,    +   540, -   460,    +   538, -   460,    +   540, -   458, 
   +   540, -   458,    +   538, -   460,    +   538, -   462,    +   538, -   460, 
   +   538, -   460,    +   538, -   462,    +   538, -   464,    +   534, -   488, 
   +   512, -   486,    +   484, -   516,    +   536, -   462,    +   538, -   460, 
   +   538, -   460,    +   536, -   462,    +   538, -   460,    +   540, -   458, 
   +   540, -   460,    +   538, -   460,    +   538, -   460,    +   538, -   460, 
   +   540, -   458,    +   538, -   460,    +   538, -  1454,    +   544, -  1482, 
   +   514, -  1462,    +   552, -  1444,    +   564, -  2942,    +  3058, -  8940, 
   +   562, -  1436,    +   564, -   460,    +   538, -   460,    +   540, -   458, 
   +   540, -   458,    +   540, -   458,    +   540, -   458,    +   540, -   458, 
   +   540, -   458,    +   540, -  1452,    +   544, -   462,    +   536, -   486, 
   +   512, -  1460,    +   560, -   466,    +   536, -  1436,    +   562, -  1434, 
   +   566, -  1434,    +   564, -  1432,    +   564, -  1434,    +   566, -  1434, 
   +   564, -   458,    +   540, -   458,    +   540, -   460,    +   540, -   460, 
   +   538, -   486,    +   512, -   486,    +   484, -   516,    +   534, -   466, 
   +   536, -   462,    +   538, -   462,    +   536, -   462,    +   538, -   460, 
   +   540, -   458,    +   540, -   458,    +   540, -   458,    +   540, -   458, 
   +   540, -   458,    +   540, -   460,    +   538, -   460,    +   540, -   460, 
   +   540, -   458,    +   540, -   460,    +   538, -   486,    +   512, -   486, 
   +   482, -   516,    +   532, -   466,    +   536, -   464,    +   536, -   462, 
   +   538, -   460,    +   540, -   460,    +   538, -   460,    +   540, -   458, 
   +   540, -   458,    +   540, -   460,    +   540, -   458,    +   540, -   458, 
   +   540, -  2944,    +  3052, -  8962,    +   542, -  1456,    +   544, -   486, 
   +   512, -   486,    +   516, -   482,    +   482, -   516,    +   534, -   464, 
   +   538, -   462,    +   538, -   460,    +   540, -   460,    +   540, -  1436, 
   +   562, -   458,    +   540, -   458,    +   540, -   460,    +   540, -  1434, 
   +   564, -  1454,    +   542, -  1456,    +   544, -   486,    +   512, -  1482, 
   +   518, -  1456,    +   558, -  1438,    +   564, -  1436,    +   564, -  1434, 
   +   564, -  1434,    +   564, -  1432,    +   588, -  1412,    +   562, -   460, 
   +   540, -   458,    +   540, -   460,    +   540, -  1482,    +   516, -  1482, 
   +   512, -  1464,    +   558, -   464,    +   536, -   462,    +   536, -   462, 
   +   540, -   460,    +   540, -   458,    +   562, -   436,    +   542, -   458, 
   +   540, -   458,    +   540, -  1456,    +   542, -  1454,    +   544, -   460, 
   +   562, -   436,    +   540, -   462,    +   536, -  1482,    +   516, -   484, 
   +   482, -   516,    +   534, -   464,    +   536, -   464,    +   536, -   462, 
   +   540, -   460,    +   540, -   458,    +   540, -  1454,    +   542, -  1436, 
   +   562, -  1436,    +   562, -  1456,    +   540

uint16_t rawData[349] = {658, 17896,  3026, 8942,  562, 462,  538, 1458,  540, 462,  536, 462,  536, 464,  536, 488,  512, 486,  516, 484,  560, 440,  536, 1434,  562, 462,  536, 462,  538, 1434,  564, 460,  538, 462,  538, 1456,  540, 1436,  562, 1438,  560, 1456,  540, 1462,  536, 488,  534, 464,  538, 462,  538, 462,  536, 462,  536, 462,  538, 460,  540, 460,  538, 460,  540, 458,  540, 458,  538, 460,  538, 462,  538, 460,  538, 460,  538, 462,  538, 464,  534, 488,  512, 486,  484, 516,  536, 462,  538, 460,  538, 460,  536, 462,  538, 460,  540, 458,  540, 460,  538, 460,  538, 460,  538, 460,  540, 458,  538, 460,  538, 1454,  544, 1482,  514, 1462,  552, 1444,  564, 2942,  3058, 8940,  562, 1436,  564, 460,  538, 460,  540, 458,  540, 458,  540, 458,  540, 458,  540, 458,  540, 458,  540, 1452,  544, 462,  536, 486,  512, 1460,  560, 466,  536, 1436,  562, 1434,  566, 1434,  564, 1432,  564, 1434,  566, 1434,  564, 458,  540, 458,  540, 460,  540, 460,  538, 486,  512, 486,  484, 516,  534, 466,  536, 462,  538,  [... cut ...] };  // SAMSUNG_AC
uint8_t state[14] = {0x02, 0xB2, 0x0F, 0x00, 0x00, 0x00, 0xC0, 0x01, 0xE2, 0xFF, 0x71, 0x00, 0x0D, 0xF0};
```

In both cases the `IRrecvDumV2` detects the power status either On or Off but the code length sent by the Original Samsung Remote is not the same one sent by the ESP8266 IRTX.

I didn't tried it yet on a proper Samsung AC Split for now, will try it soon but i just noticed this difference of the IR code transmitted between the Samsung remote and ESP8266 IRTX.

Thank you

crankyoldgit, 2018-09-26:
Two things:
1) Can please ensure you've `git pull`ed the latest in that branch? I think you might be slightly behind.
2) Can you list the code you are using to send the on/off message? If you are doing it from inside the `IRSamsungAC` class via the `send()` method, then it isn't going to work. That only sends the 112 bit message at present.
For now, you'll need to do:
```
uint8_t state[21] = {0x02, 0xB2, 0x0F, 0x00, 0x00, 0x00, 0xC0, 0x01, 0xD2, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x01, 0x02, 0xFF, 0x71, 0x80, 0x11, 0xC0};  // Taken from your Original Remote "Power off" above.
irsend.sendSamsungAC(state, 21);  // Note: 21, not the usual 14.
```

Hollako, 2018-09-26:
Well Done !!

What i did is when i want to send the power state, i send irsend.sendSamsungAC(stateon, 21) or irsend.sendSamsungAC(stateoff, 21). and the stateon or stateoff are 2 unit8_t constants defined at the beginning of the sketch.

And when i want to send the rest of the commands i use IRSamsungAc.send().

It working perfectly with the IRrecvDumpV2, hopefully today i will try it with a real Split Samsung AC to confirm 100% that Samsung AC codes are fully functional.

Thank you for your effort and support.

crankyoldgit, 2018-09-26:
Glad it's working for you.
Just added an `IRSamsungAc.sendExtended()` method to the class in that branch.
It should send a message equiv of `irsend.sendSamsungAC(internal_state)` but as `irsend.sendSamsungAC(internal_state, 21)`

crankyoldgit, 2018-09-26:
Yes, the real tests are on the actual device itself. Looking forward to the results.

Hollako, 2018-09-28:
Hello Again,

I could test the IR codes on a Split Samsung AC and i got a half success result unfortunately.

Only Power Off and On commands are working. All the rest is not working, noting that exactly the same code received from the Samsung AC remote i could receive them via the ESP8266 using the `IRrecvDumpV2`, exactly same Hex Code received. Really weird and couldn't understand why.

Power On and Off using the  `irsend.sendSamsungAC(stateon, 21)`  command while the rest i used `IRSamsungAc.send()`. Is there any difference between them, maybe you did some improvements and i used the last revision before you added `IRSamsungAc.sendExtended()`. Do you think that these improvements solved the issue cause i didn't use the `IRSamsungAc.sendExtended()`.

Any advise please ?!?

A small note David my name is Hollako not Hallako, i noticed that you mentioned someone else in 
[https://github.com/markszabo/IRremoteESP8266/pull/529]

Regards,

crankyoldgit, 2018-09-28:
First off, sorry for the typo in your name. My bad. Fixed.

>  Is there any difference between them? [edited]

Yes. `irsend.sendSamsungAC()` differs from `IRSamsungAc.sendExtended()` & `IRSamsungAc.send()` in the former case **does not** try to construct a state/settings, it only uses a captured state you've recorded. The latter two both try to build a completely new state/settings from scratch.

It's most like that the building from scratch part is not working properly yet.

> Do you think that these improvements solved the issue cause i didn't use the IRSamsungAc.sendExtended().

No. sendExtended() is only for those messages which produce a state[21], not a state[14].

> Any advise please ?!?

As I don't have access to the remote (but you do ;-), I need to you capture some messages using the dev code & the original remote, and report what the full `state[]`s are for them, and exactly what **all of the settings** are set on the remote for each one. We don't need the rawData lines. Just the state lines.
Sort of like what you did for the original spreadsheet, however this time all the numbers have changed.
Note: This is separate from what I'm asking you to do below, but do the below stuff first, if that's wrong we have more fundamental problems. The above bit is for making sure we are able to do custom state creation correctly.

I can then try to construct completely synthetic versions (i.e. changing the settings of the A/C via setMode() etc) to see if I can get it to produce the same `state[]` code.

In the meantime, there are some other things you can do to help test to see if it is a different problem.

1) Capture your desired settings (not for one of the power on/off extended sequences, just a normal '14' length setting), and write down **ALL** the settings on the remote.
2) Then try something like:
```
uint8_t the_state_you_just_collected[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0x02, 0xAF, 0x71, 0x80, 0x11, 0xF0};
irsend.sendSamsungAC(the_state_you_just_collected);
```
If that works, then we know sending "normal" codes should work.
3) Then try something like:
```
uint8_t the_state_you_just_collected[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0x02, 0xAF, 0x71, 0x80, 0x11, 0xF0};
IRSamsungAc sammy(IR_PIN_YOU_ARE_USING_TO_SEND_ON);  // Creates the class and which pin to use.
sammy.begin();  // Set everything up so we can start.
sammy.setRaw(the_state_you_just_collected);  // Set the internal state of the class to be what you captured.
sammy.send();  // Send the new internal state as a IR message.
```

In theory, they should all produce the same result. i.e. Doing what ever the collected state/message from the remote was.

If **none** of those work, then there is probably something wrong in our understanding of the protocol, and we'll need to go back to the rawData captures and frequencies etc.

If some of those work, it will confirm we understand the basic protocol (timings/freq etc) and will help point out what may be wrong.

crankyoldgit, 2018-09-29:
@Hollako I've updated the `samsung-ac-fixes` branch, but I doubt any of the changes I've added/made will affect the usage.
Of the limited data I have from your spreadsheet (I had to use the raw data for some message, and even then the description for the message really doesn't match up with the data. i.e. Please be 100% sure what the settings are for the `state[]`s you give above. i.e Double check or triple check.

I've tried constructing messages from _new_, and it seems to work. EXCEPT for the issue I've asked you to look into previously. i.e. the lowest bit in the swing byte (`state[9]`) changes, and doesn't have an explanation e.g. Sometimes 0xAE or 0XAF for swing move etc.
Some of your captures have it as 1 (0xF) or sometimes it's 0 (0xE).

Other than that, my tests as they are seem to all check out. So, the ball is now firmly in your court for more testing and data.

Hollako, 2018-09-30:
Hello,

Hopefully by tomorrow i will do some test on the real Samsung AC split.

But the thing that i don't understand is that in the `IRrecvDumpV2` the codes sent y the Samsung AC and the codes sent by the ESP8266 are typically the same.

So what make the codes from Samsung AC work and the ones from the ESP8266 don't. If the same HEX numbers is generated from both are typically the same !!

Hollako, 2018-10-01:
Hello,

I was decoding the IR codes with the latest Samsung AC library and i noticed that the bits are not the same ones that i decoded previously. and the HEX codes that was issued from the `IRrecvDumpV2` is are not the same as well. For example i had a fix sequence in all the codes which was `4049F000000003` now it's `02920F000000F00` so now the bits numbers are not the same as the previous ones. it's like inverted. for example the temp 24 was 0001 now it's 1000 and even the bits number is not same.

So now I am working on a totally new spread sheet based on the latest library. And hopefully this should solve the problem.

By the way the byte 9 what was confusing us regarding the Swing mode now i got the explanation. and it's true that the Swing mode change the byte 9 between A and F.

crankyoldgit, 2018-10-01:
@Hollako Yes. I have said this a number of times, when we changed to LSBF ordering of the data, the bit patterns/bytes will have changed. Basically as you indicated, they (the bits) are all flipped (order) for each byte. Sorry.

So, yes. Please use the new data format. It will save me having to convert to your one-off system all the time. Oh, and as with other spreadsheets for this sort of thing, I recommend the free Google Sheets spreadsheets as it's web based, no download, and updated in real-time, and if you set the options, people can comment on it directly etc. It's great for collaboration.

> By the way the byte 9 what was confusing us regarding the Swing mode now i got the explanation. and it's true that the Swing mode change the byte 9 between A and F.

Yes, but there is also another bit that changes that you haven't explained that seems linked to Swing some how.

Hollako, 2018-10-01:
Hello,

Sure i will adopt Google Sheet.

Regarding the swing other bits, they are the Checksum. The bits of checksum and Swing are close to each other.

Will upload the sheet once ready !!

crankyoldgit, 2018-10-01:
I'll reiterate what I said earlier:
> Byte 8 contains the checksum, so you can ignore that change.
But byte 9 changes from 0xAF (On) to 0xFE (Off). The first nibble (top 4 bits) of each byte (A/on and F/off respectively) is explained by your spreadsheet/comments about how swing works. However the lowest bit of that byte (Your bit 40) changes. i.e. F to E)
@Hollako You need to explain that.

The checksum is in a different byte. Yes, it's close, but no, I don't think it's the checksum that is changing it. That doesn't seem to fit with the checksum alg. But hey, it could be an overflow. It would be very very odd for it to be though. I've never seen an A/C protocol have a checksum that extends into another byte. But there is always a first time potentially. ;-)

Hollako, 2018-10-01:
Hello David,

here is below the link for the new spread sheet, you can find the Summary Rev.1, Code Rev.1 and Power Code Rev.1 sheets which are the updated ones.

https://drive.google.com/drive/folders/11P3DhyQz7O1-d74jNk1Q5rgmrh4DMw50?usp=sharing

As well i uploaded a word file whee i copied all the results received form the `IRrecvDumpV2`.

Regarding Swing, actually 3 bits are changed from On or Off, 111 for Off and 010 for On. Byte number 10 A for On and F for Off. And these are the only bits (other than the checksum) that are changing between Swing On and Swing On. Please find it in the Code Rev.1 Sheet I have Rows 18 and 19 where is only changed the Swing mode.

thank you,

crankyoldgit, 2018-10-01:
Hey @Hollako, I've converted that spreadsheet to a Google Docs/sheets spreadsheet, given you edit access, and allowed everyone with the link to view it.
That link is: https://docs.google.com/spreadsheets/d/1SlqxuxZpj8TwsZ4lNE5pXCtDx770DGWgN5Yb2aqlgD4/edit?usp=sharing

crankyoldgit, 2018-10-01:
Similarly, here is the Google Doc equivalent of the docx file: https://docs.google.com/document/d/1Sm9azPX9L38OYWIr1PrSEq8YrpN5r3F6kj_6XntmCsA/edit?usp=sharing

crankyoldgit, 2018-10-02:
> And these are the only bits (other than the checksum) that are changing between Swing On and Swing On. Please find it in the Code Rev.1 Sheet I have Rows 18 and 19 where is only changed the Swing mode.

If you go to the spreadsheet, in the 'Raw' tab/sheet, on row 19 (I've highlighted in in the Google Sheets version) it has another bit that changes.

I've used two raw samples from that tab/page to use as unit test data.
See:
https://github.com/markszabo/IRremoteESP8266/blob/samsung-ac-fixes/test/ir_Samsung_test.cpp#L813
&
https://github.com/markszabo/IRremoteESP8266/blob/samsung-ac-fixes/test/ir_Samsung_test.cpp#L854
i.e.
The decode from the raw data you collected to:

Example 1 which is derived from Row 19 of the 'Raw' tab:
```
  uint8_t expectedState[kSamsungAcStateLength] = {
      0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0,
      0x01, 0x02, 0xAF, 0x71, 0x10, 0x41, 0xF0};

```
and Example 2 which is derived from Row 7 of the 'Raw' tab:
```
  uint8_t expectedState[kSamsungAcStateLength] = {
      0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0,
      0x01, 0xE2, 0xFE, 0x71, 0x40, 0x11, 0xF0};
```

The bytes we are interested in/talking about are `expectedState[9]`. i.e. the 10th byte (or 9th when counting from zero)

They differ. e.g. `0xAF` vs `0xFE`. The upper half of they bytes (0xA0 & 0xF0) are explained by your spreadsheet. They are the swing state changing. That part is okay. It's the lower half (0x0F & 0x0E) that are different and require explanation.
I think there are other examples in the raw data you provided.

I've tried to document this as clearly and succinctly as possible here:
https://github.com/markszabo/IRremoteESP8266/blob/samsung-ac-fixes/src/ir_Samsung.cpp#L387
and here: https://github.com/markszabo/IRremoteESP8266/blob/samsung-ac-fixes/src/ir_Samsung.cpp#L393
i.e. Sometimes the swing byte changes for "swing move" not like you've explained.
Yes, all the data in your 'Swing" tab/page is consistent with your description, it's just not all the data you provided is consistent. 

The byte that stores the checksum is: expectedState[8]`. i.e. the 9th byte, ((or 8th when counting from zero)

Sidenote:
FYI, you should really number the bits from the start to the end of the code, not from end of the code to the start.

crankyoldgit, 2018-10-02:
FYI, the fixes thus far have been included in v2.5.1 that I just released.

Hollako, 2018-10-06:
Hello David,

Sorry for the late reply, but didn't have time till today to try the latest release.

At last i can say clearly that everything work perfectly including the swing On and Off.

So Samsung AC IR codes are fully supported now.

Thank you very much.

crankyoldgit, 2018-10-07:
Okay. If you say both sending and receiving with swing etc is all working fine, I'll mark it closed. But I am concerned about that unexplained bit change. i.e. https://github.com/markszabo/IRremoteESP8266/issues/505#issuecomment-426108102

Hollako, 2018-10-08:
Hello David,

I still have a doubt that something is wrong with the swing bits. You mentioned in the 2 Rows that you highlighted it that there is another bit is changing. But these 2 raws was related to the old decoding where i was working on MSB sequence. So this sheet is not valid anymore with the new sequence.

 Am totally sure that the swing mode affecting only 1 byte. which is byte N° 10. When it's Off this byte is 1111 which is F and when is On this byte is 1010 which is A. True that 2 bits are changing but both are related to Swing mode only.

Else if i still don't get your point.

[7 more replies]

closed 2018-10-26 by crankyoldgit