crankyoldgit/IRremoteESP8266 #509: Help Decoding Whirlpool AC Remote

redmusicxd, 2018-08-29:
### Version/revison of the library used
Version 2.4.3

### Expected behavior
I've managed to add the new encoding and now in IRrecvDumpV2 it appears the command but the AC has a mode called 6th Sense it's a full auto mode and if put an on offset of greater than 2 Celsius deegres the hex command stays the same for the up and down as well so i used the  auto_analyse_raw_data.py scrpt and i have a chunk of 56 bits that changes on every command but i don't know exactly how to use that piece of data

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

ON_OFF_6th_mode_0offset_fan_auto

Timestamp : 004520.075
Encoding  : WHIRLPOOL_AC
Code      : C160208E0000 (48 bits)
Library   : v2.4.3

uint16_t rawData[343] = {8952, 4482,  598, 1640,  598, 1644,  596, 532,  596, 534,  596, 540,  600, 536,  592, 550,  602, 1644,  594, 526,  604, 1636,  602, 1642,  596, 536,  594, 540,  600, 538,  602, 540,  600, 530,  598, 522,  598, 526,  594, 1648,  600, 532,  598, 536,  594, 544,  596, 546,  594, 536,  592, 1644,  596, 530,  600, 528,  602, 530,  600, 1650,  600, 1656,  594, 1664,  606, 526,  594, 528,  602, 522,  596, 530,  600, 532,  598, 536,  594, 544,  596, 544,  596, 534,  594, 526,  592, 530,  600, 528,  602, 530,  600, 534,  596, 542,  598, 542,  598, 516,  592, 7924,  602, 1638,  602, 524,  596, 532,  596, 534,  596, 1654,  596, 544,  596, 546,  596, 1652,  598, 1640,  600, 526,  594, 1650,  600, 532,  598, 1654,  596, 544,  596, 546,  596, 536,  594, 528,  592, 530,  598, 530,  600, 530,  600, 534,  596, 540,  600, 542,  598, 532,  598, 524,  596, 526,  592, 534,  596, 536,  594, 540,  600, 538,  592, 550,  602, 528,  600, 520,  600, 524,  596, 532,  598, 534,  596, 538,  592, 546,  594, 546,  594, 536,  594, 526,  592, 532,  598, 528,  602, 530,  600, 534,  596, 542,  598, 540,  600, 532,  598, 522,  598, 524,  594, 532,  598, 532,  596, 538,  592, 544,  596, 544,  596, 536,  594, 1642,  598, 526,  594, 534,  596, 536,  594, 1654,  596, 1686,  574, 1660,  600, 1656,  574, 7918,  596, 526,  594, 530,  598, 530,  600, 530,  600, 534,  596, 542,  598, 542,  598, 532,  598, 1640,  600, 524,  596, 532,  598, 534,  596, 538,  602, 536,  594, 546,  596, 536,  592, 528,  592, 532,  598, 528,  602, 530,  600, 534,  596, 542,  600, 542,  598, 532,  598, 522,  598, 526,  592, 534,  596, 534,  594, 540,  600, 536,  594, 548,  592, 540,  602, 518,  600, 524,  596, 532,  598, 532,  598, 536,  594, 544,  596, 546,  594, 536,  594, 526,  592, 532,  598, 528,  600, 530,  600, 534,  596, 542,  598, 542,  598, 534,  596, 1640,  600, 524,  594, 534,  596, 534,  596, 538,  602, 536,  594, 548,  594, 520,  598};  // WHIRLPOOL_AC
uint8_t state[6] = {0xC1, 0x60, 0x20, 0x8E, 0x00, 0x00};

6th_mode_offset_1C

Timestamp : 005070.666
Encoding  : WHIRLPOOL_AC
Code      : C160088E0000 (48 bits)
Library   : v2.4.3

uint16_t rawData[343] = {8950, 4484,  598, 1642,  598, 1646,  594, 534,  594, 538,  602, 532,  598, 540,  600, 542,  598, 1650,  600, 522,  598, 1644,  596, 1650,  600, 532,  598, 538,  602, 536,  594, 548,  592, 538,  602, 518,  600, 524,  596, 532,  598, 532,  598, 1654,  596, 544,  596, 544,  596, 536,  594, 1644,  596, 528,  600, 528,  592, 538,  602, 1648,  602, 1654,  596, 1664,  598, 534,  594, 526,  594, 530,  598, 528,  602, 530,  600, 534,  596, 542,  598, 542,  598, 534,  596, 526,  594, 530,  600, 528,  602, 530,  600, 534,  596, 542,  598, 544,  596, 518,  602, 7916,  598, 1642,  598, 528,  600, 528,  602, 530,  600, 1652,  598, 542,  598, 544,  596, 1654,  596, 1644,  596, 1648,  602, 1644,  596, 1654,  596, 1656,  604, 536,  594, 548,  602, 528,  600, 520,  600, 524,  596, 532,  598, 532,  596, 538,  602, 536,  594, 546,  594, 538,  602, 518,  600, 524,  596, 532,  598, 532,  598, 536,  594, 544,  596, 544,  596, 536,  594, 526,  592, 530,  600, 528,  600, 530,  602, 532,  596, 542,  598, 542,  598, 534,  596, 524,  596, 528,  600, 526,  592, 538,  592, 542,  598, 540,  600, 540,  600, 530,  598, 522,  598, 526,  594, 534,  596, 534,  594, 540,  602, 536,  592, 548,  592, 538,  600, 1636,  594, 1648,  602, 1642,  598, 1652,  598, 538,  602, 1680,  570, 1662,  598, 1634,  596, 7924,  600, 520,  598, 526,  592, 534,  596, 534,  596, 540,  600, 536,  604, 538,  602, 530,  600, 520,  598, 1640,  600, 528,  600, 530,  600, 534,  594, 544,  596, 544,  596, 534,  596, 526,  594, 528,  600, 526,  594, 536,  592, 542,  598, 538,  602, 538,  602, 528,  600, 520,  600, 524,  596, 530,  600, 532,  598, 534,  596, 542,  598, 542,  598, 532,  598, 524,  596, 528,  602, 526,  594, 536,  594, 540,  600, 536,  594, 548,  592, 538,  602, 518,  602, 522,  596, 530,  600, 530,  600, 534,  596, 542,  598, 544,  596, 534,  596, 524,  594, 1644,  596, 532,  596, 534,  596, 538,  602, 536,  594, 546,  594, 520,  600};  // WHIRLPOOL_AC
uint8_t state[6] = {0xC1, 0x60, 0x08, 0x8E, 0x00, 0x00};

6th_sense_offset_2C

Timestamp : 004526.142
Encoding  : WHIRLPOOL_AC
Code      : C160048E0000 (48 bits)
Library   : v2.4.3

uint16_t rawData[343] = {8950, 4482,  598, 1640,  598, 1642,  596, 532,  598, 532,  598, 536,  604, 534,  594, 546,  594, 1652,  598, 524,  594, 1646,  604, 1640,  600, 532,  598, 536,  604, 534,  596, 544,  594, 536,  594, 526,  594, 530,  600, 526,  604, 526,  604, 530,  598, 1654,  596, 546,  604, 528,  602, 1634,  606, 520,  598, 528,  602, 528,  600, 1648,  602, 1652,  596, 1662,  598, 536,  594, 526,  594, 528,  600, 526,  602, 528,  602, 532,  598, 540,  600, 540,  600, 530,  600, 520,  598, 526,  604, 524,  596, 534,  596, 538,  600, 538,  602, 538,  602, 512,  596, 7920,  604, 1636,  604, 520,  598, 530,  600, 530,  598, 1652,  598, 542,  598, 544,  596, 1650,  600, 524,  596, 1642,  596, 1650,  600, 534,  598, 1652,  596, 544,  596, 546,  594, 536,  604, 516,  602, 522,  598, 530,  600, 532,  598, 536,  604, 536,  594, 548,  602, 528,  602, 518,  600, 524,  596, 532,  598, 532,  596, 538,  602, 536,  596, 546,  604, 526,  602, 520,  600, 522,  596, 530,  598, 532,  598, 536,  602, 536,  594, 546,  604, 526,  602, 518,  602, 522,  598, 530,  598, 532,  598, 538,  602, 534,  596, 546,  604, 528,  602, 518,  602, 522,  596, 532,  598, 534,  596, 538,  602, 536,  594, 548,  602, 528,  602, 518,  600, 1640,  600, 1644,  596, 538,  602, 1646,  604, 536,  604, 1654,  606, 1628,  602, 7916,  598, 524,  594, 530,  600, 528,  602, 528,  602, 534,  596, 542,  598, 544,  596, 534,  596, 526,  594, 1646,  604, 526,  604, 528,  602, 534,  594, 542,  598, 544,  596, 536,  594, 526,  604, 520,  600, 528,  602, 530,  600, 536,  594, 544,  596, 546,  594, 538,  602, 518,  600, 524,  596, 532,  598, 534,  596, 538,  602, 536,  604, 536,  602, 530,  600, 522,  598, 528,  602, 526,  594, 538,  602, 532,  598, 540,  600, 542,  598, 532,  598, 522,  596, 528,  602, 526,  602, 528,  602, 534,  596, 542,  598, 542,  598, 534,  596, 526,  594, 1644,  606, 526,  604, 528,  602, 532,  598, 540,  600, 542,  598, 516,  592};  // WHIRLPOOL_AC
uint8_t state[6] = {0xC1, 0x60, 0x04, 0x8E, 0x00, 0x00};

6th_sense_OFFSET_3C

Timestamp : 005082.410
Encoding  : WHIRLPOOL_AC
Code      : C160048E0000 (48 bits)
Library   : v2.4.3

uint16_t rawData[343] = {8948, 4486,  606, 1634,  596, 1646,  604, 526,  604, 528,  602, 534,  598, 540,  600, 542,  598, 1650,  600, 522,  596, 1644,  606, 1640,  598, 534,  596, 538,  602, 536,  604, 538,  602, 528,  600, 520,  600, 524,  596, 532,  598, 534,  594, 540,  600, 1652,  598, 546,  594, 536,  604, 1632,  598, 528,  602, 526,  594, 536,  604, 1646,  604, 1650,  598, 1660,  600, 530,  600, 522,  598, 524,  594, 532,  598, 532,  596, 538,  602, 536,  594, 546,  604, 526,  604, 516,  604, 520,  598, 530,  600, 530,  600, 534,  596, 542,  598, 542,  598, 516,  604, 7906,  598, 1640,  600, 524,  596, 532,  596, 534,  596, 1652,  598, 540,  600, 542,  598, 1646,  604, 1634,  596, 1646,  604, 1640,  600, 1648,  602, 1648,  602, 538,  602, 538,  602, 528,  602, 518,  600, 524,  596, 530,  600, 530,  600, 534,  594, 544,  596, 544,  596, 534,  596, 524,  594, 530,  600, 526,  594, 536,  604, 530,  600, 538,  602, 540,  600, 530,  600, 520,  598, 524,  596, 532,  598, 532,  598, 536,  594, 544,  596, 544,  596, 534,  596, 522,  596, 528,  602, 524,  594, 536,  594, 540,  600, 536,  604, 536,  604, 526,  602, 518,  602, 5 [... cut ...]
#### Circuit diagram and hardware used (if applicable)
I used an ESP-01 with some IR receiver

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
YES

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2018-08-30:
First off. Wow. Congrats on writing a sender and a decoder. Top effort! I look forward to the Pull Request when it's done!

Next. I'm fairly confident that the protocol you're dealing with is NOT 48bits. It's far far bigger. Like 168 bits/21 bytes long.
e.g.
```
Found 343 timing entries.
Potential Mark Candidates:
[8950, 604]
Potential Space Candidates:
[7924, 4484, 1680, 548]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 8950
kHdrSpace  = 4484
kBitMark   = 597
kOneSpace  = 1649
kZeroSpace = 533
kSpaceGap = 7920

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+110000010110000000001000100011100000000000000000GAP(7916)
  Bits: 48
  Hex:  0xC160088E0000 (MSB first)
        0x000071100683 (LSB first)
  Dec:  212618204545024 (MSB first)
        1896875651 (LSB first)
  Bin:  0b110000010110000000001000100011100000000000000000 (MSB first)
        0b000000000000000001110001000100000000011010000011 (LSB first)
kBitMark(UNEXPECTED)1000100111111000000000000000000000000000000000000000000011110111GAP(7924)
  Bits: 64
  Hex:  0x89F80000000000F7 (MSB first)
        0xEF00000000001F91 (LSB first)
  Dec:  9941696177420370167 (MSB first)
        17221764975064784785 (LSB first)
  Bin:  0b1000100111111000000000000000000000000000000000000000000011110111 (MSB first)
        0b1110111100000000000000000000000000000000000000000001111110010001 (LSB first)
kBitMark(UNEXPECTED)00000000010000000000000000000000000000000000000001000000
  Bits: 56
  Hex:  0x00400000000040 (MSB first)
        0x02000000000200 (LSB first)
  Dec:  70368744177728 (MSB first)
        562949953421824 (LSB first)
  Bin:  0b00000000010000000000000000000000000000000000000001000000 (MSB first)
        0b00000010000000000000000000000000000000000000001000000000 (LSB first)

Total Nr. of suspected bits: 168
```

Basically there are 3 sections to the entire message, separated by an approx 8000usec pause/space.
48 bits (which you've worked out), then 64 bits, and then a final 56 bits section.

In short, you've got a lot more message to decode (and send) than you previously thought before you're close to fully reproducing it.

Here is the _raw_ output from the analyse program for suggested code (Use the '-g' flag):
```
Generating a VERY rough code outline:

// WARNING: This probably isn't directly usable. It's a guide only.
const uint16_t kHdrMark = 8950;
const uint16_t kBitMark = 597;
const uint16_t kHdrSpace = 4484;
const uint16_t kOneSpace = 1649;
const uint16_t kZeroSpace = 533;
const uint16_t kSpaceGap = 7920;
const uint16_t kXyzBits = 168;
const uint16_t kXyzStateLength = 21;
[... 59 lines cut ...]
                kBitMark, kZeroSpace,
                kBitMark
                kSpaceGap,
                data, 6,  // 6 bytes == 48 bits
                38000, // Complete guess of the modulation frequency.
                true, 0, 50);
    // Section 2
    sendGeneric(kHdrMark, kHdrSpace,
                kBitMark, kOneSpace,
                kBitMark, kZeroSpace,
                kBitMark
                kSpaceGap,
                data + 6, 8,  // 8 bytes == 64 bits
                38000, // Complete guess of the modulation frequency.
                true, 0, 50);
    sendGeneric(kHdrMark, kHdrSpace,
                kBitMark, kOneSpace,
                kBitMark, kZeroSpace,
                kBitMark
                100000,  // Completely made up "end of message" gap.
                data + 6 + 8, 7,  // 7 bytes == 56 bits
                38000, // Complete guess of the modulation frequency.
                true, 0, 50);
}
```

crankyoldgit, 2018-08-30:
FYI, I've added some comments in your git repo. Hope they help.

redmusicxd, 2018-08-30:
Ok, thank you for helping me, i'll keep you posted.

crankyoldgit, 2018-08-31:
I've created a PR/branch https://github.com/markszabo/IRremoteESP8266/tree/2.5-dev-whirlpool_ac which should send and decode the full 168 bits of the Whirlpool messages.

Can you give it a try please?

redmusicxd, 2018-08-31:
I think this is all, I hope you can help me implement all of this cause I don't  quite understand how this works
[Codes_whirlpool.txt](https://github.com/markszabo/IRremoteESP8266/files/2341884/Codes_whirlpool.txt)

crankyoldgit, 2018-09-01:
Also, another tip. In the spreadsheet, record all the settings for a remote's message/value.
e.g. mode, temp, fan, clock, turbo, sleep, etc etc. It will help in the long run.

redmusicxd, 2018-09-01:
Here is the data requested:[Google Sheets](https://docs.google.com/spreadsheets/d/1ukDBLcyO1tuC8ndHScNZSDD-Lr1VlAjPaB72wdfyKts/edit?usp=sharing)
 and also the [manual](http://docs.whirlpool.eu/_doc/853001901100.pdf)

crankyoldgit, 2018-09-02:
Yep. The temperature indicates the bit ordering is LSB first. Not MSB. i.e. bit order is reversed for each "byte"/octet etc.
e.g. In the 4th transmitted byte which stores the temp.
```
18C = 0x8 (MSB) = 0x1 (LSB)
19C = 0x4 (MSB) = 0x2 (LSB)
20C = 0xC (MSB) = 0x3 (LSB)
21C = 0x2 (MSB) = 0x4 (LSB)
...
30C = 0xB (MSB) = 0xD (LSB)
31C = 0x7 (MSB) = 0xE (LSB)
```
Makes far more sense in LSB order.

crankyoldgit, 2018-09-02:
BTW, don't be afraid to say if the data looks better in MSB order than in LSB order. It might just be they are storing/recording the temperature data oddly.
i.e. If plenty of other things look more logical in MSB vs in LSB order. Speak up. You're doing the leg work on the analysis, so it's your call.

crankyoldgit, 2018-09-14:
@redmusicxd Any luck with the [v2.5-dev](https://github.com/markszabo/IRremoteESP8266/tree/v2.5-dev) branch and any further progress on deep message decoding?

crankyoldgit, 2018-09-20:
Marking this issue closed until there is feedback on deep message decoding. Just update this issue and I'll reopen it. The basic sending/decoding support has been merged in to the 2.5 branch and will be released soon.

crankyoldgit, 2018-09-20:
v2.5.0 has been released.

crankyoldgit, 2018-10-07:
FYI, I'll try to take a look soon but I'm travelling for several weeks. So don't expect a lot of response from me for a bit.

redmusicxd, 2018-11-03:
Hello!, it's been almost a month since your last comment, could you at least give a thought at how is the checksum calculated, I don't have any experience in such things and I don't know how to begin so I really need your help, sorry if you are still traveling and I am bothering you.

crankyoldgit, 2018-11-03:
@redmusicxd I'm still travelling (for at least two more weeks).

In the meantime, can you include in the Google Spreadsheet (https://docs.google.com/spreadsheets/d/1ukDBLcyO1tuC8ndHScNZSDD-Lr1VlAjPaB72wdfyKts/edit#gid=0) a sequence of codes captured from the **_v2.5.2_** library where all the settings are the same, except varying the temperature. Starting from the lowest temp (typically 16degC) to the maximum temp (typically 30-32C), including each increment, and list what the other unchanging settings are:
e.g.
Power On, Cool Mode, fan auto, swing off, any-other-modes off, time 0:00, temp 16C
Power On, Cool Mode, fan auto, swing off, any-other-modes off, time 0:00, temp 17C
Power On, Cool Mode, fan auto, swing off, any-other-modes off, time 0:00, temp 18C
...
Power On, Cool Mode, fan auto, swing off, any-other-modes off, time 0:00, temp 30C

That will help us to determine what algorithm the checksum might be using.

raducostea, 2018-11-05:
Wow, this is really great. Is there a way to have different codes for ON and OFF? Wanting to control this remotely but i will never know if i turned it off or on since same code is being used for ON and OFF.

crankyoldgit, 2018-11-05:
@raducostea Not sure. The data in @redmusicxd 's spreadsheet indicates the state does contain info about if the power is on or off, but no idea if there is a descrete on & off signal that can be used. You'll have to get a response from them as I don't have access to the device.

Have you tried contacting the manufacturer and asking them if there is such a code for what you want?
Have you captured the on & off signals with IRrecvDumpV2? Do the codes differ at all?

raducostea, 2018-11-05:
@crankyoldgit unfortunately not, they don't change. Currently, to "safely" turn it on and off while not at home i am basically programming to ON in one minute and same for OFF, off in one minute. This will make sure its ON or OFF when i send the command.

crankyoldgit, 2018-11-05:
That's a real bummer for you/Whirlpool owners. BTW @raducostea if you can provide the state lines/codes for the above min temp to max temp in cool mode etc above, that would be a big step to help me work out what the checksum algorithm might be.

raducostea, 2018-11-06:
@crankyoldgit ,
I requested access to the document. I can provide the codes for you.

crankyoldgit, 2018-11-06:
I don't control access to that doc. Feel free to create a new stuff in the meantime.

raducostea, 2018-11-06:
@crankyoldgit posted more codes here https://docs.google.com/spreadsheets/d/1SUMVL0_LN6fI_3e_GeEPigtIe2B_PvkzKfDIA2mBiHM/edit#gid=0

crankyoldgit, 2018-11-06:
Can you share a 'public read-only' link please, so anyone following this thread can access it. It's in the sharing settings.

Thanks

raducostea, 2018-11-06:
@crankyoldgit sure https://docs.google.com/spreadsheets/d/1SUMVL0_LN6fI_3e_GeEPigtIe2B_PvkzKfDIA2mBiHM/edit?usp=sharing

Let me know if you need extra information. Looking forward to have this ready, i have couple of AC commands working already by sending uint8_t codes but its hard to map all of them. 

Btw, in the IRDumpV2 clicking power on/off on head mode and cool mode (same time) results in the same  uint8 state

> 
> Encoding  : WHIRLPOOL_AC
> Code      : 830604500000872B0000000000F800010000000001 (168 bits)
> uint16_t rawData[343] = {9088, 4446,  676, 1580,  674, 1582,  674, 490,  646, 494,  646, 496,  644, 502,  648, 502,  646, 1590,  674, 482,  646, 1584,  676, 1584,  676, 492,  644, 498,  648, 498,  646, 502,  646, 492,  648, 482,  646, 486,  644, 1588,  674, 494,  648, 496,  646, 500,  646, 504,  646, 494,  646, 484,  644, 488,  646, 488,  648, 492,  646, 1594,  674, 502,  646, 1600,  676, 492,  644, 484,  646, 486,  646, 490,  644, 494,  648, 496,  648, 500,  644, 504,  646, 494,  646, 484,  644, 488,  646, 490,  646, 494,  646, 498,  646, 500,  648, 502,  646, 476,  648, 7900,  676, 1578,  678, 1582,  674, 1588,  672, 494,  644, 498,  646, 498,  648, 502,  646, 1590,  674, 1580,  674, 1586,  674, 490,  646, 1590,  678, 496,  648, 1596,  676, 502,  646, 494,  646, 484,  646, 484,  648, 488,  646, 494,  644, 498,  644, 500,  648, 500,  648, 492,  644, 484,  646, 486,  648, 488,  646, 492,  646, 496,  646, 500,  646, 504,  648, 492,  644, 484,  644, 488,  646, 488,  646, 492,  644, 498,  646, 500,  646, 502,  646, 492,  642, 486,  644, 486,  646, 490,  644, 492,  646, 496,  646, 500,  646, 504,  644, 494,  646, 484,  646, 486,  648, 488,  648, 490,  646, 496,  644, 500,  648, 502,  644, 494,  648, 482,  644, 488,  646, 488,  648, 1592,  672, 1596,  674, 1598,  676, 1598,  676, 1572,  678, 7898,  678, 482,  646, 486,  646, 490,  644, 496,  646, 496,  646, 502,  646, 504,  644, 494,  646, 1580,  676, 486,  646, 490,  646, 494,  646, 498,  646, 500,  646, 504,  644, 496,  644, 484,  648, 484,  646, 490,  646, 492,  644, 498,  646, 498,  648, 502,  646, 492,  644, 484,  646, 486,  646, 490,  646, 494,  646, 496,  648, 498,  644, 506,  646, 492,  646, 484,  646, 486,  646, 490,  646, 492,  646, 498,  646, 500,  644, 506,  648, 492,  644, 484,  646, 488,  646, 490,  646, 492,  644, 498,  646, 502,  646, 504,  646, 492,  646, 1578,  676, 486,  646, 492,  646, 492,  644, 498,  644, 500,  646, 504,  648, 474,  632};  // WHIRLPOOL_AC
> uint8_t state[21] = {0x83, 0x06, 0x04, 0x50, 0x00, 0x00, 0x87, 0x2B, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x01};
> 
> Encoding  : WHIRLPOOL_AC
> Code      : 830604520000872B0000000000FA00010000000001 (168 bits)
> uint16_t rawData[343] = {9080, 4452,  670, 1614,  640, 1592,  666, 468,  668, 470,  668, 474,  668, 478,  668, 480,  668, 1598,  668, 458,  672, 1616,  640, 1594,  666, 470,  668, 474,  668, 478,  668, 482,  668, 472,  668, 458,  670, 460,  670, 1596,  666, 472,  668, 474,  668, 476,  670, 480,  670, 468,  670, 460,  670, 1592,  666, 466,  672, 468,  670, 1602,  666, 478,  668, 1636,  640, 470,  668, 458,  670, 464,  668, 468,  666, 470,  670, 472,  668, 478,  670, 480,  668, 468,  670, 462,  668, 464,  668, 466,  668, 472,  666, 476,  666, 478,  670, 480,  668, 454,  670, 7912,  664, 1592,  666, 1596,  664, 1598,  664, 472,  666, 476,  666, 482,  664, 484,  666, 1602,  664, 1590,  662, 1598,  662, 468,  666, 1628,  636, 478,  666, 1636,  634, 484,  664, 474,  666, 464,  664, 400,  730, 470,  666, 474,  666, 476,  664, 482,  664, 484,  664, 474,  664, 466,  662, 468,  664, 472,  664, 476,  666, 478,  662, 482,  666, 486,  662, 476,  664, 466,  664, 470,  664, 474,  664, 472,  666, 480,  664, 482,  662, 488,  662, 476,  664, 466,  664, 468,  664, 474,  662, 476,  664, 480,  662, 484,  664, 486,  664, 476,  664, 464,  664, 470,  662, 472,  664, 476,  664, 480,  664, 484,  662, 486,  662, 478,  664, 466,  660, 1626,  632, 476,  662, 1632,  634, 1636,  634, 1640,  632, 1616,  660, 1588,  660, 7918,  664, 468,  660, 472,  662, 476,  662, 476,  662, 480,  662, 484,  662, 492,  660, 476,  662, 1620,  634, 470,  662, 472,  664, 476,  662, 482,  662, 484,  662, 488,  662, 476,  664, 468,  662, 470,  660, 478,  662, 476,  662, 480,  662, 486,  660, 490,  664, 476,  662, 466,  662, 470,  664, 474,  662, 476,  662, 482,  664, 484,  662, 486,  660, 480,  660, 470,  660, 474,  658, 474,  660, 478,  660, 484,  660, 484,  658, 490,  662, 478,  662, 468,  658, 474,  658, 478,  658, 482,  658, 484,  658, 486,  660, 490,  662, 478,  660, 1596,  656, 476,  660, 476,  660, 480,  658, 482,  658, 490,  658, 494,  656, 462,  624};  // WHIRLPOOL_AC
> uint8_t state[21] = {0x83, 0x06, 0x04, 0x52, 0x00, 0x00, 0x87, 0x2B, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFA, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x01};

crankyoldgit, 2018-11-06:
FYI. Those two states are different

redmusicxd, 2018-11-06:
@raducostea I gave you access to the document if you want to make further additions.

raducostea, 2018-11-06:
Thaks @redmusicxd  
@crankyoldgit is there a way to send the 168 bits code right now? Will be easier for mapping until the library is ready.

crankyoldgit, 2018-11-06:
Try `sendWhirlpool(state);`

raducostea, 2018-11-06:
That works, i was thinking about the 168 bits code

crankyoldgit, 2018-11-06:
That *is* 168 bits. 21 * 8 = 168 bits. The longest integer size the ESP8266 supports is 64 bits. (i.e. `uint64_t`/`unsigned long long`) Hence you need to store it in an array.

crankyoldgit, 2018-11-06:
@raducostea Thanks for the data/spreadsheet. Looking at that data, I think the checksum may be based on an [XOR](https://en.wikipedia.org/wiki/Exclusive_or) function of the preceding bytes.
I still need to check that out with some firm analysis.

redmusicxd, 2018-11-06:
I checked the XOR but it would never return me the same checksum instead it would be an almost value but anyway must be a miscalculation or there are some bits that I may have missing, anyway good luck.

crankyoldgit, 2018-11-06:
Yeah. Testing that against three codes. It seems to work.
i.e. XOR of state[2..13] gives the checksum which seems to be in state[14]

josh929800, 2018-11-07:
[IR Remote Code modified4.xlsx](https://github.com/markszabo/IRremoteESP8266/files/2559046/IR.Remote.Code.modified4.xlsx)
I have 4 SAGA AC units using the whirlpool protocol I have attached my Spread sheet if that helps someone here. There are two checksums for my units and they exclude the preamble (the first part of the IR code and the checksum it's self. I also found https://www.scadacore.com/tools/programming-calculators/online-checksum-calculator/ to be a good tool for helping me solve the checksum on my ac units.

crankyoldgit, 2018-11-08:
@josh929800 Many thanks for that. That's really useful.

I'm still away from home for another week. So, I may not be able to get to coding it till then.

josh929800, 2018-11-08:
I am using HomeAssistant mqtt climate module to talk to nodeRed via MQTT. Then I have nodeRed generate the IR code to send to the ESP8266 via MQTT. If it helps you or someone else here is the nodeRed logic that generates the IR code and sends it to the AC unit. 

```nodered
[{"id":"bb7c54cc.538af8","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"83f32013.fb9a2","type":"inject","z":"bb7c54cc.538af8","name":"On","topic":"frontNode1/ac/power/set","payload":"On","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":170,"y":740,"wires":[["fcc4735.740db9"]]},{"id":"fcc4735.740db9","type":"switch","z":"bb7c54cc.538af8","name":"Power Bit","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"Off","vt":"str"},{"t":"eq","v":"On","vt":"str"}],"checkall":"true","repair":false,"outputs":2,"x":420,"y":800,"wires":[["8732aafe.a0a518"],["de3e7dd7.2753d"]]},{"id":"c6339d74.f1a6b","type":"function","z":"bb7c54cc.538af8","name":"modeBlock","func":"var acMode = flow.get(\"ac-mode\")||\"Standby\";\nvar fanSpeed = flow.get(\"fanspeed\")||\"00\";\nvar power = flow.get(\"power\")||\"0\";\nvar swing = flow.get(\"swing\")||\"0\";\nvar sleep = flow.get(\"sleep\")||\"0\";\nvar Super = flow.get(\"Super\")||\"0\";\nvar swingHex = \"0\";\nvar mode = flow.get(\"mode\")||\"4\"; \nif (Super === \"1\" && fanSpeed !== \"10\") Super = \"0\";\nif (mode === \"2\") {\n  var setTemp = flow.get(\"setTemp\")||\"8\"; // AC Default Temp 24\n} else if (mode === \"0\") {\n  var setTemp = flow.get(\"setTemp\")||\"B\"; // Heat Default Temp 27\n} else{\n  var setTemp = \"7\"; // Default Temp 23\n}\nif (swing === \"1\") {\n    swingHex = \"8\";\n}\n\nvar nowHourBinary = flow.get(\"nowHourBinary\")||\"00000\"; // 5 - 16H,8H,4H,2H,1H\nvar dimmer = flow.get(\"dimmer\")||\"0\"; // 1 - LED Off & O - LED On\nvar nowMinuteBinary = flow.get(\"nowMinuteBinary\")||\"000000\"; // 6 - 32M,16M,8M,4M,2M,1M\nif (acMode === \"Stanby\") {\n  var timerOffInBinary = flow.get(\"timerOffInBinary\")||\"1000000000001\";\n  var timerOnInBinary = flow.get(\"timerOnInBinary\")||\"0000000000000\";\n} else {\n  var timerOffInBinary = flow.get(\"timerOffInBinary\")||\"0000000000000\";\n  var timerOnInBinary = flow.get(\"timerOnInBinary\")||\"1000000000001\";\n}\nvar timerOffIO = timerOffInBinary.slice(0,-12); // 1 - I/O\nvar timerOffHour = timerOffInBinary.slice(1,-6); //6 - 32H,16H,8H,4H,2H,1H\nvar timerOffMin = timerOffInBinary.substr(7); // 6 - 32M,16M,8M,4M,2M,1M\nvar timerOnIO = timerOnInBinary.slice(0,-12); // 1 - I/O\nvar timerOnHours = timerOnInBinary.slice(1,-6); // 6 - 32H,16H,8H,4H,2H,1H\nvar timerOnMin = timerOnInBinary.substr(7); // 6 - 32M,16M,8M,4M,2M,1M\nvar ifeel = flow.get(\"ifeel\")||\"00\"; // I/O - Hex\nvar ifeelIO = ifeel.slice(0,-1);\nvar ifeelTempHex = ifeel.substr(1);\n\nvar inBin = sleep+power+fanSpeed+\"10\"+dimmer+nowHourBinary+timerOffIO+\"0\"+nowMinuteBinary+\"0\"+swing+timerOffHour+timerOnIO+\"0\"+timerOffMin+\"00\"+timerOnHours+\"00\"+timerOnMin+\"000\"+ifeelIO\n\nvar len = inBin.length;\n\n// determine how many bits in first nibble\nvar firstBits = len % 4;\n\n// if firstBits > 0 generate padding to make even nibbles\n// increase len by number of padding bits\nvar inPadded = \"\";\nif(firstBits > 0)\n{\n   var padding = \"\";\n   for(var i = 0; i < 4 - firstBits; i++)\n   {\n     padding = \"0\" + padding;\n     len++;\n   }\n   inPadded = padding + inBin;\n}\nelse\n{\ninPadded = inBin;\n}\n\n// Binary to hexadecimal lookup table in\n// a multidimensional array\nvar hexDigits = [\n[\"0000\",\"0\"],\n[\"0001\",\"1\"],\n[\"0010\",\"2\"],\n[\"0011\",\"3\"],\n[\"0100\",\"4\"],\n[\"0101\",\"5\"],\n[\"0110\",\"6\"],\n[\"0111\",\"7\"],\n[\"1000\",\"8\"],\n[\"1001\",\"9\"],\n[\"1010\",\"A\"],\n[\"1011\",\"B\"],\n[\"1100\",\"C\"],\n[\"1101\",\"D\"],\n[\"1110\",\"E\"],\n[\"1111\",\"F\"]];\n\n// grab the input bits nibble by nibble\n// look up in array and fetch hexadecimal value\nvar nibble = \"\";\nvar outHex = \"\";\nfor(var start = 0; start <= len - 4; start += 4)\n{\n   nibble = inPadded.substring(start, start + 4);\n   for(var j = 0; j < hexDigits.length; j++)\n   {\n      if(nibble == hexDigits[j][0])\n      {\n          outHex += hexDigits[j][1];\n      }\n   }\n}  \n\npowerFanHex = outHex.slice(0,-13);\noutHex = outHex.substr(1); //3\n\nhex = swingHex+powerFanHex+setTemp+mode+\"00\"+Super+\"0\"+outHex+ifeelTempHex;\nmsg.payload = hex;\nreturn msg;","outputs":1,"noerr":0,"x":230,"y":80,"wires":[["6864fa0a.f50b44"]]},{"id":"1484d8bf.5f2357","type":"switch","z":"bb7c54cc.538af8","name":"Fan","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"high","vt":"str"},{"t":"eq","v":"medium","vt":"str"},{"t":"eq","v":"low","vt":"str"},{"t":"eq","v":"auto","vt":"str"}],"checkall":"true","repair":false,"outputs":4,"x":410,"y":580,"wires":[["fa9b3d46.1c8ac"],["1caa22bc.c7fe9d"],["498cf32e.40921c"],["7d111777.d60b88"]]},{"id":"d68b8b52.903b88","type":"mqtt in","z":"bb7c54cc.538af8","name":"frontNode1/ac/fan/set","topic":"frontNode1/ac/fan/set","qos":"0","broker":"393ef689.1db1da","x":200,"y":640,"wires":[["1484d8bf.5f2357"]]},{"id":"7d111777.d60b88","type":"change","z":"bb7c54cc.538af8","name":"Auto","rules":[{"t":"set","p":"fanspeed","pt":"flow","to":"00","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":610,"y":640,"wires":[["bcf81a2b.b94cd8"]]},{"id":"fa9b3d46.1c8ac","type":"change","z":"bb7c54cc.538af8","name":"High","rules":[{"t":"set","p":"fanspeed","pt":"flow","to" [... cut ...] af8","name":"frontNode1/ir/sent","topic":"frontNode1/ir/sent","qos":"0","broker":"393ef689.1db1da","x":190,"y":380,"wires":[["6ed0d57e.6b756c","fb1f1e4.edbe8e"]]},{"id":"393ef689.1db1da","type":"mqtt-broker","z":"","name":"","broker":"localhost","port":"1883","clientid":"NodeRED","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]
```

crankyoldgit, 2018-11-14:
My travel is over, and I'll try to get to this in the next few days as I deal with jet lag etc.

josh929800, 2018-11-19:
Thank you for the update... I did some testing with the IRrecvDumpV2 app and here are my findings:

powerToggle was not working - Fixed the mask
fanSpeed was not working - Fixed the mask
Temp was off for one of my remotes - added a check for the second remote

ir_Whirlpool.cpp
line 141: // Return the set temp. in deg C
uint8_t IRWhirlpoolAc::getTemp() {
  uint8_t minTemp = kWhirlpoolAcMinTemp;
  if (((remote_state[kWhirlpoolAcRemotePos] & kWhirlpoolAcRemoteMask) >> 2) == 2) minTemp = kWhirlpoolAcMinTemp1;
  return ((remote_state[kWhirlpoolAcTempPos] & kWhirlpoolAcTempMask) >> 4) +minTemp;
}

ir_Whirlpool.h
line 41: const uint8_t kWhirlpoolAcFanMask = 0b00000011;
line 51: const uint8_t kWhirlpoolAcPowerToggleMask = 0b00000100;
line 59 (New): const uint8_t kWhirlpoolAcRemoteMask = 0b00001000;
line 60 (New): const uint8_t kWhirlpoolAcRemotePos = 18;

Results before the changes...

Remote 1 
Note Temp was 19
Timestamp : 000027.938
Encoding  : WHIRLPOOL_AC
Code      : 83060130000092360000000000950002000008000A (168 bits)
Mesg Desc.: Power toggle: Off, Mode: 0 (HEAT), Temp: 21C, Fan: 0 (AUTO), Swing: Off, Light: On, Clock: 18:54, On Timer: Off, Off Timer: Off, Command: 2 (TEMP)
Library   : v2.5.2

Raw Timing[343]:
   +  9036, -  4442,    +   656, -  1590,    +   654, -  1594,    +   652, -   476, 
   +   656, -   478,    +   658, -   482,    +   656, -   484,    +   658, -   488, 
   +   656, -  1604,    +   652, -   468,    +   656, -  1592,    +   654, -  1598, 
   +   654, -   478,    +   656, -   482,    +   656, -   484,    +   656, -   488, 
   +   656, -   478,    +   656, -  1616,    +   628, -   472,    +   656, -   474, 
   +   656, -   480,    +   654, -   482,    +   654, -   486,    +   656, -   488, 
   +   656, -   478,    +   654, -   468,    +   656, -   472,    +   654, -   476, 
   +   656, -   480,    +   654, -  1606,    +   652, -  1612,    +   650, -   490, 
   +   656, -   478,    +   654, -   468,    +   656, -   472,    +   654, -   478, 
   +   652, -   480,    +   656, -   482,    +   656, -   488,    +   654, -   490, 
   +   654, -   478,    +   656, -   468,    +   656, -   472,    +   654, -   474, 
   +   658, -   478,    +   656, -   482,    +   656, -   486,    +   656, -   488, 
   +   656, -   462,    +   656, -  7882,    +   656, -   468,    +   656, -  1594, 
   +   654, -   476,    +   656, -   478,    +   656, -  1604,    +   654, -   484, 
   +   656, -   488,    +   656, -  1604,    +   650, -   468,    +   656, -  1594, 
   +   654, -  1598,    +   654, -   478,    +   656, -  1604,    +   654, -  1608, 
   +   654, -   490,    +   656, -   480,    +   656, -   468,    +   656, -   470, 
   +   656, -   474,    +   656, -   478,    +   656, -   482,    +   656, -   484, 
   +   656, -   488,    +   658, -   478,    +   656, -   466,    +   656, -   470, 
   +   656, -   476,    +   656, -   478,    +   656, -   480,    +   658, -   484, 
   +   658, -   488,    +   656, -   476,    +   658, -   466,    +   658, -   472, 
   +   656, -   476,    +   656, -   478,    +   656, -   480,    +   658, -   484, 
   +   658, -   488,    +   656, -   478,    +   656, -   468,    +   656, -   470, 
   +   656, -   474,    +   656, -   478,    +   656, -   482,    +   656, -   484, 
   +   656, -   488,    +   656, -   476,    +   656, -   468,    +   656, -   472, 
   +   656, -   476,    +   656, -   478,    +   658, -   480,    +   656, -   486, 
   +   656, -   488,    +   656, -   480,    +   656, -  1590,    +   654, -   472, 
   +   656, -  1596,    +   656, -   478,    +   656, -  1604,    +   654, -   486, 
   +   656, -   488,    +   656, -  1584,    +   654, -  7882,    +   656, -   468, 
   +   656, -   470,    +   656, -   474,    +   656, -   476,    +   658, -   480, 
   +   656, -   484,    +   656, -   488,    +   658, -   478,    +   658, -   468, 
   +   656, -  1594,    +   654, -   476,    +   656, -   478,    +   656, -   482, 
   +   656, -   484,    +   656, -   488,    +   656, -   478,    +   656, -   466, 
   +   658, -   472,    +   654, -   474,    +   656, -   478,    +   656, -   480, 
   +   658, -   484,    +   656, -   488,    +   656, -   478,    +   656, -   466, 
   +   658, -   470,    +   656, -   476,    +   656, -   478,    +   656, -   480, 
   +   658, -   484,    +   656, -   488,    +   658, -   476,    +   656, -   466, 
   +   658, -   468,    +   658, -   474,    +   656, -  1602,    +   652, -   482, 
   +   656, -   484,    +   658, -   488,    +   658, -   478,    +   658, -   466, 
[... 1192 lines cut ...]
   +   652, -  1600,    +   652, -   482,    +   654, -  1632,    +   626, -   488, 
   +   654, -   490,    +   654, -   482,    +   656, -   466,    +   656, -  1596, 
   +   654, -   474,    +   658, -  1600,    +   652, -  1606,    +   652, -  1610, 
   +   654, -   490,    +   654, -   478,    +   656, -   470,    +   654, -   474, 
   +   656, -   476,    +   654, -   482,    +   656, -   482,    +   656, -   486, 
   +   658, -   490,    +   654, -   478,    +   656, -   468,    +   654, -  1598, 
   +   652, -   476,    +   656, -   482,    +   654, -   484,    +   656, -  1610, 
   +   652, -  1616,    +   652, -  1588,    +   654, -  7892,    +   656, -   468, 
   +   656, -   472,    +   654, -   480,    +   654, -   478,    +   656, -   482, 
   +   656, -   488,    +   654, -   492,    +   654, -   480,    +   658, -  1592, 
[... cut ...] , 476,  656, 478,  654, 482,  656, 486,  656, 490,  656, 480,  656, 470,  656, 472,  654, 478,  652, 1600,  656, 480,  656, 486,  656, 488,  656, 480,  656, 470,  654, 472,  654, 476,  656, 480,  654, 484,  656, 486,  656, 490,  654, 480,  654, 1590,  654, 474,  654, 1598,  656, 478,  656, 482,  654, 488,  654, 488,  656, 462,  608};  // WHIRLPOOL_AC
uint8_t state[21] = {0x83, 0x06, 0x00, 0x71, 0x00, 0x00, 0x93, 0x8D, 0x00, 0xA0, 0x17, 0x3A, 0x00, 0xE2, 0x00, 0x0D, 0x00, 0x00, 0x08, 0x00, 0x05};

redmusicxd, 2018-11-19:
I also tested the decoder and here is the result:
1.the temp is working as expected
2.mode is working as expected
3.Clock is just right
4.Fan Mode is always on AUTO
5.Swing is never On
6.Light is never On
I'll provide some data when i can.

crankyoldgit, 2018-11-19:
@josh929800 Thanks for all of that. The data is super helpful. Yes. You are correct. I was off by a nibble in my calculations for Power and Fan speed. Fixed.
I'm intrigued by the difference in remotes. I'm glad you've found the distinguishing bit. Can you tell me what the model numbers are for the "Remote 1" and "Remote 2" please? I'll need to add a `set/getModel()` so we can handle temperatures correctly.
[Q] Do both remotes work correctly on all the units? i.e. Are they interchangeable?

@redmusicxd Ditto for you. Can you please tell me what the model number is for your remote? Swing appears to be working for one of @josh929800's remotes, so I'm going to need some `state[21]` lines from you with some descriptions to the Swing & Light(dim) behavior on yours to get it working for you.

@josh929800 I trimmed your earlier response. If the code is detecting it fine as Whirlpool etc and everything is _mostly_ okay, you don't need to include all the `rawData` stuff. Just the `Mesg Desc.:` and `uint8_t state[]` lines are all that's need when we are working out the sub-parts of the "working" protocol decode.

redmusicxd, 2018-11-19:
The Remote Model is: DG11J1-3A

crankyoldgit, 2018-11-19:
Basic model detection has now been added to that branch.
@josh929800 Re: Temperature.Can you please test it with your "Remote 1", especially can you determine what the maximum temperature is for that remote model. I suspect it's 30C.

josh929800, 2018-11-19:
Remote 1 (18-32C) Saga by Tornado DG11J1-04 and Remote 2 (16-30C) DG11J1-91

josh929800, 2018-11-19:
All is working for me... 😃
DG11J1-91 (using new code)
Code      : 830602020000821A0000000000980002000008000A (168 bits)
Mesg Desc.: Power toggle: Off, Mode: 2 (COOL), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Light: On, Clock: 02:26, On Timer: Off, Off Timer: Off, Command: 2 (TEMP)
Code      : 830682020000821D40000000005F0007000008000F (168 bits)
Mesg Desc.: Power toggle: Off, Mode: 2 (COOL), Temp: 16C, Fan: 0 (AUTO), Swing: On, Light: On, Clock: 02:29, On Timer: Off, Off Timer: Off, Command: 7 (SWING)
Code      : 830602020000A21D0000000000BF00000000080008 (168 bits)
Mesg Desc.: Power toggle: Off, Mode: 2 (COOL), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Light: Off, Clock: 02:29, On Timer: Off, Off Timer: Off, Command: 0 (LIGHT)

DG11J1-04
Code      : 830600020000931600000000008700020000000002 (168 bits)
Mesg Desc.: Power toggle: Off, Mode: 2 (COOL), Temp: 18C, Fan: 0 (AUTO), Swing: Off, Light: On, Clock: 19:22, On Timer: Off, Off Timer: Off, Command: 2 (TEMP)
Code      : 830680020000931940000000004800070000000007 (168 bits)
Mesg Desc.: Power toggle: Off, Mode: 2 (COOL), Temp: 18C, Fan: 0 (AUTO), Swing: On, Light: On, Clock: 19:25, On Timer: Off, Off Timer: Off, Command: 7 (SWING)
Code      : 830600020000B31A0000000000AB00000000000000 (168 bits)
Mesg Desc.: Power toggle: Off, Mode: 2 (COOL), Temp: 18C, Fan: 0 (AUTO), Swing: Off, Light: Off, Clock: 19:26, On Timer: Off, Off Timer: Off, Command: 0 (LIGHT)

crankyoldgit, 2018-11-20:
Thanks @josh929800 However I think you were not using the correct code in the above results. You may have been still using your own/modified code. The code in that branch should have also listed the "Model:" info in the "Mesg Desc.:". I added that at the same time.

Eitherway, I've made some more changes to that branch which should report your new model info you supplied and handle max temp setting etc. Please try that out.

@josh929800 @raducostea @redmusicxd 
I still have Smart, IFeel, Quiet, Turbo (Super?) etc support to still workout and add. Am I missing any other obvious/important modes?

crankyoldgit, 2018-11-20:
@josh929800 What does "Super" do on your device? The manual I have has "Jet" and "Super Silent" modes. I'm guessing the later. Correct?

closed 2018-11-20 by crankyoldgit

crankyoldgit, 2018-11-20:
Code for all of this is now in the `master` branch and will be included in the next release.

If you have any issues or feature requests, please create a new issue, as this issue is now closed (For the second time ;-)

crankyoldgit, 2018-11-23:
v2.5.3 has been release. It includes the code discussed here.

MorisBR, 2024-01-17:
Actually, there is a code on 18th byte of raw state that inform when control send to turn on and off. this code is 0x2 (on) and 0x3 (off). So it need to be implemented.
Attached is my suggestion for implementation.
[ir_Whirlpool.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/13958008/ir_Whirlpool.zip)

PR #572 (2018-11-20): Deep decoding for Whirlpool A/C
Support deep decoding and creation of Whirlpool A/C messages
- Supports two different models.
- Unit test coverage for all supported modes/settings.

- Modes fully supported:
  - Toggling power mode
  - Cool/Heat/Fan
  - Clock/On & Off Timer
  - Temperature
  - Swing
  - Light/Dimmer
- Partial Support:
  - Dry mode.
  - 6th Sense (Auto)
  - Super/Jet
  - Sleep
  - Button Pressed.
- Not supported:
  - iFeel
  - AboutU
  - Quiet
  - Smart
  - Powersave

Help from @josh929800 @raducostea @redmusicxd

Fixes #509
commit 2cb74450 (2018-11-20): Deep decoding for Whirlpool A/C (#572)

Support deep decoding and creation of Whirlpool A/C messages

- Supports two different models.
- Unit test coverage for all supported modes/settings.
- Modes fully supported:
  - Toggling power mode
  - Cool/Heat/Fan
  - Clock/On & Off Timer
  - Temperature
  - Swing
  - Light/Dimmer
- Partial Support:
  - Dry mode.
  - 6th Sense (Auto)
  - Super/Jet
  - Sleep
  - Button Pressed.
- Not supported:
  - iFeel
  - AboutU
  - Quiet
  - Smart
  - Powersave

Help from @josh929800 @raducostea @redmusicxd

Fixes #509