crankyoldgit/IRremoteESP8266 #515: Sending signals to Lutron Devices

jxwolstenholme, 2018-09-07:
Hi all,

Wondering if anybody might have had any luck sending codes with the Lutron protocol - or even getting raw codes to be read by any of their devices.

The codes that they provide in their manual don't seem to be very long and are perhaps sent at a slightly odd timing? Can anybody shed some light as to why they might not work out of the box with this library?

Full Off (Pronto Hex): 0000 0069 0007 0000 032a 010e 005a 010e 005a 005a 0168 005a 005a 010e 005a 00b4 005a 01c2

Full Off (Lutron Protocol): FF88BD120h

Full Off (Raw - Generated from IrScrutinizer): +20518 -6839 +2280 -6839 +2280 -2280 +9119 -2280 +2280 -6839 +2280 -4560 +2280 -11399

The following is provided in their user guide: http://www.lutron.com/TechnicalDocumentLibrary/048158.doc

IR Carrier Frequency: 40.0 kHz 
Duty Cycle: 40% 
Single Bit Time: 2.288 milliseconds 
Baud Rate: 437 bps 
Command Length: 36 bits 
Command Duration: 82.368 milliseconds 
Logic One: Presence of IR modulated at 40.0 kHz 
Logic Zero: Absence of IR 
Transmit Order: Transmit the most significant bit first 
General Function: IR code is transmitted while a button is held down 
Timeout Function: Timeouts may not occur until at least seven-seconds of continuous IR transmission has taken place.

[image]

I'm very new to this and am unsure as to howI would translate the above in to a protocol template that would work with this library.

ANY help or guidance would be greatly appreciated.

Many Thanks

crankyoldgit, 2018-09-07:
After a quick look at the raw data you provided (and that I decoded from the Pronto code) it looks like it is some sort of multi-value mark & space encoding.
```
tools$ ./gc_decode -prontohex "0000 0069 0007 0000 032a 010e 005a 010e 005a 005a 0168 005a 005a 010e 005a 00b4 005a 01c2"
Code length 18
Code type      -1 (UNKNOWN)
Code bits      7
Code value     0xf59156f4
Code address   0x0
Code command   0x0
uint16_t rawbuf[14] = {
    20250, 6750, 2250, 6750, 2250, 2250, 9000, 2250, 
    2250, 6750, 2250, 4500, 2250, 11250};
```

According to the info you listed, every ~2250usecs is a bit and there are a total of 36 of them.
rawbuf[0] is 0b111111111 {9 bits} (e.g. 2250 * 9 = 20250, even array values are Mark/On. i.e. 1's)
rawbuf[1] is 0b000 {3 bits} (e.g. 2250 * 3 = 6750, odd array values are Space/Off. i.e. 0's)
rawbuf[2] is 0b1 (1 bit} (e.g. 2250 * 1 = 2250, even array values are Mark/On. i.e. 1's)
rawbuf[3] is 0b000 (3)
rawbuf[4] is 0b1 {1}
rawbuf[5] is 0b0 {1}
rawbuf[6] is 0b1111 {4}
rawbuf[7] is 0b0 {1}
rawbuf[8] is 0b1 {1}
rawbuf[9] is 0b000 {3}
rawbuf[10] is 0b1 {1}
rawbuf[11] is 0b00 {2}
rawbuf[12] is 0b1 {1}
rawbuf[13] is 0b00000 {5}
Sum of {}'s is 36 bits.
The data is 0b111111111000100010111101000100100000 = 0xFF88BD120

Now. We could write a sender routine for this, and a decoder, but if you don't really have a remote then it's not exactly urgent. And .. it's quite unusual.
If there are existing code (from where ever you got the ones you listed.) then I would *strongly* suggest you use those.
e.g. See the [sendRaw()](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino#L68) and [sendPronto()](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendProntoDemo/IRsendProntoDemo.ino) in the examples.

crankyoldgit, 2018-09-07:
e.g. These would be how you send the message you used in the example:
```
// Pronto code
uint16_t pronto[18] = {0x0000, 0x0069, 0x0007, 0x0000. 0x032a, 0x010e, 0x005a, 0x010e,
    0x005a, 0x005a, 0x0168, 0x005a, 0x005a, 0x010e. 0x005a, 0x00b4, 0x005a, 0x01c2);
irsend.SendPronto(pronto, 18);  // Pronto code include the frequency in itself.

// Raw code
uint16_t rawbuf[14] = {
    20250, 6750, 2250, 6750, 2250, 2250, 9000, 2250, 
    2250, 6750, 2250, 4500, 2250, 11250};
irsend.sendRaw(rawbuf, 14, 40000);  // 40kHz.
```

jxwolstenholme, 2018-09-07:
Thanks for the pointers @crankyoldgit - I'll give this a go

Is there any chance that you might give me some instructions as to how you generated that raw code?

I'm guessing that I must have been doing it wrong.

MUCH appreciated!!

crankyoldgit, 2018-09-07:
The program I used to convert the Pronto code to raw is the `gc_decode` tool at https://github.com/markszabo/IRremoteESP8266/blob/master/tools/gc_decode.cpp

or you could use the values in the raw array from your example data:
```
Full Off (Raw - Generated from IrScrutinizer): +20518 -6839 +2280 -6839 +2280 -2280 +9119 -2280 +2280 -6839 +2280 -4560 +2280 -11399
```
Just remove the +'s and -'s. and put those values into the array for `sendRaw()`. They will be close enough.

crankyoldgit, 2018-09-07:
I've added experimental support for this protocol in https://github.com/markszabo/IRremoteESP8266/tree/v2.5-dev-lutron

Can you please try it out an let me know if it works for you?

crankyoldgit, 2018-09-15:
This has been merged into the v2.5-dev branch, and thus will be in the next major release.

crankyoldgit, 2018-09-20:
Marking this issue closed due to no feedback and experimental code added to support the protocol.

closed 2018-09-20 by crankyoldgit

crankyoldgit, 2018-09-20:
v2.5.0 has been released.

jxwolstenholme, 2018-09-25:
So sorry @crankyoldgit  - Went away for a couple of weeks and just got back to test it.

I can confirm that this does work - at least in my case - many thanks for making the effort!

Experimental code should definitely work for others too.

Thanks again.

crankyoldgit, 2018-09-26:
Glad to hear it. Can you confirm what you tried? i.e. sending, decoding, or both?

jxwolstenholme, 2018-09-26:
Unfortunately only got to try sending the codes that I had as I've never had a remote for the system to be able to decode anything with.

You are correct about using a bit length of 35 though. I believe that this is standard across all of their IR products.

crankyoldgit, 2018-09-26:
Cool. Thanks for the confirmation. I'll mark sending as working then.
FYI, If you do send a 36-bit data to the routine, by default it will just ignore the first bit.
i.e It will send only the lower 35-bits.